pikvm/ustreamer #2: Feature Suggestion: MJPG-Source Support

chron0, 2019-03-03:
I was looking for mjpg-streamer alternatives and found ustreamer. I've played a little with it now and like it so far but without being able to use the MJPG source stream of the cam, my max fps is capped by the cam itself to 5 fps. Only the MJPG stream delivers FHD@30fps. I'd really love to try it again if you get around to teach it to also just forward the MJPG stream coming from the device. Building itself went without issues and it runs smoothly here. Thanks for sharing.

mdevaev, 2019-03-03:
I have not needed to add support for the MJPG, but I will try to do it for you :)

chron0, 2019-03-03:
Not for me per se, for all of us out there who'd want a little more choice than just mjpg-streamer...

```
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : PTZ Pro Camera
	Bus info      : usb-0000:0e:00.0-2.4
	Driver version: 4.18.7
	Capabilities  : 0x84A00001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1920/1080
	Pixel Format      : 'MJPG'
	Field             : None
	Bytes per Line    : 0
	Size Image        : 4147200
	Colorspace        : sRGB
	Transfer Function : Default
	YCbCr Encoding    : Default
	Quantization      : Default
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1920, Height 1080
	Default     : Left 0, Top 0, Width 1920, Height 1080
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1920, Height 1080
Selection: crop_bounds, Left 0, Top 0, Width 1920, Height 1080
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation (int)    : min=0 max=255 step=1 default=128 value=110
 white_balance_temperature_auto (bool)   : default=1 value=0
                           gain (int)    : min=0 max=255 step=1 default=0 value=0
           power_line_frequency (menu)   : min=0 max=2 default=2 value=0
      white_balance_temperature (int)    : min=2000 max=7500 step=1 default=4000 value=3300
                      sharpness (int)    : min=0 max=255 step=1 default=128 value=135
         backlight_compensation (int)    : min=0 max=1 step=1 default=0 value=0
                  exposure_auto (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute (int)    : min=3 max=2047 step=1 default=250 value=77
         exposure_auto_priority (bool)   : default=0 value=1
                   pan_relative (int)    : min=-4480 max=4480 step=1 default=0 value=200
                  tilt_relative (int)    : min=-1920 max=1920 step=1 default=0 value=-200
                      pan_reset (button) : flags=write-only
                     tilt_reset (button) : flags=write-only
                 focus_absolute (int)    : min=0 max=255 step=1 default=0 value=36
                     focus_auto (bool)   : default=1 value=0
                  zoom_absolute (int)    : min=100 max=1000 step=1 default=100 value=122
                      pan_speed (int)    : min=-1 max=1 step=1 default=1 value=0
                     tilt_speed (int)    : min=-1 max=1 step=1 default=1 value=0
                      led1_mode (menu)   : min=0 max=3 default=0 value=1
                 led1_frequency (int)    : min=0 max=255 step=1 default=0 value=0
```

```$ v4l2-ctl --list-formats-ext -d /dev/video0 

ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
[... 237 lines cut ...]
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1600x896
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
```

mdevaev, 2019-03-03:
Done 154192107033314d25238a53ccb5e816b491e41f
Please try it: `./ustreamer -m jpeg`.
I've also implemented hw jpeg quality control but i don't any webcams which support this feature :(
```
-- ERROR [54717.743 tid=11160] -- Can't query HW JPEG compressor params and set quality (unsupported)
```

chron0, 2019-03-03:
Thanks Maxim, great work, latest master working here:

``` 
./stream.sh                                                                                                                                              
-- INFO  [101283.243 tid=16956] -- Installing SIGINT handler ...
-- INFO  [101283.243 tid=16956] -- Installing SIGTERM handler ...
-- INFO  [101283.243 tid=16956] -- Ignoring SIGPIPE ...
-- INFO  [101283.243 tid=16956] -- Using JPEG quality: 100%
-- INFO  [101283.243 tid=16956] -- Listening HTTP on [0.0.0.0]:18000
-- INFO  [101283.243 tid=16957] -- Using V4L2 device: /dev/video0
-- INFO  [101283.243 tid=16957] -- Using desired FPS: 30
================================================================================
-- INFO  [101283.243 tid=16958] -- Starting HTTP eventloop ...
-- INFO  [101283.244 tid=16957] -- Device fd=8 opened
-- INFO  [101283.244 tid=16957] -- Using input channel: 0
-- INFO  [101283.244 tid=16957] -- Using TV standard: DEFAULT
-- INFO  [101283.244 tid=16957] -- Using resolution: 1920x1080
-- INFO  [101283.244 tid=16957] -- Using pixelformat: MJPG
-- INFO  [101283.264 tid=16957] -- Requested 17 HW buffers, got 17
-- INFO  [101283.340 tid=16957] -- Capturing started
-- INFO  [101283.340 tid=16957] -- Switching to HW encoder because the input format is (M)JPEG
-- ERROR [101283.340 tid=16957] -- Can't query HW JPEG compressor params and set quality (unsupported)
-- INFO  [101283.340 tid=16957] -- Spawning 17 workers ...
-- INFO  [101283.340 tid=16957] -- Capturing ...
```

It seems ustreamer consumes a little more CPU than mjpg-streamer does with similar settings, is there maybe any kind of re-encoding happening or any other idea why?

[image]
[image]

I'll wrap a small ebuild around it to make it easy for gentoo users to use as well.

mdevaev, 2019-03-03:
Try option `-n 1` to reduce worker threads. Also show me your /proc/cpuinfo plz

chron0, 2019-03-03:
I'll try to pin the workers and see what that does, gentoo ebuild is ready, shall I just paste it here or do you rather want a PR?

```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 44
model name	: Intel(R) Xeon(R) CPU           E5640  @ 2.67GHz
stepping	: 2
microcode	: 0x1d
cpu MHz		: 1846.991
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 popcnt aes lahf_lm epb pti tpr_shadow vnmi flexpriority ept vpid dtherm ida arat
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf
bogomips	: 5331.44
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 44
model name	: Intel(R) Xeon(R) CPU           E5640  @ 2.67GHz
stepping	: 2
microcode	: 0x1d
cpu MHz		: 2665.723
cache size	: 12288 KB
physical id	: 1
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 32
initial apicid	: 32
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 popcnt aes lahf_lm epb pti tpr_shadow vnmi flexpriority ept vpid dtherm ida arat
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf
bogomips	: 5331.28
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 44
model name	: Intel(R) Xeon(R) CPU           E5640  @ 2.67GHz
stepping	: 2
microcode	: 0x1d
cpu MHz		: 1755.698
cache size	: 12288 KB
physical id	: 0
siblings	: 8
core id		: 10
cpu cores	: 4
[... 340 lines cut ...]
cpu family	: 6
model		: 44
model name	: Intel(R) Xeon(R) CPU           E5640  @ 2.67GHz
stepping	: 2
microcode	: 0x1d
cpu MHz		: 2665.724
cache size	: 12288 KB
physical id	: 1
siblings	: 8
core id		: 9
cpu cores	: 4
apicid		: 51
initial apicid	: 51
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 popcnt aes lahf_lm epb pti tpr_shadow vnmi flexpriority ept vpid dtherm ida arat
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf
bogomips	: 5331.28
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management:
```

mdevaev, 2019-03-03:
It doesn't matter to me

chron0, 2019-03-03:
ustreamer-9999.ebuild

```
 # Copyright 2019 Gentoo Authors
 # Distributed under the terms of the GNU General Public License v2
 
 EAPI=7
 
 inherit git-r3
 
 DESCRIPTION="µStreamer - Lightweight and fast MJPG-HTTP streamer"
 HOMEPAGE="https://github.com/pi-kvm/ustreamer"
 EGIT_REPO_URI="https://github.com/pi-kvm/ustreamer.git"
 
 LICENSE="GPL-3"
 SLOT="0"
 KEYWORDS="~amd64"
 IUSE=""
 
 DEPEND=">=dev-libs/libevent-2.1.8
         >=media-libs/libjpeg-turbo-1.5.3
         >=sys-apps/util-linux-2.33"
 RDEPEND="${DEPEND}"
 BDEPEND=""
 
 src_install() {
     dobin ustreamer 
 }
```

mdevaev, 2019-03-03:
By default ustreamer spawn one worker per core for parallel compressing. In some cases, this may provide an additional overhead but you can adjust this parameter to reduce cpu consuming.

chron0, 2019-03-03:
ye, thats what I thought. My question is only, why does it need to compress? the data coming from the cam is already compressed MJPG, is it not?

mdevaev, 2019-03-03:
True, but no compression occurs. I think that the cpu load is associated with a large number of workers. Regardless of the compression method at least one thread is spawned to process data from the camera. It seems that 17 streams were simply trying to read the data from your camera in sequence.

chron0, 2019-03-03:
Now with -w 1:

[image]
[image]

mdevaev, 2019-03-03:
Try to add `refresh_interval.tv_usec = 8000;` in src/http.c after `}` in line 144 and run it again

chron0, 2019-03-03:
No real change that I can see. I was testing idle (about 10% cpu) and with a client (about 25%).

mdevaev, 2019-03-03:
Now I have no idea what the reason can be. I thought that it could be the http stream update timer but it seems everything is fine with it. Also I can't reproduce it:
[image]

mdevaev, 2019-03-03:
One of the differences with mjpg-streamer is that ustreamer always reads the data from the camera, regardless of whether there are any clients who are looking at the stream. The main reason is that the usteramer was designed to use with a separate broadcast control daemon (https://github.com/pi-kvm/kvmd, https://github.com/pi-kvm/kvmd/blob/master/kvmd/apps/kvmd/streamer.py). I assumed that if there were clients, this would launch the ustreamer; after disconnecting the last one, it would kill the process. Therefore, in an inactive state, my program consumes more cpu time. However, if there are clients, the result should not differ significantly from mjpg-streamer.

mdevaev, 2019-03-03:
Show me `ltrace -c ./ustreamer ...` please

mdevaev, 2019-03-10:
@chron0 Is the problem with CPU still relevant?

chron0, 2019-03-10:
sorry... busy week - well, it depends, it clearly consumes more CPU than mjpg-streamer. For me, in this setup it's no problem at all, I've got plenty of resources on this machine. It would be great to be lean on all arch's but I guess it's fine if it performs well on small-scale/embedded devices.

```-- INFO  [728598.600 tid=31549] -- Installing SIGINT handler ...
-- INFO  [728598.601 tid=31549] -- Installing SIGTERM handler ...
-- INFO  [728598.602 tid=31549] -- Ignoring SIGPIPE ...
-- INFO  [728598.604 tid=31549] -- Using JPEG quality: 100%
-- INFO  [728598.606 tid=31549] -- Listening HTTP on [0.0.0.0]:18000
-- INFO  [728598.611 tid=31550] -- Using V4L2 device: /dev/video0
-- INFO  [728598.614 tid=31550] -- Using desired FPS: 30
-- INFO  [728598.616 tid=31551] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [728598.645 tid=31550] -- Device fd=8 opened
-- INFO  [728598.649 tid=31550] -- Using input channel: 0
-- INFO  [728598.652 tid=31550] -- Using TV standard: DEFAULT
-- INFO  [728598.658 tid=31550] -- Using resolution: 1920x1080
-- INFO  [728598.660 tid=31550] -- Using pixelformat: MJPG
-- INFO  [728598.687 tid=31550] -- Requested 17 HW buffers, got 17
-- INFO  [728598.794 tid=31550] -- Capturing started
-- INFO  [728598.798 tid=31550] -- Switching to HW encoder because the input format is (M)JPEG
-- ERROR [728598.802 tid=31550] -- Can't query HW JPEG compressor params and set quality (unsupported)
-- INFO  [728598.807 tid=31550] -- Spawning 1 workers ...
-- INFO  [728598.812 tid=31550] -- Capturing ...
enable_breakpoint pid=31552, addr=0x555dce96a500, symbol=round: No such process
% time     seconds  usecs/call     calls      function
------ ----------- ----------- --------- --------------------
  7.61    0.000923         115         8 calloc
  7.39    0.000896          89        10 getopt_long
  6.64    0.000805         115         7 clock_gettime
  6.62    0.000803         803         1 event_base_new
  5.91    0.000717         143         5 __printf_chk
  5.89    0.000714         357         2 pthread_create
  5.62    0.000682         682         1 evhttp_bind_socket
  5.45    0.000661          94         7 round
  4.61    0.000559         111         5 syscall
  4.30    0.000521          86         6 strtol
  4.29    0.000520          86         6 __errno_location
  4.20    0.000510         127         4 evhttp_set_cb
  4.10    0.000497          99         5 fflush
  4.02    0.000488          97         5 pthread_mutex_unlock
  3.77    0.000457          91         5 pthread_mutex_lock
  2.85    0.000346          86         4 strcasecmp
  2.67    0.000324         108         3 pthread_mutex_init
  2.23    0.000270         135         2 sigaction
  1.65    0.000200         200         1 sysconf
  1.37    0.000166          83         2 sigaddset
  1.12    0.000136         136         1 signal
  1.05    0.000127         127         1 realloc
  0.96    0.000116         116         1 event_add
  0.96    0.000116         116         1 evthread_use_pthreads
  0.88    0.000107         107         1 evhttp_new
  0.84    0.000102         102         1 memcpy
  0.84    0.000102         102         1 evhttp_set_allowed_methods
  0.80    0.000097          97         1 evhttp_set_timeout
  0.78    0.000095          95         1 event_new
  0.60    0.000073          73         1 sigemptyset
------ ----------- ----------- --------- --------------------
100.00    0.012130                    99 total
```

I think it's always worthwhile to investigate room for optimization :)

mdevaev, 2019-03-11:
I need more statistics. Let the ltrace work for about 30 seconds.

Optimization is a necessary and good thing, although the ustreamer was primarily designed to fully utilize the available resources for maximum image quality and fps. I use it on raspberry pi to encoding and streaming fullhd video with 24 fps.

chron0, 2019-03-11:
I din't stop it, it stopped automatically once it started the worker. I was surprised to see it finish so quickly but I haven't used ltrace thus far. Maybe we need additional parameters to keep running and tracing the worker?

mdevaev, 2019-03-11:
What kernel version you use?

chron0, 2019-03-11:
```
Linux phoenix 4.18.7-rt-rt5 #2 SMP PREEMPT RT Tue Oct 30 21:05:43 UTC 2018 x86_64 Intel(R) Xeon(R) CPU E5640 @ 2.67GHz GenuineIntel GNU/Linux
```

mdevaev, 2019-03-12:
That's strange. Could you use perf and attach perf.data?
```
perf record ./ustreamer
```

chron0, 2019-03-12:
[perf.data.gz](https://github.com/pi-kvm/ustreamer/files/2959074/perf.data.gz)

mdevaev, 2019-03-14:
The situation has not become clearer, but I have one hypothesis. Do you have another camera? Not so cool :) Will the problem be reproduced with another camera?

chron0, 2019-03-15:
hmm, I don't think my other webcams do mjpg, only YUV - I do have one of those HDMI/USB UVC adapters I haven't tried yet - I could rig my big cam to it and try to capture from there, maybe it will do mjpg, let me see if I can rig another test setup up.

mdevaev, 2019-03-15:
Usually video capture devices don't know MJPG but I would be interested to see if I am right. In contrast, almost all even the oldest cameras supports MJPG.

Also I started to do an option that reduces the CPU usage if there are no clients looking to stream.

mdevaev, 2019-03-15:
Try to use fresh version from git with option `--slowdown`. It will reduce cpu consumption when no clients connected.

mdevaev, 2019-04-13:
@chron0 could you try the latest version?

chron0, 2019-04-13:
**BUILD**

```
>>> Emerging (1 of 1) media-video/ustreamer-9999::local-overlay
>>> Unpacking source...
 * Repository id: pi-kvm_ustreamer.git
 * To override fetched repository properties, use:
 *   EGIT_OVERRIDE_REPO_PI_KVM_USTREAMER
 *   EGIT_OVERRIDE_BRANCH_PI_KVM_USTREAMER
 *   EGIT_OVERRIDE_COMMIT_PI_KVM_USTREAMER
 *   EGIT_OVERRIDE_COMMIT_DATE_PI_KVM_USTREAMER
 * 
 * Fetching https://github.com/pi-kvm/ustreamer.git ...
git fetch https://github.com/pi-kvm/ustreamer.git +HEAD:refs/git-r3/HEAD
remote: Enumerating objects: 239, done.
remote: Counting objects: 100% (239/239), done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 567 (delta 202), reused 218 (delta 188), pack-reused 328
Receiving objects: 100% (567/567), 103.69 KiB | 512.00 KiB/s, done.
Resolving deltas: 100% (392/392), completed with 31 local objects.
From https://github.com/pi-kvm/ustreamer
   869d127..2077d94             -> refs/git-r3/HEAD
 * [new tag]         v0.68      -> v0.68
 * [new tag]         v0.53      -> v0.53
 * [new tag]         v0.54      -> v0.54
 * [new tag]         v0.55      -> v0.55
 * [new tag]         v0.56      -> v0.56
 * [new tag]         v0.57      -> v0.57
 * [new tag]         v0.58      -> v0.58
 * [new tag]         v0.59      -> v0.59
 * [new tag]         v0.60      -> v0.60
 * [new tag]         v0.61      -> v0.61
 * [new tag]         v0.62      -> v0.62
 * [new tag]         v0.63      -> v0.63
 * [new tag]         v0.64      -> v0.64
 * [new tag]         v0.65      -> v0.65
 * [new tag]         v0.66      -> v0.66
 * [new tag]         v0.67      -> v0.67
git symbolic-ref refs/git-r3/media-video/ustreamer/0/__main__ refs/git-r3/HEAD
 * Checking out https://github.com/pi-kvm/ustreamer.git to /var/tmp/portage/media-video/ustreamer-9999/work/ustreamer-9999 ...
git checkout --quiet refs/git-r3/HEAD
GIT update -->
   repository:               https://github.com/pi-kvm/ustreamer.git
   updating from commit:     869d12759c4d4e1ccf127db545d2e79c660991b4
   to commit:                2077d94edce24e7d524b94c81c417c763684ef5f
 .bumpversion.cfg                      |   8 ++++--
 .gitignore                            |   9 +++---
 Makefile                              |  25 +++++++++--------
 README.md                             |   8 +++---
 README.ru.md                          |  10 +++----
 PKGBUILD => pkg/arch/PKGBUILD         |   4 +--
 pkg/gentoo/ustreamer-9999.ebuild      |  27 ++++++++++++++++++
 pkg/openwrt/Makefile                  |  45 ++++++++++++++++++++++++++++++
 pkg/openwrt/files/ustreamer.config    |  20 ++++++++++++++
 pkg/openwrt/files/ustreamer.init      |  55 +++++++++++++++++++++++++++++++++++++
 src/config.h                          |   2 +-
 src/device.c                          | 285 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++----------------------------
 src/device.h                          |  45 +++++++++++++++++++++++-------
 src/encoder.c                         |  69 +++++++++++++++++++++++-----------------------
 src/encoder.h                         |  14 +++++-----
 src/{jpeg => encoders/cpu}/encoder.c  | 163 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-------------------------------------------
 src/encoders/cpu/encoder.h            |  28 +++++++++++++++++++
 src/encoders/hw/encoder.c             | 116 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 src/{ => encoders}/hw/encoder.h       |   2 +-
 src/encoders/hw/huffman.h             |  66 ++++++++++++++++++++++++++++++++++++++++++++
 src/{ => encoders}/omx/component.c    |   2 +-
 src/{ => encoders}/omx/component.h    |   0
 src/{ => encoders}/omx/encoder.c      | 113 +++++++++++++++++++++++++++++++++++++++++++++-------------------------------
 src/{ => encoders}/omx/encoder.h      |   7 +++--
 src/{ => encoders}/omx/formatters.c   |   2 +-
[... 67 lines cut ...]
-- INFO  [511796.641 tid=20039] -- Using desired FPS: 30
================================================================================
-- INFO  [511796.641 tid=20040] -- Starting HTTP eventloop ...
-- INFO  [511796.641 tid=20039] -- Device fd=8 opened
-- INFO  [511796.641 tid=20039] -- Using input channel: 0
-- INFO  [511796.642 tid=20039] -- Using TV standard: DEFAULT
-- INFO  [511796.642 tid=20039] -- Using resolution: 1920x1080
-- INFO  [511796.642 tid=20039] -- Using pixelformat: JPEG
-- INFO  [511796.642 tid=20039] -- Using HW FPS: 30
-- INFO  [511796.648 tid=20039] -- Requested 5 HW buffers, got 5
-- INFO  [511796.726 tid=20039] -- Capturing started
-- INFO  [511796.726 tid=20039] -- Switching to HW JPEG encoder because the input format is (M)JPEG
-- ERROR [511796.726 tid=20039] -- Can't query HW JPEG encoder params and set quality (unsupported)
-- INFO  [511796.726 tid=20039] -- Using JPEG quality: HW-default
-- INFO  [511796.726 tid=20039] -- Spawning 1 workers ...
-- INFO  [511796.726 tid=20039] -- Capturing ...
-- INFO  [511866.360 tid=20040] -- HTTP: Registered the new stream client: [127.0.0.1]:43024; id=54e3a06b-fe1b-4d01-841c-1c72bd788b59; clients now: 1
```
**LOAD**

[image]

It's even less load when idle (highest thread tops at 2.2%). I think I'll redo the setup and use a little gl.inet  with an openwrt and attach it directly to the cam to get rid of the USB connection between the cam and my client. That'll give me an opportunity to test it on small non-x86.

This is a great improvement. Thank you for your time and effort you put into making and improving it.

mdevaev, 2019-04-13:
Thanks for testing! I think this issue can be considered solved. If you have more suggestions or requests for improving the program - I will be glad to hear them :)

chron0, 2019-04-13:
See #2 :)

closed 2019-04-13 by chron0

mdevaev, 2019-04-13:
Also use option `-l / --slowdown`, it should further reduce CPU usage in idle mode without clients.
PS: oh I see, ok.

chron0, 2019-04-13:
[image]

confirmed!

mdevaev, 2019-04-13:
Nice)

chron0, 2019-04-13:
very! :)