pikvm/ustreamer #4: Feature Suggestion: Stream Fallback Source

chron0, 2019-04-13:
I'll just throw it in here for discussion: Wouldn't it be a good benefit to have a fallback option for the streamer, if the USB device is disconnected or doesn't stream data anymore, to have an option to detect source disconnection/stream-stall and stream either a JPG as fallback "source" or loop an mjpg video instead...

mdevaev, 2019-04-13:
You can try to pull out the camera wire during a stream right now and see image with "NO SIGNAL". The broadcast will resume if the wire is connected again. Is this what you want?

chron0, 2019-04-13:
Only half of it. This is already better than mjpg-streamer but I'd imagine a --fallback-jpg or --fallback-mjpg option, where we can give the path of a jpg file to show instead of the "NO SIGNAL" or the path to a video file (mjpg to keep things simple) that would be looped instead.

mdevaev, 2019-04-13:
I can make the `--fallback-jpg` option, but not `--fallback-mjpg`, because in fact the MJPG is not a pure video format and it cannot exist as file outside the HTTP protocol. This is a sequence of frames, each of which is an independent JPEG file. The server determines all broadcast parameters, for example, the delay between frames.

chron0, 2019-04-13:
MJPG can very well be a video format outside of HTTP, see https://www.redsharknews.com/production/item/3209-why-would-canon-use-mjpeg-for-video-in-the-new-1-dx-mk-ii

mdevaev, 2019-04-13:
Oookay that's interesting. Sounds like non-standard format. https://en.wikipedia.org/wiki/Motion_JPEG#Disadvantages

> Unlike the video formats specified in international standards such as MPEG-2 and the format specified in the JPEG still-picture coding standard, there is no document that defines a single exact format that is universally recognized as a complete specification of “Motion JPEG” for use in all contexts. This raises compatibility concerns about file outputs from different manufacturers. However, each particular file format usually has some standard on how M-JPEG is encoded. For example, Microsoft documents their standard format to store M-JPEG in AVI files,[6] Apple documents how M-JPEG is stored in QuickTime files, RFC 2435 describes how M-JPEG is implemented in an RTP stream, and an M-JPEG CodecID is planned for the Matroska file format.

chron0, 2019-04-13:
ye, currently it's mostly wrapped in mov files.

mdevaev, 2019-04-13:
Ok, I'll do the first option, but I'm not sure about the second one. It will complicate the already complex server too much. At least as I imagine it now.

mdevaev, 2019-04-14:
Try it. Option `-k/--blank <path/to/file.jpg>`. I made it without handling some errors, but should work quite well.

mdevaev, 2019-04-17:
@chron0 I added error handling, now this feature is done.

closed 2019-04-30 by mdevaev