pikvm/ustreamer #5: Error compiling on Raspbian Raspbian GNU/Linux 8 (jessie)

dandfra, 2019-05-17:
in the linking phase it says that the symbol libevent_global_shutdown is missing (in the compilethere is a warning).
I workarounded by commenting out (it's used to cleanup before destroy, but is not really neded)

mdevaev, 2019-05-17:
Which version of libevent you used?

dandfra, 2019-05-17:
r $ apt list --installed | grep libevent

WARNING: apt does not have a stable CLI interface yet. Use with caution in
scripts.

libevent-2.0-5/oldstable,now 2.0.21-stable-2+deb8u1 armhf
[installed,automatic]
libevent-core-2.0-5/oldstable,now 2.0.21-stable-2+deb8u1 armhf
[installed,automatic]
libevent-dev/oldstable,now 2.0.21-stable-2+deb8u1 armhf [installed]
libevent-extra-2.0-5/oldstable,now 2.0.21-stable-2+deb8u1 armhf
[installed,automatic]
libevent-openssl-2.0-5/oldstable,now 2.0.21-stable-2+deb8u1 armhf
[installed,automatic]
libevent-pthreads-2.0-5/oldstable,now 2.0.21-stable-2+deb8u1 armhf
[installed]

mdevaev, 2019-05-17:
Gotcha. ```libevent_global_shutdown()``` was implemented in version 2.1.1. I've added additional check for this. Please try it from master. Also I recommend to use a newer version of libevent because it's being actively developed and contains many bugfixes.

dandfra, 2019-05-17:
Ok, I confirm that now it compiles, thanks a lot.
FYI libevent 2.0 is the version in both jesse and stretch (https://packages.debian.org/search?keywords=libevent).

closed 2019-05-17 by dandfra

mdevaev, 2019-05-17:
Cool. Thanks for the issue! I'll make a new ustreamer release.