pikvm/ustreamer #6: Installs an unstripped eewcutable

yurivict, 2019-05-18:
```
====> Running Q/A tests (stage-qa)
Warning: 'bin/ustreamer' is not stripped consider trying INSTALL_TARGET=install-strip or using ${STRIP_CMD}
```

Please strip it!

mdevaev, 2019-05-18:
Umm, what is that? It's a FreeBSD package?

yurivict, 2019-05-18:
yes

mdevaev, 2019-05-18:
I've added make targets `install-strip` (and `uninstall` also but who cares) following this doc: https://www.gnu.org/prep/standards/html_node/Standard-Targets.html
That's what you need?

mdevaev, 2019-05-18:
Wait wat? Ustreamer working under FreeBSD? But how? I thought it's not portable because I used Linux-specific APIs.

yurivict, 2019-05-18:
> UPD: This is it? https://www.freshports.org/multimedia/ustreamer

Yes, it build on FreeBSD.

> I can add this link to readme.

Please do. Thank you!

mdevaev, 2019-05-18:
Done! Bumped to 0.75

closed 2019-05-18 by mdevaev