pikvm/ustreamer #7: Stream crashing

lbr38, 2019-09-06:
Hi,

I had the stream working for all day, and tonight it crashed. I tried to stop ans start it again and now it still crash (loop) : 

```
================================================================================
-- INFO  [124679.232 tid=1019] -- Device fd=8 opened
-- INFO  [124679.232 tid=1019] -- Using input channel: 0
-- INFO  [124679.232 tid=1019] -- Using TV standard: DEFAULT
-- INFO  [124679.291 tid=1019] -- Using resolution: 1280x720
-- INFO  [124679.291 tid=1019] -- Using pixelformat: JPEG
-- INFO  [124679.348 tid=1019] -- Using HW FPS: 30 -> 60 (coerced)
-- INFO  [124679.358 tid=1019] -- Requested 5 HW buffers, got 5
-- INFO  [124679.395 tid=1019] -- Capturing started
-- INFO  [124679.395 tid=1019] -- Switching to HW encoder because the input format is (M)JPEG
-- ERROR [124679.395 tid=1019] -- Can't query HW encoder params and set quality (unsupported)
-- INFO  [124679.395 tid=1019] -- Using JPEG quality: encoder default
-- INFO  [124679.395 tid=1019] -- Creating pool with 1 workers ...
-- INFO  [124679.396 tid=1019] -- Capturing ...
-- ERROR [124680.397 tid=1019] -- Mainloop select() timeout
-- INFO  [124680.397 tid=1019] -- Destroying workers pool ...
-- INFO  [124680.409 tid=1019] -- Capturing stopped
-- INFO  [124680.415 tid=1019] -- Device fd=8 closed
```
Also some logs from kern.log : 
```
Sep  6 20:46:52 pi1 kernel: [124048.471629] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:46:53 pi1 kernel: [124049.472968] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:46:54 pi1 kernel: [124050.474253] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:46:55 pi1 kernel: [124051.475526] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:46:56 pi1 kernel: [124052.476797] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:46:57 pi1 kernel: [124053.478214] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:46:58 pi1 kernel: [124054.479344] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:46:59 pi1 kernel: [124055.480746] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:47:00 pi1 kernel: [124056.482118] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
Sep  6 20:47:01 pi1 kernel: [124057.483476] uvcvideo: Failed to set UVC probe control : -71 (exp. 26).
```

The command I used : 
`ustreamer --device=/dev/video0  -r 1280x720 -f 30 -m JPEG --host 0.0.0.0 --port 8888 2>&1 | tee /home/pi/scripts/ustreamer-live.log &`

I'm pretty sure it will work again if I reboot the RPi but have you got something to "reset" the camera without rebooting ?

mdevaev, 2019-09-06:
I think you can remove uvcvideo and laod it again: `rmmod uvcvideo && modprobe uvcvideo`.

Also try to append `dwc_otg.fiq_fsm_mask=0x3` to `/boot/cmdline.txt` and reboot. I think it will have to help.

mdevaev, 2019-09-06:
When you restart, after the timeout did the program exit?

lbr38, 2019-09-08:
Okay so I had the same crash today, here are the logs before killing the program. Those logs are looping and the black No signal page is diplaying on the stream:

```
================================================================================
-- INFO  [143879.739 tid=658] -- Device fd=9 opened
-- INFO  [143879.740 tid=658] -- Using input channel: 0
-- INFO  [143879.740 tid=658] -- Using TV standard: DEFAULT
-- ERROR [143879.740 tid=658] -- Unable to set pixelformat=JPEG, resolution=1280x720: Input/output error
-- INFO  [143879.740 tid=658] -- Device fd=9 closed
-- INFO  [143879.740 tid=658] -- Sleeping 1 seconds before new stream init ...
================================================================================
-- INFO  [143880.741 tid=658] -- Device fd=9 opened
-- INFO  [143880.741 tid=658] -- Using input channel: 0
-- INFO  [143880.741 tid=658] -- Using TV standard: DEFAULT
-- ERROR [143880.741 tid=658] -- Unable to set pixelformat=JPEG, resolution=1280x720: Input/output error
-- INFO  [143880.741 tid=658] -- Device fd=9 closed
-- INFO  [143880.741 tid=658] -- Sleeping 1 seconds before new stream init ...
================================================================================
-- INFO  [143881.742 tid=658] -- Device fd=9 opened
-- INFO  [143881.742 tid=658] -- Using input channel: 0
-- INFO  [143881.742 tid=658] -- Using TV standard: DEFAULT
-- ERROR [143881.742 tid=658] -- Unable to set pixelformat=JPEG, resolution=1280x720: Input/output error
-- INFO  [143881.743 tid=658] -- Device fd=9 closed
-- INFO  [143881.743 tid=658] -- Sleeping 1 seconds before new stream init ...
================================================================================
...
Continue looping until I kill the program
```
So I tried to:
* stop the program 
* rmmod uvcvideo && modprobe uvcvideo as you suggested 
* then start the program, but still crash and loop with the 'select timeout' logs:

```
================================================================================
-- INFO  [144412.221 tid=7311] -- Device fd=8 opened
-- INFO  [144412.221 tid=7311] -- Using input channel: 0
-- INFO  [144412.221 tid=7311] -- Using TV standard: DEFAULT
-- INFO  [144412.279 tid=7311] -- Using resolution: 1280x720
-- INFO  [144412.279 tid=7311] -- Using pixelformat: JPEG
-- INFO  [144412.337 tid=7311] -- Using HW FPS: 30 -> 60 (coerced)
-- INFO  [144412.347 tid=7311] -- Requested 5 HW buffers, got 5
-- INFO  [144412.383 tid=7311] -- Capturing started
-- INFO  [144412.383 tid=7311] -- Switching to HW encoder because the input format is (M)JPEG
-- ERROR [144412.383 tid=7311] -- Can't query HW encoder params and set quality (unsupported)
-- INFO  [144412.384 tid=7311] -- Using JPEG quality: encoder default
-- INFO  [144412.384 tid=7311] -- Creating pool with 1 workers ...
-- INFO  [144412.384 tid=7311] -- Capturing ...
-- ERROR [144413.385 tid=7311] -- Mainloop select() timeout
-- INFO  [144413.385 tid=7311] -- Destroying workers pool ...
-- INFO  [144413.398 tid=7311] -- Capturing stopped
-- INFO  [144413.403 tid=7311] -- Device fd=8 closed
================================================================================
-- INFO  [144413.404 tid=7311] -- Device fd=8 opened
-- INFO  [144413.404 tid=7311] -- Using input channel: 0
-- INFO  [144413.404 tid=7311] -- Using TV standard: DEFAULT
-- INFO  [144413.461 tid=7311] -- Using resolution: 1280x720
-- INFO  [144413.461 tid=7311] -- Using pixelformat: JPEG
-- INFO  [144413.519 tid=7311] -- Using HW FPS: 30 -> 60 (coerced)
-- INFO  [144413.529 tid=7311] -- Requested 5 HW buffers, got 5
-- INFO  [144413.566 tid=7311] -- Capturing started
-- INFO  [144413.566 tid=7311] -- Switching to HW encoder because the input format is (M)JPEG
-- ERROR [144413.566 tid=7311] -- Can't query HW encoder params and set quality (unsupported)
-- INFO  [144413.566 tid=7311] -- Using JPEG quality: encoder default
-- INFO  [144413.566 tid=7311] -- Creating pool with 1 workers ...
-- INFO  [144413.567 tid=7311] -- Capturing ...
-- INFO  [144414.289 tid=7309] -- ===== Stopping by SIGTERM =====
-- INFO  [144414.289 tid=7312] -- HTTP eventloop stopped
-- ERROR [144414.568 tid=7311] -- Mainloop select() timeout
-- INFO  [144414.568 tid=7311] -- Destroying workers pool ...
-- INFO  [144414.581 tid=7311] -- Capturing stopped
-- INFO  [144414.586 tid=7311] -- Device fd=8 closed
================================================================================
...
Continue looping until I kill the program
```

> When you restart, after the timeout did the program exit?

Nope it doesn't exit and try to restart the capture (logs above)

For the moment the only way to get the camera and the stream working again is to reboot the system.

> Also try to append `dwc_otg.fiq_fsm_mask=0x3` to `/boot/cmdline.txt` and reboot. I think it will have to help.
> 
> Related: [raspberrypi/linux#623](https://github.com/raspberrypi/linux/issues/623)

Ok I will test this ;-)

Thank you for your time and help, I don't know if there is a solution for those crashs maybe the camera is too buggy..

mdevaev, 2019-09-08:
Random crashes always mean a problem with hardware or drivers. Sometimes you can make a simple workarund in software for this, but sometimes there’s no easy solution.

lbr38, 2019-09-11:
Still haven't found a solution.. the stream works for 1 day then it crashes :(

Saw many threads on the internet, everybody seems to have this problem since years, and nobody ever found a real solution except rebooting the system.
I'll have to make a script that monitors the stream and reboot the system if the camera crashes. This is a bit frustrating to do it this way but there's no other way.

Anyway ustreamer works like a charm, you did a nice job! :)

mdevaev, 2019-09-11:
Thanks! :)

It looks like it's still in the driver. Unfortunately, I have no way to debug it. If you find out any solution in the future - please let me know about it.

closed 2019-09-11 by mdevaev

xmmwc, 2020-06-09:
I solve it by set --device-timeout to 2 sec
just wait longer

lbr38, 2020-06-11:
Indeed it seems to work fine with timeouts!

> -- ERROR [61756.456    stream] -- Mainloop select() timeout
> -- INFO  [61756.456    stream] -- Destroying workers pool ...
> -- INFO  [61756.472      http] -- HTTP: Freezing last alive frame forever
> -- INFO  [61756.586    stream] -- Capturing stopped
> -- INFO  [61756.603    stream] -- Device fd=8 closed

But will see if it works on long term and how it deals with "I/O errors" and "Can't open device" errors (which need to reboot the system to get the camera back in my case)

Thanks for the info and thanks @mdevaev for this new feature :)