pikvm/ustreamer #9: Feature Request: Alternate options for "NO SIGNAL" condition

dstulken, 2019-09-11:
Hello!
Would it be possible to add some alternate options for the "blank" JPG configuration?
A few ideas:

1)  Allow the internal blank "NO SIGNAL" image to be generated at a resolution matching the regular stream output... ie, the same as what was specified with the "-r" parameter.   Some stream recipients really struggle with resolution changes mid-stream, so having the "NO SIGNAL" page match in resolution would avoid tearing, buffer glitches, and other artifacts.   (This might actually be nice as the default option, unless there was a benefit to using 640x480 as the default?)

2) Add an option to hold (and re-send) the last valid image frame prior to the disconnect, instead of showing the "NO SIGNAL" image.  This might alternately have a "NO SIGNAL" text overlay, which may be advantageous to have be in small text.   
The use case for this mode, would be for systems where a downstream consumer is doing motion analysis on the stream.   In the case of a system which experiences frequent disconnects, such as the nuisance "select() timeout" condition, each camera disconnect and its full screen "NO SIGNAL" page will generate cascading nuisance motion triggers in the downstream system.   Holding and retransmitting the last valid image would allow these intermittent video streams to bridge the camera disconnect gaps without triggering motion actions in the downstream systems.   (And likewise, having a "NO SIGNAL" overlay in SMALL text on the frozen image may be sufficient to allow downstream viewers to be aware of this condition, yet not change so much of the video frame that it triggers a motion event.)

Thanks in advance

mdevaev, 2019-09-11:
I was thinking of something like that. I have never tried generating pictures using C. Perhaps it’s time to experiment :) However, I can’t promise that I'll do it quickly. First of all, I'll try to save the last valid image until the stream is turned off.

mdevaev, 2019-09-11:
Also in current implementation you can use `-k/--blank` to specify any jpeg file to be used as blank (with any resolution ofc).

mdevaev, 2019-09-17:
@dstulken I slightly changed this new option. If you specify `-K 0`, then the last frame will be displayed forever. If you specify `-K 10`, then no more than 10 seconds.

mdevaev, 2019-10-14:
I thought about how to better implement this feature and came to the conclusion that the settings for displaying the label on the screen that I can add will always be insufficient for some users, since I want to have a general and not a private solution. Label settings will require parameters such as font, size, screen layout, color, effects (you probably need some kind of outline or background so that the label does not merge with a bright or dark image). Then someone will want localization, and I will all turn into a nightmare :)

However, right now ustreamer allows you to implement what you want in combination with the option `-K <N>`. It will allow to smooth out insignificant short delays when the source is turned off (or to completely ignore it and broadcast the image forever). Along with this, you can write some javascript code in the browser, and use the information from the ustreamer's `/state` URL. It contains comprehensive information about what happens to the source: its resolution, status flags, and more. Then you can use the `<DIV>` located on top of the `<IMG>` to display the information you need. Even in my project, for which I created ustreamer, this approach is used. By the way, now I use your idea there and show the picture so as to avoid flickering :)

It seems to me that there is nothing more flexible than the CSS, and any attempts to give the software its limited functionality will always give rise to support problems.

Although there is still one case when you watch a video stream not in the browser, but with the player. Is it so in your case?

mdevaev, 2019-10-21:
I close this because I think that at the moment the current solution is best. In the event of new circumstances it's possible to consider other solutions and additional work on this task.

closed 2019-10-21 by mdevaev