pikvm/ustreamer #10: ov2710 error

DaTTcz, 2019-11-23:
Hello, i can't start the ov2710 camera. What am I doing wrong ? Thank you.

**lsusb**
`ID 0c45: 6368 Microdia`

**./ustreamer --device /dev/video0 -r 1920x1080 -f 30 -m JPEG**

```
- INFO  [45121.867      main] -- Installing SIGINT handler ...
-- INFO  [45121.867      main] -- Installing SIGTERM handler ...
-- INFO  [45121.867      main] -- Ignoring SIGPIPE ...
-- INFO  [45121.867      main] -- Using internal blank placeholder
-- INFO  [45121.868      main] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [45121.868    stream] -- Using V4L2 device: /dev/video0
-- INFO  [45121.868    stream] -- Using desired FPS: 30
================================================================================
-- INFO  [45121.868      http] -- Starting HTTP eventloop ...
-- INFO  [45121.955    stream] -- Device fd=8 opened
-- INFO  [45121.955    stream] -- Using input channel: 0
-- INFO  [45121.955    stream] -- Using TV standard: DEFAULT
-- INFO  [45122.151    stream] -- Using resolution: 1920x1080
-- INFO  [45122.151    stream] -- Using pixelformat: JPEG
-- INFO  [45122.175    stream] -- Using HW FPS: 30 -> 10000000 (coerced)
-- INFO  [45122.235    stream] -- Requested 3 HW buffers, got 3
-- ERROR [45122.236    stream] -- Can't map device buffer 0: No such device
-- INFO  [45122.381    stream] -- Device fd=8 closed
-- INFO  [45122.381    stream] -- Sleeping 1 seconds before new stream init ...
-- INFO  [45123.206      main] -- ===== Stopping by SIGINT =====
-- INFO  [45123.206      http] -- HTTP eventloop stopped
-- INFO  [45123.382      main] -- Bye-bye

```
**v4l2-ctl -list-formats-ext**

```
ioctl: VIDIOC_ENUM_FMT
        Index: 0
        Type: Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name: MJPEG
                Size: Discrete 1920x1080
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 432x240
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 320x184
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 160x120mm
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (31.000 fps)
                Size: 1280x960
                        Interval: Discrete 0.03s (30.000 fps)

        Index: 1
        Type: Video Capture
        Pixel Format: 'YUYV'
        Name: YUV 4: 2: 2
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5,000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 432x240
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 320x184
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 160x120mm
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.045s (21.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.093s (11.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.075s (13.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.113s (9.000 fps)
                Size: 1280x960
                        Interval: Discrete 0.167s (6.000 fps)
```

mdevaev, 2019-11-23:
Does the camera work with other software?
Try using the YUYV format.
And show me the log of this command: `mplayer -v -tv driver=v4l2:device=/dev/video0:outfmt=mjpg tv://`

DaTTcz, 2019-11-23:
I tested the camera under windows10 and it works. Camera is OK.
i tried mjpg-streamer-experimental under linux server and it doesn't work too.

**./mjpg_streamer -i "./input_uvc.so -r 1920x1080 -f 30" -o "./output_http.so"**
```
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1920 x 1080
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 30 to 10000000
Unable to map buffer: No such device
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```
**mplayer -v -tv driver=v4l2:device=/dev/video0:outfmt=mjpg tv://**
```
MPlayer 1.3.0 (Debian), built with gcc-7 (C) 2000-2016 MPlayer Team
CPU vendor name: GenuineIntel  max cpuid level: 11
CPU: Intel(R) Celeron(R) CPU  N2830  @ 2.16GHz (Family: 6, Model: 55, Stepping: 8)
extended cpuid-level: 8
extended cache-info: 67141696
Detected cache-line size is 64 bytes
CPUflags:  MMX: 1 MMX2: 1 3DNow: 0 3DNowExt: 0 SSE: 1 SSE2: 1 SSE3: 1 SSSE3: 1 SSE4: 1 SSE4.2: 1 AVX: 0
Compiled with runtime CPU detection.
get_path('codecs.conf') -> '/home/david/.mplayer/codecs.conf'
Reading optional codecs config file /home/david/.mplayer/codecs.conf: No such file or directory
Reading optional codecs config file /etc/mplayer/codecs.conf: No such file or directory
Using built-in default codecs.conf.
init_freetype
Using MMX (with tiny bit MMX2) Optimized OnScreenDisplay
get_path('fonts') -> '/home/david/.mplayer/fonts'
Configuration: --prefix=/usr --confdir=/etc/mplayer --enable-debug --enable-xvmc --enable-menu --disable-arts --language=all --disable-libmpeg2-internal --disable-ffmpeg_a --disable-esd --extra-cflags=-g -O2 -fdebug-prefix-map=/build/mplayer-brS2lj/mplayer-1.3.0=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 --extra-ldflags=-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now -Wl,--as-needed --enable-runtime-cpudetection --enable-joystick --enable-radio --enable-radio-capture --disable-gui
CommandLine: '-v' '-tv' 'driver=v4l2:device=/dev/video0:outfmt=mjpg' 'tv://'
Using nanosleep() timing
get_path('input.conf') -> '/home/david/.mplayer/input.conf'
Reading optional input config file /home/david/.mplayer/input.conf: No such file or directory
Parsing input config file /etc/mplayer/input.conf
Input config file /etc/mplayer/input.conf parsed: 92 binds
Setting up LIRC support...
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.
get_path('.conf') -> '/home/david/.mplayer/.conf'

Playing tv://.
get_path('sub/') -> '/home/david/.mplayer/sub/'
STREAM: [tv] tv://
STREAM: Description: TV Input
STREAM: Author: Benjamin Zores, Albeu
STREAM: Comment:
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM was used instead.
Selected device: USB Camera
 Capabilities:  video capture  read/write  streaming
 supported norms:
 inputs: 0 = Camera;
 Current input: 0
 Format MJPEG  ( 0 bits, MJPEG): Unknown 0x47504a4d
 Format YUYV   (16 bits, YUV 4:2:2 (YUYV)): Packed YUY2
 Current format: MJPEG
==================================================================
 WARNING: UNTESTED OR UNKNOWN OUTPUT IMAGE FORMAT REQUESTED (0x47504a4d)
 This may cause buggy playback or program crash! Bug reports will
 be ignored! You should try again with YV12 (which is the default
 colorspace) and read the documentation!
==================================================================
v4l2: set format: MJPEG
v4l2: set input: 0
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
Selected norm : pal
v4l2: ioctl enum norm failed: Invalid argument
Error: Cannot set norm!
Selected input hasn't got a tuner!
==> Found video stream: 0
v4l2: get format: MJPEG
v4l2: get fps: 30.000031
v4l2: get width: 640
v4l2: get height: 480
Using a ring buffer for maximum 2 frames, 0 MB total size.
v4l2: mmap capture buffer failed: No such device
v4l2: ioctl set mute failed: Invalid argument
v4l2: 0 frames successfully processed, 0 frames dropped.
v4l2: up to 0 video frames buffered.

vo: x11 uninit called but X11 not initialized..

Exiting... (End of file)
```

mdevaev, 2019-11-23:
Where did you find the driver for this camera? It's not supported by the default kernel.

DaTTcz, 2019-11-23:
I did not install any driver. I thought it was part of the kernel. That's probably the problem... But I haven't found it anywhere...

mdevaev, 2019-11-23:
Show me dmesg plz.

mdevaev, 2019-11-23:
... And lsmod

DaTTcz, 2019-11-23:
**dmesg**
```
[  173.554586] usb 2-3: USB disconnect, device number 3
[  176.738548] usb 2-3: new high-speed USB device number 5 using xhci_hcd
[  176.993654] usb 2-3: New USB device found, idVendor=0c45, idProduct=6368
[  176.993671] usb 2-3: New USB device strings: Mfr=2, Product=1, SerialNumber=0
[  176.993680] usb 2-3: Product: USB 2.0 Camera
[  176.993690] usb 2-3: Manufacturer: Sonix Technology Co., Ltd.
```
**lsmod**
```
Module                  Size  Used by
ccm                    20480  9
nls_iso8859_1          16384  1
intel_rapl             20480  0
arc4                   16384  2
intel_soc_dts_thermal    16384  0
intel_soc_dts_iosf     16384  1 intel_soc_dts_thermal
intel_powerclamp       16384  0
coretemp               16384  0
snd_hda_codec_hdmi     49152  1
kvm_intel             217088  0
snd_hda_codec_realtek   106496  1
snd_hda_codec_generic    73728  1 snd_hda_codec_realtek
kvm                   610304  1 kvm_intel
iwlmvm                368640  0
irqbypass              16384  1 kvm
snd_hda_intel          45056  0
punit_atom_debug       16384  0
mac80211              786432  1 iwlmvm
intel_cstate           20480  0
snd_hda_codec         126976  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
iwlwifi               290816  1 iwlmvm
btusb                  45056  0
btrtl                  16384  1 btusb
snd_hda_core           81920  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
btbcm                  16384  1 btusb
btintel                16384  1 btusb
snd_hwdep              20480  1 snd_hda_codec
snd_pcm                98304  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
bluetooth             544768  5 btrtl,btintel,btbcm,btusb
cfg80211              622592  3 iwlmvm,iwlwifi,mac80211
mei_txe                20480  0
snd_timer              32768  1 snd_pcm
cdc_acm                32768  0
ecdh_generic           24576  1 bluetooth
mei                    94208  1 mei_txe
snd                    81920  8 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm
soundcore              16384  1 snd
shpchp                 36864  0
lpc_ich                24576  0
ir_rc6_decoder         16384  0
rc_rc6_mce             16384  0
ir_lirc_codec          16384  0
lirc_dev               16384  2 ir_lirc_codec
ite_cir                28672  0
rc_core                36864  6 lirc_dev,ir_lirc_codec,ite_cir,ir_rc6_decoder,rc_rc6_mce
mac_hid                16384  0
intel_smartconnect     16384  0
sch_fq_codel           20480  5
ib_iser                49152  0
rdma_cm                61440  1 ib_iser
iw_cm                  45056  1 rdma_cm
ib_cm                  53248  1 rdma_cm
ib_core               225280  4 rdma_cm,iw_cm,ib_iser,ib_cm
iscsi_tcp              20480  0
libiscsi_tcp           20480  1 iscsi_tcp
libiscsi               53248  3 libiscsi_tcp,iscsi_tcp,ib_iser
scsi_transport_iscsi    98304  3 iscsi_tcp,ib_iser,libiscsi
cuse                   16384  1
ip_tables              28672  0
x_tables               40960  1 ip_tables
autofs4                40960  2
btrfs                1138688  0
zstd_compress         163840  1 btrfs
raid10                 53248  0
raid456               147456  0
async_raid6_recov      20480  1 raid456
async_memcpy           16384  2 raid456,async_raid6_recov
async_pq               16384  2 raid456,async_raid6_recov
async_xor              16384  3 async_pq,raid456,async_raid6_recov
async_tx               16384  5 async_pq,async_memcpy,async_xor,raid456,async_raid6_recov
xor                    24576  2 async_xor,btrfs
raid6_pq              114688  4 async_pq,btrfs,raid456,async_raid6_recov
libcrc32c              16384  1 raid456
raid1                  40960  0
raid0                  20480  0
multipath              16384  0
linear                 16384  0
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
i915                 1622016  1
ghash_clmulni_intel    16384  0
cryptd                 24576  1 ghash_clmulni_intel
i2c_algo_bit           16384  1 i915
drm_kms_helper        172032  1 i915
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
drm                   401408  3 drm_kms_helper,i915
pata_acpi              16384  0
video                  45056  1 i915

```
**v4l2-ctl -d /dev/video0 -D**
```
Driver Info (not using libv4l2):
        Driver name   : uvc
        Card type     : USB Camera
        Bus info      : usb-2-5
        Driver version: 4.4.98
        Capabilities  : 0x05000001
                Video Capture
                Read/Write
                Streaming

```

DaTTcz, 2019-11-23:
**v4l2-ctl --all**

```
Driver Info (not using libv4l2):
        Driver name   : uvc
        Card type     : USB Camera
        Bus info      : usb-2-8
        Driver version: 4.4.98
        Capabilities  : 0x05000001
                Video Capture
                Read/Write
                Streaming
Video input : 0 (Camera: ok)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'MJPG'
        Field             : None
        Bytes per Line    : 0
        Size Image        : 4147789
        Colorspace        : sRGB
        Transfer Function : Default (maps to sRGB)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (10000000/333333)
        Read buffers     : 4

```

mdevaev, 2019-11-24:
As far as I understand, Linux does not support this camera and tries to use the default driver (uvc). It seems that a special module is needed for this camera: https://github.com/allwinner-zh/linux-3.4-sunxi/blob/master/drivers/media/video/sunxi-vfe/device/ov2710.c
This driver is written for an old version of the kernel, for some fork. Perhaps it can be adapted for the new kernel, but unfortunately I can't help you with this.

DaTTcz, 2019-11-24:
I agree with you. I was looking for a small camera (usb module) with high resolution and fast FPS for linux. I don't need a microphone. Can you please recommend one? Thank you for your time.

mdevaev, 2019-11-24:
The only camera I can advise you is Logitech HD Webcam C930e. I use it and it works fine in Linux. I'm sorry, but I haven't encountered the camera modules you require.

I think you can write to the LKML mailing list with a message that the UVC driver doesn't work with your camera, and give a link to the module. Perhaps kernel hackers will either fix the module or advise you on how to make your camera work.

closed 2019-11-24 by mdevaev