pikvm/ustreamer #11: Build warnings: ormat specifies type 'unsigned long long' but the argument has type 'uint64_t', etc

yurivict, 2020-01-18:
clang-8 prints these:
```
src/device.c:393:31: warning: format specifies type 'unsigned long long' but the argument has type 'uint64_t' (aka 'unsigned long') [-Wformat]
                        dv.bt.width, dv.bt.height, dv.bt.pixelclock);
                                                   ^~~~~~~~~~~~~~~~
src/logging.h:121:48: note: expanded from macro 'LOG_INFO'
                LOG_PRINTF(COLOR_GREEN, "INFO ", "", _msg, ##__VA_ARGS__); \
                                                     ~~~~    ^~~~~~~~~~~
src/logging.h:106:63: note: expanded from macro 'LOG_PRINTF'
                LOG_PRINTF_NOLOCK(_label_color, _label, _msg_color, _msg, ##__VA_ARGS__); \
                                                                    ~~~~    ^~~~~~~~~~~
src/logging.h:95:34: note: expanded from macro 'LOG_PRINTF_NOLOCK'
                                get_now_monotonic(), _buf, ##__VA_ARGS__); \
                                                             ^~~~~~~~~~~
src/device.c:393:31: warning: format specifies type 'unsigned long long' but the argument has type 'uint64_t' (aka 'unsigned long') [-Wformat]
                        dv.bt.width, dv.bt.height, dv.bt.pixelclock);
                                                   ^~~~~~~~~~~~~~~~
src/logging.h:121:48: note: expanded from macro 'LOG_INFO'
                LOG_PRINTF(COLOR_GREEN, "INFO ", "", _msg, ##__VA_ARGS__); \
                                                     ~~~~    ^~~~~~~~~~~
src/logging.h:106:63: note: expanded from macro 'LOG_PRINTF'
                LOG_PRINTF_NOLOCK(_label_color, _label, _msg_color, _msg, ##__VA_ARGS__); \
                                                                    ~~~~    ^~~~~~~~~~~
src/logging.h:98:34: note: expanded from macro 'LOG_PRINTF_NOLOCK'
                                get_now_monotonic(), _buf, ##__VA_ARGS__); \
                                                             ^~~~~~~~~~~
2 warnings and 1 error generated.
```

http://beefy9.nyi.freebsd.org/data/113amd64-default/522982/logs/ustreamer-1.10.log

mdevaev, 2020-01-18:
Ok I'll check it.

PS: Use `make WITH_PTHREAD_NP=0` to build without pthread error

mdevaev, 2020-01-18:
Fixed warnings, please check it from master

yurivict, 2020-01-18:
This warning is gone, but there are still some:
```
In file included from <built-in>:348:
<command line>:3:9: warning: 'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]
#define WITH_PTHREAD_NP 1
        ^
<command line>:1:9: note: previous definition is here
#define WITH_PTHREAD_NP 0
        ^
cc: warning: -lm: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -ljpeg: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -levent: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -levent_pthreads: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -luuid: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -lm: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -ljpeg: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -levent: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -levent_pthreads: 'linker' input unused [-Wunused-command-line-argument]
cc: warning: -luuid: 'linker' input unused [-Wunused-command-line-argument]
```

mdevaev, 2020-01-19:
Fixed `'linker' input unused`. Can't reproduce first error on clang 9. Is it specific for clang 8?

yurivict, 2020-01-19:
linker flags shouldn't be supplied to compiler, and compiler flags shouldn't be supplied to linker. You give ex. ```-ljpeg``` to compiler which is wrong. This shouldn't be specific to compiler versions.

mdevaev, 2020-01-19:
> `-ljpeg` to compiler which is wrong. This shouldn't be specific to compiler versions.

Yes. I already fixed. I mean I can't reproduce this error:

```
In file included from <built-in>:348:
<command line>:3:9: warning: 'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]
#define WITH_PTHREAD_NP 1
        ^
<command line>:1:9: note: previous definition is here
#define WITH_PTHREAD_NP 0
```

yurivict, 2020-01-19:
You have a bug. On one hand you recommend to add ```WITH_PTHREAD_NP=0```, implying that ```WITH_PTHREAD_NP=1``` is the opposite, on the other hand you always check ```#ifdef WITH_PTHREAD_NP``` which catches both 0 and 1.

You need to use ```#if WITH_PTHREAD_NP``` everywhere.

mdevaev, 2020-01-19:
I define `-DWITH_PTHREAD_NP` only if the option `WITH_PTHREAD_NP=0` is not specified. Otherwise `WITH_PTHREAD_NP` is not defined. If there was a bug, you wouldn't be able to complete the build at all.

yurivict, 2020-01-19:
When the user sets ```-DWITH_PTHREAD_NP=0``` and ```-DWITH_PTHREAD_NP=1``` he means two different things, but your check ```#ifdef WITH_PTHREAD_NP``` catches them both.

yurivict, 2020-01-19:
I am just saying that the code should be very simple and readable. You should either have -DXXX and ```#if defined()``` or -DXXX=1 and ```#if XXX```, not both.

mdevaev, 2020-01-19:
If don't set `-DWITH_PTHREAD_NP=0` and `-DWITH_PTHREAD_NP=1` anywere. Look:
```
$ make CC=clang WITH_PTHREAD_NP=0
clang src/device.c -o build/src/device.o -O3 -c -std=c11 -Wall -Wextra -D_GNU_SOURCE -DWITH_SETPROCTITLE
```
```
$ make CC=clang
clang src/device.c -o build/src/device.o -O3 -c -std=c11 -Wall -Wextra -D_GNU_SOURCE -DWITH_PTHREAD_NP -DWITH_SETPROCTITLE
```
So I use only `#ifdef` or `#if defined`.

yurivict, 2020-01-19:
Ok, thanks.

closed 2020-01-19 by yurivict

mdevaev, 2020-01-19:
I don't stand my ground, maybe I don't understand something? I thought the problem was in the line `In file included from <built-in>:348:`. It seems that the compiler is swearing at some built-in "files".

yurivict, 2020-01-19:
Sorry, I just don't have resources to delve into this any deeper. The FreeBSD port builds fine now, warnings are a minor thing. I would let you deal with it.

I'll let you know if there are any other issues with the FreeBSD port.

Thank you for your help!

mdevaev, 2020-01-19:
Thanks for the report!

commit b5375b83 (2020-01-18): Fixed #11: clang warning