pikvm/ustreamer #12: Breaks "thread_get_name() not implemented, you can disable it using WITH_PTHREAD_NP=0" even with -DWITH_PTHREAD_NP=0

yurivict, 2020-01-25:
http://beefy9.nyi.freebsd.org/data/113amd64-default/523997/logs/ustreamer-1.11.log

I recall that we had a conversation that ```WITH_PTHREAD_NP=0``` vs. ```WITH_PTHREAD_NP=1``` is clearly incompatible with ```#ifdef WITH_PTHREAD_NP``` that you do. ```#ifdef WITH_PTHREAD_NP``` is ```true``` when regardless of what value you supply to ```WITH_PTHREAD_NP=0```.

Please fix your ifdefs.

mdevaev, 2020-01-25:
Two quick questions.
* Where did the paramete `-DWITH_PTHREAD_NP=0` come from in `CFLAGS`?
* How does the program build start? Some script calls `make`, `gmake`, or something else?

yurivict, 2020-01-25:
> Where did the parameter -DWITH_PTHREAD_NP=0 come from in CFLAGS?

Port sets it in CFLAGS.

> How does the program build start? Some script calls make, gmake, or something else?

gmake

mdevaev, 2020-01-25:
Don't pass this option via CFLAGS. Right now you have a two options:
* Run `gmake WITH_PTHREAD_NP=0`
* Or create file `config.mk` with a single line `WITH_PTHREAD_NP=0`
Try it.

mdevaev, 2020-01-29:
@yurivict I looked at your patch in the ports. I think the best solution would be to add `MAKE_ARGS += WITH_PTHREAD_NP=0` to your port. This will solve the problem of defines.

https://svnweb.freebsd.org/ports/head/Mk/bsd.port.mk?revision=523641&view=markup#l791

closed 2020-01-30 by mdevaev