pikvm/ustreamer #13: Wrong JPEG library version: library is 80, caller expects 62

djkraven, 2020-01-27:
I just installed ustreamer today on raspberry pi to address issues with mjpg-streamer and multiple cameras. Unfortunately, I get the error of the wrong jpeg library.

```
/ustreamer --device /dev/video2 --host=0.0.0.0 --port=3080
-- INFO  [7874.091      main] -- Installing SIGINT handler ...
-- INFO  [7874.092      main] -- Installing SIGTERM handler ...
-- INFO  [7874.092      main] -- Ignoring SIGPIPE ...
-- INFO  [7874.093      main] -- Using internal blank placeholder
-- INFO  [7874.094      main] -- Listening HTTP on [0.0.0.0]:3080
-- INFO  [7874.095    stream] -- Using V4L2 device: /dev/video2
-- INFO  [7874.095    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [7874.095      http] -- Starting HTTP eventloop ...
-- INFO  [7874.306    stream] -- Device fd=8 opened
-- INFO  [7874.306    stream] -- Using input channel: 0
-- INFO  [7874.306    stream] -- Using TV standard: DEFAULT
-- INFO  [7874.314    stream] -- Using resolution: 640x480
-- INFO  [7874.314    stream] -- Using pixelformat: YUYV
-- INFO  [7874.321    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [7874.324    stream] -- Requested 5 HW buffers, got 5
-- INFO  [7874.330    stream] -- Capturing started
-- INFO  [7874.331    stream] -- Using JPEG quality: 80%
-- INFO  [7874.331    stream] -- Creating pool with 4 workers ...
-- INFO  [7874.331    stream] -- Capturing ...
Wrong JPEG library version: library is 80, caller expects 62
```

mdevaev, 2020-01-28:
Is this a problem on raspbian? Can you remove the libjpeg6 and rebuild ustreamer?

djkraven, 2020-01-28:
it is on a Raspbian GNU/Linux 10 (buster)

Trying to remove returns errors due to dependencies. I have the libjpeg8 and libjpeg8-dev installed if there is a way to point the make to the correct version?

```
apt-get remove libjpeg6
The following packages have unmet dependencies:
 ant : Depends: default-jre-headless (>= 2:1.8) but it is not going to be installed or
                java8-runtime-headless
 ca-certificates-java : Depends: default-jre-headless but it is not going to be installed or
                                 java8-runtime-headless
 libavcodec58 : Depends: librsvg2-2 (>= 2.14.4) but it is not going to be installed
 libgdk-pixbuf2.0-0 : Depends: libjpeg62-turbo (>= 1.3.1) but it is not going to be installed
                      Depends: libtiff5 (>= 4.0.3) but it is not going to be installed
                      Recommends: libgdk-pixbuf2.0-bin but it is not going to be installed
 libgtk-3-0 : Depends: adwaita-icon-theme but it is not going to be installed
              Recommends: libgtk-3-bin but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```
dpkg -l | grep libjpeg
```
libjpeg62-turbo:armhf                 1:1.5.2-2+b1                        armhf        libjpeg-turbo JPEG runtime library
libjpeg8:armhf                        8d1-2                               armhf        Independent JPEG Group's JPEG runtime library
libjpeg8-dev:armhf                    8d1-2                               armhf        Development files for the IJG JPEG library
```

mdevaev, 2020-01-28:
Okay, try to remove libjpeg8(-dev)

djkraven, 2020-01-28:
That worked. I had to install the libjpeg62-turbo-dev, run make, and it runs. Now to figure out a few items. 

1.  Why is streaming two cameras results in 'no signal'. I had to add the --workers 2 to it would at least run two instances.

```
nohup ./ustreamer --process-name-prefix ustreamer --device /dev/video2 --resolution 1280x720  --desired-fps 4 --format YUYV --workers 2 --host=0.0.0.0 --port=60808 &
nohup ./ustreamer --process-name-prefix ustreamer --device /dev/video0 --resolution 1280x1024 --desired-fps 5 --format YUYV --workers 2 --host=0.0.0.0 --port=60809 &
```

2. How to control the exposure (or something) that is causing the stream to flash brighter periodically. The --brightness-auto and --gain-auto are not supported by the cameras.

mdevaev, 2020-01-28:
Show me the log

djkraven, 2020-01-28:
Here are the log files created by the follow: 
```
# ./ustreamer --process-name-prefix ustreamer --log-level 3 --device /dev/video0 --workers 2 --format YUYV --host=0.0.0.0 --port=60809>video0.log
# ./ustreamer --process-name-prefix ustreamer --log-level 3 --device /dev/video2 --workers 2 --format YUYV --host=0.0.0.0 --port=60808>video2.log
```
[video0.log](https://github.com/pikvm/ustreamer/files/4124469/video0.log)
[video2.log](https://github.com/pikvm/ustreamer/files/4124470/video2.log)

mdevaev, 2020-01-28:
Try `--device-timeout=8` and `--format=jpeg` to use hardware webcam encoding. Also remove option `--workers`.

djkraven, 2020-01-28:
That did it. Both cameras are streaming. Now make an init.d script and figure out the custom web page to mirror what I had going with mjpg-streamer. Is it possible to adjust the image control settings with ustreamer running or does the process have to stop and restart? 

The logging is good to have connected IP information so it can be monitored. 
Thank you for your assistance.

```
nohup ./ustreamer --process-name-prefix ustreamer --log-level 0 --device /dev/video0 --device-timeout=8 --format jpeg --resolution 528x297 --host=0.0.0.0 --port=3080>video0.log &
nohup ./ustreamer --process-name-prefix ustreamer --log-level 0 --device /dev/video2 --device-timeout=8 --format jpeg --resolution 880x497 --host=0.0.0.0 --port=3080>video2.log &
```

mdevaev, 2020-01-28:
> That did it. Both cameras are streaming.

Great!

> custom web page

You can serve static files and directories using the option `--static=path/to/dir`.

>  Is it possible to adjust the image control settings **with ustreamer running** or does the process have to stop and restart?

This cannot be done right now, because this feature has not yet been implemented. Ustreamer was designed to work with video capture devices, and this feature was secondary.

In our system, ustreamer is run using a small python server. The js code from the browser is connected to the server via a websocket, then the server, while there is at least one client, starts ustreamer and restarts it when the parameters change.

I understand that this is probably not as easy as you would like, and I plan to implement image parameter settings when I have time. Someday.

djkraven, 2020-01-30:
An option is to use **v4l2-ctl** to control the camera setting while ustreamer is running.
`https://www.linuxtv.org/wiki/index.php/V4l-utils`

mdevaev, 2020-02-06:
Good idea!

Okay, the problem described in the name of this ticket can be considered solved. I'll implement the configurable image parameters sometime later.

closed 2020-02-06 by mdevaev