pikvm/ustreamer #14: How does the --static work?

djkraven, 2020-01-29:
I have added the --static and don't see reference to it in the logging or use of the directory.
```
nohup ./ustreamer --process-name-prefix ustreamer --log-level 3 --device /dev/video2 --device-timeout=8 --format jpeg --resolution 880x497 --host=0.0.0.0 --port=3080 --static=/var/www/html/ustreamer/ >video2.log &
```
```
-- INFO  [76378.461      main] -- Installing SIGINT handler ...
-- INFO  [76378.461      main] -- Installing SIGTERM handler ...
-- INFO  [76378.461      main] -- Ignoring SIGPIPE ...
-- DEBUG [76378.461      main] -- Increasing picture 0x0x7d1b70 buffer: 0 -> 13845 (+13845)
-- INFO  [76378.461      main] -- Using internal blank placeholder
-- DEBUG [76378.461      main] -- Increasing picture 0x0x7d14a8 buffer: 0 -> 13845 (+13845)
-- DEBUG [76378.461      main] -- Binding HTTP to [0.0.0.0]:3080 ...
-- INFO  [76378.462      main] -- Listening HTTP on [0.0.0.0]:3080
-- INFO  [76378.462    stream] -- Using V4L2 device: /dev/video2
-- INFO  [76378.462    stream] -- Using desired FPS: 0
-- DEBUG [76378.462    stream] -- _stream_init_loop: stream->proc->stop=0
================================================================================
-- INFO  [76378.463      http] -- Starting HTTP eventloop ...
-- DEBUG [76378.485      http] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [76378.485      http] -- HTTP: Dropped same frame (BLANK) number 0
-- DEBUG [76378.501      http] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [76378.501      http] -- HTTP: Dropped same frame (BLANK) number 1
-- DEBUG [76378.510      http] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [76378.511      http] -- HTTP: Dropped same frame (BLANK) number 2
```

mdevaev, 2020-01-29:
Just place `index.html` to this folder.

Ustreamer does not log accesses to static files, so as not to clutter the log. Is this necessary?

mdevaev, 2020-01-29:
JFYI: Fixed some minor cases of static files in master.

djkraven, 2020-01-29:
Wasn't looking for file access logging specifically though at a debug level the http access and error log could be useful. What I was looking for is confirmation ustreamer started properly the expected options were implemented correctly. I have confirmed the --static is working.

djkraven, 2020-01-29:
What are the fixes you implemented? Anything that would be notice in using the --static option?

mdevaev, 2020-01-29:
Yes. Fixed mime types processing and serving `.` (dot) directory.

mdevaev, 2020-01-29:
Also I'll add message about serving

djkraven, 2020-01-29:
Good to know and will check that out. Right now I trying to get the streaming image to scale properly like was on the mjpg-streamer page I had.

djkraven, 2020-01-31:
I have the custom static pages working, though when running the two cameras one them has a flashing that is frustrating. I have tried various settings to reduce it but the results end up being worse than the flashing. This does happen with mjpg-streamer when the cameras are streamer so I figure there is either a setting tweak I better understanding on or something. Any suggestions? 

```
nohup ./ustreamer --process-name-prefix ustreamer-0 --log-level 0 --device /dev/video2 --device-timeout=8 --format jpeg --quality 100 --resolution 1920x1080 --host=0.0.0.0 --port=60808 --static /var/www/html/ustreamer-0/ > /var/log/ustreamer-0.log &
nohup ./ustreamer --process-name-prefix ustreamer-1 --log-level 0 --device /dev/video0 --device-timeout=8 --format jpeg --quality 100 --resolution 1280x1024 --host=0.0.0.0 --port=60809 --static /var/www/html/ustreamer-1/ > /var/log/ustreamer-1.log &

```

mdevaev, 2020-01-31:
I once heard of a similar problem. It looks like it's some kind of driver glitch. No idea how to fix it yet.

djkraven, 2020-01-31:
If it were a driver glitch wouldn't mjpg-streamer present the same behavior?

mdevaev, 2020-01-31:
Sorry, I didn't understand you a bit. English is not my native language.

Show me your logs. Also, if possible, take a video of what these glitches look like.

djkraven, 2020-02-03:
Here are the logs and the short video. ustreamer-0.log is the camera on the left side of the video and ustreamer-1.log the right side.

[2020-02-03 14-31-19.zip](https://github.com/pikvm/ustreamer/files/4149810/2020-02-03.14-31-19.zip)
[ustreamer-1.log](https://github.com/pikvm/ustreamer/files/4149868/ustreamer-1.log)
[ustreamer-0.log](https://github.com/pikvm/ustreamer/files/4149869/ustreamer-0.log)

mdevaev, 2020-02-03:
Can you estimate how much FPS was when using mjpg-streamer?

What happens if you try to limit fps for ustreamer using option `--desired-fps=12`?

djkraven, 2020-02-06:
the --desired-fps=12 seems to have reduced maybe stopped the flashing. I am encountering the same problem I was having with mjpg-streamer of no stream loading for one or both when the page is reloaded.

mdevaev, 2020-02-06:
What browser do you use?

My hypothesis is that the browser is experiencing problems when rendering two video streams with high resolution and fps.

djkraven, 2020-02-06:
Firefox is what I prefer to use but for these streaming pages it is either Chrome or Edge. The page I have setup is actually streaming 5 cameras. These two are the only ones that have the issue.

mdevaev, 2020-02-06:
> I am encountering the same problem I was having with mjpg-streamer of no stream loading for one or both when the page is reloaded.

I'm sorry, I'm not sure what you mean. You mean that on a page where several mjpg-streamers broadcast a stream, in case one of the cameras doesn't load, the other camera might start flickering?

> These two are the only ones that have the issue. Other cameras show a stream from the same machine (raspberry?)? Do they show a stream using ustreamer? How high is the fps and resolution of other cameras?

djkraven, 2020-02-06:
Here the setup
- Linux Mint  running Apache for the presentation of 6 different streams from different sources
  - 2 usb cameras mjpg-streamer Linux Mint (same box Apache is on)  
  - 1 usb camera  mjpg-streamer Raspberry Pi 3b+ 
  - 2 usb cameras ustreamer        Raspberry Pi 3b+ 
  - Remote web camera stream 

The Pi that is running ustreamer is the one having issues with the streams failing to load when the presentation page of all 6 is refreshed in a browser. This was not a problem with an earlier release of mjpg-streamer. In searching for the cause I was the post reference ustreamer as a solution for sounded like a similar problem. This may be an issue with the Pi more than the streaming solution. If you have thoughts on where to look for the failures

mdevaev, 2020-02-07:
I believe the problem may be the slow return of traffic on the raspberry pi. That's why I suggested reducing the fps to reduce the load on the channel and/or cpu. In fact, I think the reason lies outside of ustreamer, since it doesn't perform any image transformations to make the characteristic blinking distortions. If I were you, I would start by reducing the resolution of all cameras to 640x480 and 8fps. The fundamental difference between ustreamer and mjpg-streamer is that ustreamer tries to occupy the entire channel without restrictions in order to transmit the maximum possible fps from the device. When combined with high resolution, this can be a problem if you are broadcasting multiple streams.

djkraven, 2020-02-07:
I can try that though the 640x480 crops the image which defeats the reason for wide-angle camera.

mdevaev, 2020-02-07:
See what resolutions your camera supports and use the nearest appropriate one.

djkraven, 2020-02-07:
I noticed that the Pi was using swap and realized the GUI was not necessary so I switch the boot to use CLI. I also stopped the DHCPD service since it uses a static. This helps with the frame rate flashing. It still has the stream/image broken issue and which I need to test the other resolutions for. When I was trying them in the past all the lower options cropped the image. The following is what I have working currently. killall does not work on the ustreamer running processes.

```
kill `ps axu |grep -e ' ustreamer-'| grep -v grep|awk '/ustreamer/ {print $2 " " $11 }'`
sleep 2 

nohup ./ustreamer --process-name-prefix ustreamer-1 --log-level 0 --device /dev/video0 --device-timeout=8 --format jpeg --quality 100 --resolution 1280x1024  --desired-fps=14 --host=0.0.0.0 --port=3081   --static /var/www/html/ustreamer-1/ > /var/log/ustreamer-1.log &

nohup ./ustreamer --process-name-prefix ustreamer-0 --log-level 0 --device /dev/video2 --device-timeout=8 --format jpeg --quality 100 --resolution 1920x1080 --desired-fps=29 --host=0.0.0.0 --port=3080 --static /var/www/html/ustreamer-0/ > /var/log/ustreamer-0.log &

multitail /var/log/ustreamer-0.log /var/log/ustreamer-1.log 

```

djkraven, 2020-02-07:
This warning shows up in the logs. What causes this and is there something that should be changed to avoid it from happening?

```
[warn] Epoll ADD(1) on fd 11 failed. Old events were 0; read change was 1 (add); write change was 0 (none); close change was 0 (none): Bad file descriptor
```

mdevaev, 2020-02-10:
> killall does not work on the ustreamer running processes

If it can't find the process, then that's fine. This is because you renamed it.

```shell
$ ps x | grep ustreamer
  73669 pts/0    Sl+    0:00 ustreamer-0: ./ustreamer --process-name-prefix ustreamer-0 --log-level 0 --device /dev/video0 --device-timeout=8 --format jpeg --quality 100 --resolution 1920x1080 --desired-fps=29 --host=0.0.0.0 --port=3080 --static .
$ killall ustreamer
ustreamer: no process found
$ kill `ps axu |grep -e ' ustreamer-'| grep -v grep|awk '/ustreamer/ {print $2}'`
$ ps x | grep ustreamer
$
```

> This warning shows up in the logs

I've never seen this before. How can I reproduce it?

mdevaev, 2020-05-04:
I added some logging about files at the verbose level for debugging. This is not a full-fledged access log, but it is better than what it was before. At this moment I suggest using nginx reverse proxy if you need to have full access logs. The current functionality is sufficient for organizing a stream from an embedded system.

Since the work on this ticket has already been done, I believe that it can be closed. Also, if you have problems with hup, please open a new ticket and describe in detail how I can reproduce them.

closed 2020-05-04 by mdevaev