pikvm/ustreamer #16: init.d and systemctl scripts

djkraven, 2020-02-06:
Has anyone put together scripts to start, restart, and stop the service so ustreamer could start at boot or easily managed?

mdevaev, 2020-02-06:
No, noone did it. But I suppose you can use a script from mjpg-streamer: https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/mjpg_streamer%40.service

Like:
```ini
[Unit]
Description=uStreamer service
After=network.target

[Service]
User=ustreamer
ExecStart=/usr/bin/ustreamer --device /dev/%I ... (other opts)

[Install]
WantedBy=multi-user.target
```

Create a system user `useradd -r ustreamer`; if necessary, add it to the `video` group: `usermod -a -G video ustreamer`. Then run `systemctl enable ustreamer@video0.service` for `video0`, `video1`, etc.

You might want to make sure that each usb port has a specific camera. This is easy to do with udev:

https://unix.stackexchange.com/questions/66901/how-to-bind-usb-device-under-a-static-name

https://wiki.archlinux.org/index.php/Udev#Setting_static_device_names

For example: https://github.com/pikvm/kvmd/blob/master/configs/os/udev/v0-vga-rpi3.rules#L3

mdevaev, 2020-02-10:
Looks like I answered your question :) I've added a link to this issue to README for those who will encounter this problem.

closed 2020-02-10 by mdevaev

Taudris, 2021-08-31:
Here's a udev rules file I wrote for my C920. I put it in `/etc/udev/rules.d/99-video-c920.rules`, working on Raspbian buster.
```
# reference: https://man7.org/linux/man-pages/man7/udev.7.html

ACTION=="remove", GOTO="end"

SUBSYSTEM=="video4linux", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="082d", ATTRS{serial}=="XXXXXXXX", ATTR{index}="0", GOTO="is_rear"
GOTO="end"

LABEL="is_rear"
RUN="/usr/bin/logger -t udev /dev/video-rear connected"
OWNER="root",GROUP="video",MODE="0660",SYMLINK="video-rear"
RUN+="/usr/bin/v4l2-ctl --device /dev/video-rear --set-ctrl=exposure_auto_priority=0 --set-ctrl=exposure_auto=1 --set-ctrl=white_balance_temperature_auto=0 --set-ctrl=power_line_frequency=0 --set-ctrl=focus_auto=0"
RUN+="/usr/bin/v4l2-ctl --device /dev/video-rear --set-ctrl=exposure_absolute=55 --set-ctrl=gain=0 --set-ctrl=white_balance_temperature=3100 --set-ctrl=focus_absolute=25 --set-ctrl=sharpness=224"

LABEL="end"
```

cthulux, 2022-02-13:
Hi! Thank you for the systemctl script, it works great! Do you know if I can create a systemd socket unit instead of a systemd service unit? I want it to be activated when requested and deactivated after the client disconnects. Is that possible? Thanks!

mdevaev, 2022-02-13:
Sup. This is not supported now because it requires some code on the part of ustreamer.

cthulux, 2022-02-13:
Ok, anyway without it works great! Thanks!

mdevaev, 2022-02-13:
@cthulux please create the new issue for this feature, I will add it. I can't promise how soon, but I'll try to find the time.

mdevaev, 2022-02-13:
@cthulux Implemented socket activation. Please try it from master branch. Build with `make WITH_SYSTEMD=1` and use startup command `ustreamer --systemd`.

mdevaev, 2022-02-13:
@cthulux upd: added option `--exit-on-no-clients=N`. N is the number of seconds without clients after which ustreamer will finish execution.

cthulux, 2022-02-14:
Thank you @mdevaev ! This is great! I had an issue I solved by installing libsystemd-dev (just commenting it for anyone else that wants to try it). I'll try it and come back to let you know how it was. Thanks!

cthulux, 2022-02-14:
Hi @mdevaev, I've been trying to follow online tutorials in order to create the systemd socket file with the new options but I was not very sucessful, I don't know what is missing:

`cat /lib/systemd/system/rpicam@.service
[Unit]
Description=RaspberryPi camera stream
After=network.target
Wants=network.target
Requires=rpicam.socket

[Service]
Type=simple
User=ustreamer
#StandardOutput=socket
PrivateUsers=yes
DeviceAllow=/dev/video0
PrivateTmp=yes
ProtectHome=yes
Restart=on-abnormal
RestartSec=20

ExecStart=/usr/local/bin/ustreamer --process-name-prefix ustreamer-%I --log-level 0 --drop-same-frames=30 --desired-fps 6 --device-timeout=8 --buffers=3 -r 1296x972 --device=/dev/video%I --quality 80 --slowdown --systemd --host 127.0.0.1 --port=8089

[Install]
WantedBy=multi-user.target`

And the systemd socket file is:
`cat /lib/systemd/system/rpicam.socket
[Unit]
Description=CUPS Scheduler
PartOf=rpicam@0.service

[Socket]
ListenStream=/run/ustreamer.sock
SocketUser=ustreamer

Accept=yes

[Install]
WantedBy=sockets.target`

Thanks!

mdevaev, 2022-02-14:
Remove `--host 127.0.0.1 --port=8089`

mdevaev, 2022-02-14:
Also remove `Accept=yes`

cthulux, 2022-02-14:
Hi @mdevaev , I followed your suggestions and had to change the service template file to service file only because there was some issue with the template I couldn't manage. Finaly I was able to make the sytemd socket file run without issues but I couldn't connect to it.

I am using Haproxy in order to add TLS encryption and authentication. Previous to adding the systemd socket file, It listened in port 8090 and forwarded it to ustreamer localhost port 8089. 

Now I adapted it so it listens on the tcp unix path but I am unable to see any video since it looks like the systemd service is not trigered or I don't know what.

This is the Haproxy which worked previously with "server rpicam 127.0.0.1:8089" and now for the socket systed file I changed to "server rpicam /run/ustreamer.sock":
```
userlist rpicamcredentials
   user USERNAME password HASHED_PASSWORD

frontend rpilibcam
    bind-process 2-3
    bind :8090 tfo ssl crt /etc/haproxy/certs/ process 2-3 alpn h2,http/1.1 curves X25519:P-256:secp384r1
    bind abns@haproxy-clt7 accept-proxy tfo ssl crt /etc/haproxy/certs/ process 2-3 alpn h2,http/1.1 curves X25519:P-256:secp384r1
    mode http
    option forwardfor
    http-request redirect scheme https unless { ssl_fc }
    http-request auth unless { http_auth(rpicamcredentials) }
    default_backend rpicam_8089

backend rpicam_8089
    mode http
    option forwardfor
    retry-on all-retryable-errors
    #server rpicam 127.0.0.1:8089
    server rpicam /run/ustreamer.sock
```

This is the systemd service file:

```
cat /lib/systemd/system/ustreamer.service
[Unit]
Description=Ustreamer RaspberryPi camera stream
After=network.target ustreamer.socket
Wants=network.target
Requires=ustreamer.socket

[Service]
#Environment="SCRIPT_ARGS=%I" 
Type=simple
User=ustreamer
#StandardOutput=socket
PrivateUsers=yes
DeviceAllow=/dev/video0
PrivateTmp=yes
ProtectHome=yes
Restart=on-abnormal
RestartSec=20

ExecStart=/usr/local/bin/ustreamer --process-name-prefix ustreamer-0 --log-level 0 --drop-same-frames=30 --desired-fps 6 --device-timeout=8 --buffers=3 -r 1296x972 --device=/dev/video0 --quality 80 --slowdown --systemd

[Install]
WantedBy=multi-user.target
```
And this is the systemd socket file:
```
cat /lib/systemd/system/ustreamer.socket
[Unit]
Description=Ustreamer socket unit
PartOf=ustreamer.service

[Socket]
ListenStream=/run/ustreamer.sock
SocketUser=ustreamer
SocketMode=755

[Install]
WantedBy=sockets.target
```
I added ustreamer user for both systemd unit files, also added haproxy user to the ustreamer group and just in case made the ustreamer socket permissions 755 so haproxy can read the socket (with option SocketMode=755).

```
ls -lrth /run/ | grep -i ustreamer
srwxr-xr-x  1 ustreamer ustreamer    0 Feb 14 11:57 ustreamer.sock
```
After I login to https://host:8090 I can see the error 
`503 Service Unavailable`

Do you know if there is another way (instead of Haproxy) to test if the /run/ustreamer.sock.

If this is something dificult, I can use the previous configuration without socket file which worked really fine.

Thanks a lot!

mdevaev, 2022-02-14:
Try `curl --unix /run/ustreamer.sock http://127.0.0.1:8080/state`.

I used the systemd debugging tool to check. Everything works:
```
$ systemd-socket-activate -l /tmp/foo.sock ./ustreamer --systemd
$ curl --unix /tmp/foo.sock http://127.0.0.1:8080/state
```

cthulux, 2022-02-15:
Thank you @mdevaev! It works! Now I'll have to figure out myself howto make it work with Haproxy. :)

mdevaev, 2022-02-15:
Glad to hear it!

TheNinsa, 2022-07-04:
Hello o/

I need some insight on the System service described above.
(Its a small Raspberry Pi camera module v2 inside a 3d Printer)

```
[Unit]
Description=uStreamer Raspberrypi Camera stream
After=network.target
Wants=network.target
[Service]
Type=simple
User=ustreamer
DeviceAllow=/dev/video0
Restart=on-abnormal
RestartSec=20

ExecStart=usr/bin/ustreamer/ustreamer --process-name-prefix ustreamer-0 --log-level 0 --drop-same-frame 30 --device-timeout=8 --buffers=3 --device=/dev/video0

[Install]
WantedBy=multi-user.target
```

Throws this:
```
pi@Voron-v0615:~ $ sudo systemctl status ustreamer@0.service
● ustreamer@0.service - picam stream
     Loaded: loaded (/etc/systemd/system/ustreamer@.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-07-04 14:23:23 BST; 35s ago
   Main PID: 546 (main)
      Tasks: 3 (limit: 3597)
        CPU: 292ms
     CGroup: /system.slice/system-ustreamer.slice/ustreamer@0.service
             └─546 ustreamer-0: /usr/bin/ustreamer/ustreamer --process-name-prefix ustreamer-0 --log-level 0 --device /dev/video0 --device-timeout=8

Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.340    stream] -- Device fd=8 opened
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.340    stream] -- Using input channel: 0
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.340    stream] -- Using resolution: 640x480
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.341    stream] -- Using format: YUYV
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.341    stream] -- Querying HW FPS changing is not supported
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.341    stream] -- Using IO method: MMAP
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.351    stream] -- Requested 5 device buffers, got 5
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- ERROR [31.351    stream] -- Can't start capturing: Invalid argument
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.353    stream] -- Device fd=8 closed
Jul 04 14:23:59 Voron-v0615 ustreamer[546]: -- INFO  [31.353    stream] -- Sleeping 1 seconds before new stream init ...

```

Any insight as to what "invalid argument" refers to?

Thank you for any help in advance ^^

-Ninsa

robertsamples, 2022-12-18:
I'm having some trouble trying to parse the comments here and am not sure if I am having the same issues that others are having.

after enabling the script with systemctl I recieve this error:
Failed to enable unit: File multi-user.target: Identifier removed
I found people discussing changing the target to default.target which caused an error when enabling the script but by the command to the following I could enable, start and get the status with log files indicating it was working:
sudo systemctl enable ustreamer@**user**.service

Unfortunately when I try to connect to the specified port I can't connect as I can when I manually start ustreamer. Do you have any suggestions? Thank you!

ZyberSE, 2023-03-31:
I tried setting up the socket but failed, anyone care to give me a step-by-step example? Would be an nice addition to the otherwise well written guide.

cthulux, 2023-04-05:
Hi @ZyberSE, It is a bit difficult for me to answer this since I am not currently using it. 

This are the instructions I was able to recover from what I did to get it working:

Step 1: Download latest version and compile with "WITH_SYSTEMD=1". You don't need to use git, since current releases may include the systemd compilation flag. At the time I did it, I had to use git due to it was a new feature @mdevaev implemented:

```
git fetch --depth=1 https://github.com/pikvm/ustreamer
make WITH_SYSTEMD=1
make install
```

If you need to update it later:

```
git fetch
git reset --hard HEAD
git merge '@{u}'
make WITH_SYSTEMD=1
make install
```

Create specific camera user:

`sudo useradd -r ustreamer; sudo usermod -a -G video ustreamer`

Check if you have the systemd services for ustreamer:

```
root@rpi4:/home/pi# systemctl list-unit-files | grep -i ustream
ustreamer.service                      disabled        enabled
ustreamer@.service                     disabled        enabled
ustreamer.socket                       masked          enabled
root@rpi4:/home/pi#

```
If you have those, then just start/enable the socket:

`systemctl daemon-reload && sudo systemctl enable –now ustreamer.socket`

Else, create the systemd files. Here you are some systemd examples (they are overloaded with options that certainly are not needed and others may not be up to date,  that's why I said examples):

root@rpi4:/home/pi# systemctl cat ustreamer@.service
```
# /lib/systemd/system/ustreamer@.service
[Unit]
Description=Ustreamer RaspberryPi camera stream
After=network.target 
Wants=network.target

[Service]
Environment="SCRIPT_ARGS=%I"
Type=simple
User=ustreamer
PrivateUsers=yes
DeviceAllow=/dev/video0
PrivateTmp=yes
ProtectHome=yes
ProtectSystem=strict
Restart=on-abnormal
RestartSec=5
NoNewPrivileges=yes

# SIN systemd socket #
ExecStart=/usr/local/bin/ustreamer --process-name-prefix ustreamer-%I --log-level 0 --drop-same-frames=30 --desired-fps 6 --device-timeout=8 --buffers=3 -r 1296x972 --d>

[Install]
WantedBy=multi-user.target
```

root@rpi4:/home/pi# systemctl cat ustreamer.service
```
# /lib/systemd/system/ustreamer.service
[Unit]
Description=Ustreamer RaspberryPi camera stream
After=network.target ustreamer.socket
Wants=network.target
Requires=ustreamer.socket
KillMode=control-group

[Service]
#Environment="SCRIPT_ARGS=%I"
NoNewPrivileges=yes
Type=simple
User=ustreamer
ProcSubset=pid
ProtectProc=noaccess
ProtectClock=yes
#StandardOutput=socket
PrivateUsers=yes
DeviceAllow=/dev/video0
InaccessiblePaths=/etc/
InaccessiblePaths=/boot/
InaccessiblePaths=/home/
PrivateTmp=yes
ProtectHome=tmpfs
ProtectSystem=strict
#ProtectHome=yes
Restart=on-abnormal
#RestrictNetworkInterfaces=lo
PrivateNetwork=yes
RestartSec=10
#NoExecPaths=/
UMask=0007
#ExecPaths=/usr/local/bin/ustreamer
ExecStart=/usr/local/bin/ustreamer --process-name-prefix ustreamer-0 --log-level 0 --drop-same-frames=30 --desired-fps 6 --device-timeout=8 --buffers=3 -r 1296x972 --device=/dev/video0 --quality 80 --slowdown --exit-on-no-clients=300 --systemd

[Install]
WantedBy=multi-user.target
```

This is were you create the socket file, please edit and replace "/var/lib/haproxy/ustreamer.sock" with your socket name
root@rpi4:/home/pi# cat /lib/systemd/system/ustreamer.socket 
```
[Unit]
Description=Ustreamer socket unit
PartOf=ustreamer.service

[Socket]
#Please, your socket here!
ListenStream=/var/lib/haproxy/ustreamer.sock
SocketUser=ustreamer
SocketMode=006

[Install]
WantedBy=sockets.target
```

> I tried setting up the socket but failed, anyone care to give me a step-by-step example? Would be an nice addition to the otherwise well written guide.

cthulux, 2023-04-05:
> I'm having some trouble trying to parse the comments here and am not sure if I am having the same issues that others are having.
>
> after enabling the script with systemctl I recieve this error: [...]

Hi! I just saw your question today, are you still facing this issue?

korylek1231, 2023-05-10:
Hi. I Found this thread and decided to create my own autostart script. After some time i manage to make full working and enabling on start. Wanted to share it. I'm running OctoPrint on LUbuntu. Video is from USB camera

```
[Unit]
Description=uStreamer service
**After=network-online.target
Wants=network-online.target** // Those two are very important! If u wont add it script doesn't turn on system boot!
Restart=on-failure
[Service]
DeviceAllow=/dev/video0
User=[username]
ExecStart= [_Here i put path to ./ustreamer script. In my case it's "/home/me/ustreamer/src/ustreamer.bin"  with addons like --device=/dev/video0 etc._]
[Install]
WantedBy=default.target
```

All other things like creating etc. same as @mdevaev mentioned in first post (btw thx for it). If u're gonna use only one camera it better to create script without "@" allias.

jacob-swanson, 2024-07-19:
I slapped together a quick nixos module for this with socket activation, since there wasn't one in nixpkgs, and figured I'd share it here.

```nix
{
  config,
  lib,
  pkgs,
  ...
}:
with lib;
let
  cfg = config.services.ustreamer;
in
{
  options = {
    services.ustreamer = {
      enable = mkEnableOption "ustreamer webcam streamer";

      user = mkOption {
        type = types.str;
        default = "ustreamer";
        description = "ustreamer user name.";
      };

      group = mkOption {
        type = types.str;
        default = "video";
        description = "ustreamer group name.";
      };

      listenStream = mkOption {
        type = types.str;
        default = "/run/ustreamer.sock";
        description = "ustreamer socket.";
      };

      extraArgs = mkOption {
        type = types.str;
        default = "";
        description = "ustreamer cli args";
      };
    };
  };

  config = mkIf cfg.enable {
    users.users = optionalAttrs (cfg.user == "ustreamer") {
      ustreamer = {
        group = cfg.group;
        isSystemUser = true;
      };
    };

    systemd.services.ustreamer = {
      description = "ustreamer webcam streamer";
      after = [
        "network.target"
        "ustreamer.socket"
      ];
      requires = [ "ustreamer.socket" ];
      serviceConfig = {
        User = cfg.user;
        Group = cfg.group;
        ExecStart = "${pkgs.ustreamer}/bin/ustreamer --systemd --exit-on-no-clients 300 ${cfg.extraArgs}";
      };
    };

    systemd.sockets.ustreamer = {
      description = "ustreamer webcam socket";
      partOf = [ "ustreamer.service" ];
      wantedBy = [ "sockets.target" ];
      listenStreams = [ cfg.listenStream ];
    };
  };
}
```

Then you can use it for nginx+mainsail like this!
```nix
services.ustreamer = {
  enable = true;
  extraArgs = "--resolution 1280x720";
};
services.nginx.upstreams.ustreamer = {
  servers = {
    "unix:${config.services.ustreamer.listenStream}" = { };
  };
};
services.mainsail = {
  enable = true;
  nginx = {
    # ...
    locations."/webcam/" = {
      proxyPass = "http://ustreamer/";
      extraConfig = ''
        postpone_output 0;
        proxy_buffering off;
        proxy_ignore_headers X-Accel-Buffering;
      '';
    };
  };
};
```

bamhm182, 2024-08-01:
@jacob-swanson, your config looks good to me. You should submit it to nixpkgs and help everyone out.

oldmanbluntz, 2025-02-05:
is there any way this can be updated to work on Ubuntu 24.10 in 2025 by any chance? I keep running into issues trying to get any of this to work with all the changes to ubuntu and systemctl in the last 5 years since the original writeup was made.

mdevaev, 2025-02-05:
Which problem do you have?

oldmanbluntz, 2025-02-06:
nevermind. it was a little bit of my own confusion, and a little bit on the end of the changes between ubuntu versions in 2022 and versions from 2024. I got it all sorted out.

I did have to make 2 different streamer@.service files (streamer1@.service and streamer@.service), since 2 of my cameras are max 640x480 res and my 3rd one does 1080p max.