pikvm/ustreamer #17: Dependencies are unclear

FalconFour, 2020-02-22:
Trying to build this, I spent a lot of time researching errors and trying to figure out what the following line meant:

> You'll need  ```make```, ```gcc```, ```libevent``` with ```pthreads``` support, ```libjpeg8```/```libjpeg-turbo```, ```libuuid``` and ```libbsd``` (only for Linux).

```make``` and ```gcc``` were pretty obvious, but the rest were a significant challenge to untangle - every one of them had a different package name from what was written there, and each had their own naming style to understand (e.g. "uuid-dev" instead of "libuuid-dev", and two different packages for libevent).

I ended up fixing it, and the following line is the result of a good chunk of research time:
```sudo apt install libevent-dev libevent-pthreads-2.1-6-dev libjpeg8-dev uuid-dev libbsd-dev```

I might end up adding a patch to support GRBG format cameras (an academic exercise I'm undertaking with a 20-year-old webcam and a Pi Zero, that hit a dead-end with a "select() timeout" error with mjpg-streamer), so if I figure that out, I'll try and remember to patch the doc as well in a separate pull. I'm just not terribly well versed in Git, soo... ;)

mdevaev, 2020-02-22:
Hi! This confusion arose because these packages are named differently on different operating systems. I have given names for Arch Linux.

> sudo apt install libevent-dev libevent-pthreads-2.1-6-dev libjpeg8-dev uuid-dev libbsd-dev

For which system are these names relevant? I'll add this to the README.

> GRBG

Cool! What encoding method do you implement this for? CPU, HW or OMX?

FalconFour, 2020-02-25:
Those are for Raspbian 😅 presumably the main audience for this, but I understand the confusion there.

As for GRBG, oof... I'm having so much trouble getting the driver itself working ([saga](https://www.reddit.com/r/linuxquestions/comments/f7zlb1/debugging_gspca_with_v4l2_select_timeout/)), I'm chasing the core driver issues around first. It seems like OMX won't support it (it seems to have a fairly limited feature set...?) so I wrote it at first for CPU. It's really simple, just taking each pixel (one color at a time), updating a local variable for green, red, or blue depending on which one it should be "seeing" at the time, then blasting out R, G, and B to the line array as it goes - so only one color changes at a time, but on sum, it should be a fairly approximate color image.

On second thought, while I'm busy hacking the driver itself, I might just push the change back up to the driver-level, emulating RGB24 instead, since that's a universal format (and GRBG definitely isn't). If I can get the darn camera working, we'll see! I just tested the camera in Windows, and it still works... ;)

A little off topic there, but fun. =)

mdevaev, 2020-02-25:
Awesome :D

Okay, I've added a note about packages naming. I'm closing this issue, join the discord of this project if you want, I think your research may be interesting to many people: https://discord.gg/bpmXfz5

closed 2020-02-25 by mdevaev