pikvm/ustreamer #18: Video capture not supported by the device

KiillThemAll, 2020-04-24:
Mjpg streamer failing with select timeout at random time and its such pain, because camera controlling laser cutter. Tried ustreamer, output as follows:

sudo ./ustreamer --host=0.0.0.0
-- INFO  [6670.969      main] -- Installing SIGINT handler ...
-- INFO  [6670.969      main] -- Installing SIGTERM handler ...
-- INFO  [6670.969      main] -- Ignoring SIGPIPE ...
-- INFO  [6670.969      main] -- Using internal blank placeholder
-- INFO  [6670.970      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [6670.970    stream] -- Using V4L2 device: /dev/video0
-- INFO  [6670.970    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [6670.970    stream] -- Device fd=8 opened
-- ERROR [6670.970    stream] -- Video capture not supported by the device
-- INFO  [6670.970    stream] -- Device fd=8 closed
-- INFO  [6670.970    stream] -- Sleeping 1 seconds before new stream init ...
-- INFO  [6670.971      http] -- Starting HTTP eventloop ...

Mjpg streamer starts fine with this:

mjpg_streamer -i "input_uvc.so -y -d /dev/video0 -r 640x480" -o "output_http.so -w ./www"
MJPG Streamer Version: git rev: 58e952383cbe973641a3ce6c6a738bafc1605337
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled

Nadeyus na pomosh)

mdevaev, 2020-04-24:
Что за камера?

KiillThemAll, 2020-04-24:
Вопрос сложный, досталась от коллеги, который в свое время закупал. Снять к сожалению могу ее очень редко, сбивается настройка, придется таблицы высот переделывать. Она регулирует высоту линзы относительно материала

KiillThemAll, 2020-04-24:
а патч для mjpg как я понимаю только timeout добавляет?

KiillThemAll, 2020-04-24:
По логам без названия камеры ничего понять не выйдет?

mdevaev, 2020-04-24:
> а патч для mjpg как я понимаю только timeout добавляет?
Какой патч?

> По логам без названия камеры ничего понять не выйдет?
Можно погрепать dmesg, там должно быть написано.

KiillThemAll, 2020-04-24:
vot etot https://github.com/jacksonliam/mjpg-streamer/pull/164

mdevaev, 2020-04-24:
Это для устройства видеозахвата по HDMI. Там есть таймаут, да, но не только он.

KiillThemAll, 2020-04-24:
dmesg по поводу камеры не грепается, никаких упоминаний модели не нашел, скорее всего изза того что она на порту csi стоит, а не в usb воткнута, хотя точно не знаю, не знаток.
Попробую побольше поставить timeout в mjpg, может перестанет вылетать

mdevaev, 2020-04-24:
Таймаут не поможет, камера хочет работать с каким-то странным режимом.
Попробуйте `dmesg | grep video0`.

KiillThemAll, 2020-04-24:
пробовал, пусто

KiillThemAll, 2020-04-24:
dmesg | grep video
[    1.996917] iommu: Adding device ff9a0000.video-codec to group 3
[    2.001299] Linux video capture interface: v2.00
[    3.103501] probe device ff9a0000.video-codec
[    3.103583] rockchip-vpu ff9a0000.video-codec: can't request region for resource [mem 0xff9a0000-0xff9a07ff]
[    3.103617] rockchip-vpu ff9a0000.video-codec: rockchip_vpu_hw_probe failed
[    3.103721] rockchip-vpu: probe of ff9a0000.video-codec failed with error -16
[    3.104032] usbcore: registered new interface driver uvcvideo

KiillThemAll, 2020-04-24:
v4l2-ctl --all
Driver Info (not using libv4l2):
	Driver name   : rkisp1
	Card type     : rkisp1
	Bus info      : platform:ff910000.isp
	Driver version: 4.4.132
	Capabilities  : 0x84201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
Priority: 2
Format Video Capture Multiplanar:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV'
	Field             : None
	Number of planes  : 1
	Flags             : 
	Colorspace        : Default
	Transfer Function : Default
	YCbCr Encoding    : Default
	Quantization      : Full Range
	Plane 0           :
	   Bytes per Line : 1280
	   Size Image     : 614400
Crop: Left 0, Top 0, Width 1296, Height 972
Selection: crop, Left 0, Top 0, Width 1296, Height 972
Selection: crop_bounds, Left 0, Top 0, Width 1296, Height 972
Selection: crop, Left 0, Top 0, Width 1296, Height 972
Selection: crop_bounds, Left 0, Top 0, Width 1296, Height 972

User Controls

                       exposure (int)    : min=0 max=1048575 step=1 default=4096 value=4096

Image Source Controls

              vertical_blanking (int)    : min=448 max=448 step=1 default=448 value=448
            horizontal_blanking (int)    : min=600 max=600 step=1 default=600 value=600
                  analogue_gain (int)    : min=0 max=1023 step=1 default=256 value=256

Image Processing Controls

                 link_frequency (intmenu): min=0 max=0 default=0 value=0 flags=read-only
                     pixel_rate (int64)  : min=0 max=0 step=0 default=0 value=67308000 flags=read-only

mdevaev, 2020-04-24:
Так, зайдем с другой стороны. А какой софт с ней вообще работает?

KiillThemAll, 2020-04-24:
Тольк mjpg пробовал

mjpg_streamer -i "input_uvc.so -y -d /dev/video0 -r 640x480" -o "output_http.so -w ./www"
MJPG Streamer Version: git rev: 58e952383cbe973641a3ce6c6a738bafc1605337
i: Using V4L2 device.: /dev/video0
i: Desired Resolution: 640 x 480
i: Frames Per Second.: -1
i: Format............: YUYV
i: JPEG Quality......: 80
i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
o: www-folder-path......: ./www/
o: HTTP TCP port........: 8080
o: HTTP Listen Address..: (null)
o: username:password....: disabled
o: commands.............: enabled

mdevaev, 2020-04-24:
Как я понял, у этой камеры какой-то нестандартный способ захвата изображения. Я нашел описание драйвера: https://dl.vamrs.com/products/rock960/docs/sw/Rockchip%C2%A0Linux%20Camera%C2%A0Developer%20Guide%20V1.1.pdf
Там на 42 странице есть пример того, как можно снимать с него поток. Вероятно, камеру как-то можно настроить, чтобы она отдавала его обычным способом, но я с наскоку не разобрался, как это сделать. Попробуйте изучить этот док. Ну и поищите все же точную модель камеры, вдруг кто-нибудь уже сделал инструкцию.

KiillThemAll, 2020-04-24:
Спасибо большое за информацию, попробую покопать. Если что получится - напишу

mdevaev, 2020-06-02:
Судя по всему, сейчас неактуально. Если нужно - можете переоткрыть этот тикет, а пока что - закрываю.

closed 2020-06-02 by mdevaev