pikvm/ustreamer #22: Instant segfault after upgrading to 1.19

mtlynch, 2020-07-16:
I'm seeing a segfault instantly after loading uStreamer after upgrading to ustreamer v1.19 from 1.1.8:

```bash
$ sudo ./ustreamer --log-level 3
-- INFO  [145.254      main] -- Installing SIGINT handler ...
-- INFO  [145.254      main] -- Installing SIGTERM handler ...
-- INFO  [145.254      main] -- Ignoring SIGPIPE ...
-- DEBUG [145.254      main] -- Increasing picture 0x0x3efe78 buffer: 0 -> 13845 (+13845)
-- INFO  [145.254      main] -- Using internal blank placeholder
-- DEBUG [145.254      main] -- Increasing picture 0x0x3ef380 buffer: 0 -> 13845 (+13845)
-- DEBUG [145.254      main] -- Binding HTTP to [127.0.0.1]:8080 ...
-- INFO  [145.255      main] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [145.255    stream] -- Using V4L2 device: /dev/video0
-- INFO  [145.255    stream] -- Using desired FPS: 4
-- DEBUG [145.255    stream] -- _stream_init_loop: stream->proc->stop=0
================================================================================
-- INFO  [145.255      http] -- Starting HTTP eventloop ...
-- DEBUG [145.387      http] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [145.387      http] -- HTTP: Dropped same frame (BLANK) number 0
-- INFO  [145.492    stream] -- Device fd=8 opened
-- DEBUG [145.492    stream] -- Calling ioctl(VIDIOC_QUERYCAP) ...
-- INFO  [145.492    stream] -- Using input channel: 0
-- INFO  [145.492    stream] -- Using TV standard: DEFAULT
-- DEBUG [145.492    stream] -- Calling ioctl(VIDIOC_S_FMT) ...
-- INFO  [145.493    stream] -- Using resolution: 640x480
-- INFO  [145.493    stream] -- Using pixelformat: YUYV
-- DEBUG [145.493    stream] -- Calling ioctl(VIDIOC_G_PARM) ...
-- INFO  [145.494    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [145.494    stream] -- Using IO method: MMAP
-- DEBUG [145.494    stream] -- Calling ioctl(VIDIOC_REQBUFS) for V4L2_MEMORY_MMAP ...
-- INFO  [145.499    stream] -- Requested 5 HW buffers, got 5
-- DEBUG [145.499    stream] -- Allocating HW buffers ...
-- DEBUG [145.499    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 0 ...
-- DEBUG [145.499    stream] -- Mapping device buffer 0 ...
-- DEBUG [145.500    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 1 ...
-- DEBUG [145.500    stream] -- Mapping device buffer 1 ...
-- DEBUG [145.501    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 2 ...
-- DEBUG [145.501    stream] -- Mapping device buffer 2 ...
-- DEBUG [145.501    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 3 ...
-- DEBUG [145.501    stream] -- Mapping device buffer 3 ...
-- DEBUG [145.502    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 4 ...
-- DEBUG [145.502    stream] -- Mapping device buffer 4 ...
-- DEBUG [145.502    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 0 ...
-- DEBUG [145.502    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 1 ...
-- DEBUG [145.502    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 2 ...
-- DEBUG [145.502    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 3 ...
-- DEBUG [145.502    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 4 ...
-- DEBUG [145.503    stream] -- Allocating picture buffers ...
-- DEBUG [145.503    stream] -- Pre-allocating picture buffer 0 sized 1228800 bytes...
-- DEBUG [145.503    stream] -- Increasing picture 0x0xb5a00768 buffer: 0 -> 1228800 (+1228800)
-- DEBUG [145.503    stream] -- Pre-allocating picture buffer 1 sized 1228800 bytes...
-- DEBUG [145.503    stream] -- Increasing picture 0x0xb5a007a0 buffer: 0 -> 1228800 (+1228800)
-- DEBUG [145.503    stream] -- Pre-allocating picture buffer 2 sized 1228800 bytes...
-- DEBUG [145.503    stream] -- Increasing picture 0x0xb5a007d8 buffer: 0 -> 1228800 (+1228800)
-- DEBUG [145.503    stream] -- Pre-allocating picture buffer 3 sized 1228800 bytes...
-- DEBUG [145.503    stream] -- Increasing picture 0x0xb5a00810 buffer: 0 -> 1228800 (+1228800)
-- DEBUG [145.503    stream] -- Pre-allocating picture buffer 4 sized 1228800 bytes...
-- DEBUG [145.503    stream] -- Increasing picture 0x0xb5a00848 buffer: 0 -> 1228800 (+1228800)
-- DEBUG [145.503    stream] -- Device fd=8 initialized
-- DEBUG [145.503    stream] -- Calling ioctl(VIDIOC_STREAMON) ...
-- INFO  [145.511    stream] -- Capturing started
Segmentation fault
```

I tried getting a backtrace with gdb but it isn't very useful:

```text
(gdb) backtrace
#0  0x0002261c in encoder_prepare ()
#1  0x00030b94 in stream_loop ()
#2  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Is there a way to compile with debug symbols?

## Environment

* Video Capture device: [HDMI to USB dongle](https://twitter.com/Ascii211/status/1268631069051453448)
* OS: Raspberry Pi OS Buster Lite (2012-05-27)

mdevaev, 2020-07-17:
1) Perhaps from my previous build was an old object files, this rarely happens for some reasons. Try `make clean` before building.

2) Run it via `valgrind` and add the options that it will offer.

3) Compile using `make CFLAGS=-g`

mdevaev, 2020-07-17:
4. Install `libc-dbg` or whatever it's called in Raspbian

mtlynch, 2020-07-17:
Ah, it was (1).

I'll update [ansible-role-tinypilot](https://github.com/mtlynch/ansible-role-tinypilot) to `make clean` whenever the source changes.

Thanks!

closed 2020-07-17 by mtlynch

mdevaev, 2020-07-17:
Cool!