pikvm/ustreamer #23: OMX encoder unkillably hangs process after subsequent restarts of ustreamer

kobuki, 2020-08-12:
I'm experiencing a weird error when I use --encoder=omx. ustream hangs and only produces a 'NO SIGNAL' screen. It doesn't stop on CTRL-C, I can only kill it using SIGKILL. Using CPU encoder works, sort of (see my other ticket soon). The error comes up after killing ustreamer after the first run. Only subsequent starts cause the error and software encoder works meanwhile. Resetting the USB capture device somehow solves the issue and ustreamer can run again. The capture device itself works, verified using the following simple streaming command:

`ffmpeg -f v4l2 -r 30 -video_size 1280x720 -i /dev/video0 -c:v mjpeg -qmin 1 -q:v 8 -f nut tcp://10.11.12.101:1234`

On the other end, mpv plays the stream fine.

Setup: Pi 4 4 GB, kernel 5.4.51-v7l+ #1327, Raspbian 10.4

Command line:

`./ustreamer --device=/dev/video0 --format=YUYV --workers=4 --persistent --drop-same-frames=30 --host=0.0.0.0 --port=8080 --resolution=1280x720 --desired-fps=30 --encoder=omx`

Version: 1.20, directly from git

Log of an unsuccessful run: https://pastebin.com/b97G4yTM

mdevaev, 2020-08-12:
What capture device do you use?

kobuki, 2020-08-12:
The device: Bus 002 Device 002: ID 1bcf:2c99 Sunplus Innovation Technology Inc.
This exact model: https://www.aliexpress.com/item/32991659016.html

I can dump more info if needed.

mdevaev, 2020-08-19:
Fixed with #33

closed 2020-08-19 by mdevaev