pikvm/ustreamer #25: error: field has incomplete type 'struct in_addr'

yurivict, 2020-08-13:
```
In file included from src/http/server.c:35:
/usr/include/netinet/ip.h:71:17: error: field has incomplete type 'struct in_addr'
        struct  in_addr ip_src,ip_dst;  /* source and dest address */
                        ^
/usr/include/netinet/ip.h:71:9: note: forward declaration of 'struct in_addr'
        struct  in_addr ip_src,ip_dst;  /* source and dest address */
                ^
/usr/include/netinet/ip.h:71:24: error: field has incomplete type 'struct in_addr'
        struct  in_addr ip_src,ip_dst;  /* source and dest address */
                               ^
```

OS: FreeBSD 12.1
clang-9

mdevaev, 2020-08-14:
Check from master please.

mdevaev, 2020-08-18:
Compiles. The fix will be released in the next release.

Also you can remove [this patch](https://svnweb.freebsd.org/ports/head/multimedia/ustreamer/files/patch-src_tools.h?view=markup). I've applied it to upstream.

closed 2020-08-18 by mdevaev