pikvm/ustreamer #26: ustreamer wont start - Unable to set pixelformat=YUYV, resolution=640x480: Input/output error - constant looping.

changedsoul, 2020-08-15:
So if I just start the pi, I can issue "./ustreamer --host=0.0.0.0" and it starts streaming at 640x480 res and every thing is fine.
I dont know if I am stopping it correctly or not, but I hit CTRL-C to stop the stream, and after that, I am unable to restart the stream unless I reboot.

I keep getting this error: Unable to set pixelformat=YUYV, resolution=640x480: Input/output error.

I feel there is something that is not getting stopped or reset or something, but I dont have any idea what it might be.
Any info would be appreciated.

mdevaev, 2020-08-15:
what video device do you use?

mdevaev, 2020-08-15:
Also give me logs with --debug option for successful and unsuccessful start.

changedsoul, 2020-08-15:
Ok, well my raspberry pi is making a fool of me. I must have performed this more that 5 times with the same result.  Start, Stop, Fail to start, Reboot Pi, Starts fine.
As soon as you responded and I went to get the logs, ustreamer is working as I suppose its intended to. 
I was having problems changing the camera resolution too and now thats working. 
Perhaps the camera driver is acting up.

The camera I am using is:          lsusb :Bus 001 Device 005: ID 045e:0294 Microsoft Corp.  (Its an old xbox web cam)
If it starts happening again, Ill capture the debug and stuff and report back, but presently I tried and cannot get it to fail.

mdevaev, 2020-08-15:
For ustreamer, ctrl+c is the normal way to quit. There are probably problems with the driver, but I will wait for logs if this happens again.

changedsoul, 2020-08-15:
Thank you for responding so quick.

mdevaev, 2020-08-16:
Feel free to reopen if needed.

closed 2020-08-16 by mdevaev