pikvm/ustreamer #27: How to video-capture a device that changes resolutions at runtime without blurred images?

rgl, 2020-08-16:
I'm using [one of the video-capture dongles](https://github.com/rgl/usb-hdmi-video-capture) (similar to the one used/shipped by pikvm) and set the capture resolution with `ustreamer --resolution`, but that resolution is not really the resolution the BIOS/OS ends up using, and as a result, the image I see in the ustream `/stream` endpoint is blurred.

When I match the ustream resolution with the OS resolution (as returned by `fbset` or `xrandr`), it works fine at the OS level, but not when it enters the BIOS, which uses yet another resolution... so I either have a blurred image at the BIOS or at the OS, which is a sub-optimal experience.

Do you known how to fix this? Maybe, somehow query V4L2 (or the video-capture dongle) to known the current resolution of the device and switch to it at runtime? Or force the video-capture dongle to always return a single supported resolution?

Are you/@mtlynch seeing this with the dongles shipped with pikvm?

mdevaev, 2020-08-16:
Hello.

> similar to the one used/**shipped** by pikvm

Pi-KVM does not ship any dongles. Pi-KVM is not TinyPilot and it has nothing to do with kits. That project just uses uStreamer for its own purposes. I wrote uStreamer for my own KVM project: Pi-KVM: https://github.com/pikvm/pikvm.

> Maybe, somehow query V4L2 (or the video-capture dongle) to known the current resolution of the device and switch to it at runtime?

You are right that V4L2 can send a signal about the screen resolution. The only problem is that the dongle does not do this, since it pretends to be a normal webcam with a normal driver. In other words, it is fundamentally impossible to configure the resolution automatically. This is one of the reasons why I don't recommend using the dongle and offer an alternative to it (see the link).

If you are interested in this question in the context of using KVM, then I can recommend that you install my [PI-KVM OS](https://github.com/pikvm/pikvm) instead of TinyPilot. It will run on the same hardware configuration. Pi-KVM provides a menu for switching resolutions in real time in the web interface.

As bonus, you will get a working mouse, a virtual drive, and other things.

rgl, 2020-08-17:
@mdevaev Oh, I'm so sorry, for some reason I've mixed up the two... I meant to say tinypilot. I've fixed the question.

I'm not using a rpi at all. I'm using an Ubuntu 20.04 Desktop to plug the dongle on. I will try with PI-KVM OS on a RPI!

Do you known if the dongle provides other USB protocol/interfaces to control the HDMI besides the normal UVC interface?

I'm now seeing the pikvm hardware that you've got in the repo, and I'm linking it!

With the pikvm hardware none of these blurred images issues happen? It can detect resolution changes and notify the web app?

Can it tell the HDMI device to switch resolution or report a fixed resolution to the device connected to HDMI?

Can it somehow use Consumer Electronics Control (CEC) protocol to control the HDMI device (like the cable set-top box)?

mdevaev, 2020-08-17:
> Do you known if the dongle provides other USB protocol/interfaces to control the HDMI besides the normal UVC interface?

Not as far as I can see.

> With the pikvm hardware none of these blurred images issues happen?

TC358743 (recommended CSI bridge) can understand permission changes and can configure EDID to restrict the list of resolutions. So the image will always be good.

> Can it somehow use Consumer Electronics Control (CEC) protocol to control the HDMI device (like the cable set-top box)?

As I remember no.

mdevaev, 2020-08-18:
There is nothing else to do within this issue.

closed 2020-08-18 by mdevaev