pikvm/ustreamer #28: Is there a way to rotate the image?

PascalHonegger, 2020-08-16:
I've just stumbled upon this project from an issuewithin mjpg-streamer and am really liking the experience so far. I used the available docker image from [beholder-rpa](https://github.com/beholder-rpa/ustreamer-docker) and was running in no time.
I was just wondering if there's a way to flop / rotate the image, as my camera is mounted upside down. Manually calling `v4l2-ctl --set-ctrl horizontal_flip=1` does fix my issue, but it would be nice if this was available as a handy command line argument. I was previously using the [--hf flag from mjpg-streamer](https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_uvc/README.md#usage).

mdevaev, 2020-08-16:
No problem, I can add this flag.

mdevaev, 2020-08-17:
OK, I think I made this feature, but it needs to be tested. None of my cameras support image flipping. Contact me via Discord (the link is in README) and let's test it together using [tmate](https://tmate.io).

mdevaev, 2020-08-17:
Done with #31

closed 2020-08-17 by mdevaev