pikvm/ustreamer #29: Stream stops showing an image / capturing after a while when connected to a video-capture dongle

rgl, 2020-08-16:
I'm using [one of the video-capture dongles](https://github.com/rgl/usb-hdmi-video-capture) (similar to the one used/shipped by pikvm) to capture the HDMI output of device that is running Ubuntu Server but after a while the ustream `/stream` endpoint shows a blank/black screen and never shows anything anymore. It only starts working again after I re-plug the HDMI input from the video-capture dongle.

**NB** the linux console turns off the screen after a while, and it also turns it on when I press a keyboard key, but that did not make the video streaming work again, hence, this cry for help issue.

Is the dongle in "sleep" mode? Maybe there is something that needs to be done to "wake" it up again from the V4L2 side?

Are you/@mtlynch seen this with the dongles shipped with pikvm?

mdevaev, 2020-08-16:
What happens if you restart the streamer?

bbeaudoin, 2020-08-17:
What do you mean "shipped with pikvm?"

bbeaudoin, 2020-08-17:
Similar devices can be found under the brand names "BlueAVS" "MavisLink", "Tengchi", "DigitNow", "HAYOX", "Kedok", and possibly others on Amazon. I bought mine on May 24, 2020 and it appears to be similar to yours:

~~~
Jul 28 23:40:47 pikvm kernel: usb 1-1.4: New USB device found, idVendor=534d, idProduct=2109, bcdDevice=21.00
Jul 28 23:40:47 pikvm kernel: usb 1-1.4: New USB device strings: Mfr=1, Product=0, SerialNumber=0
Jul 28 23:40:47 pikvm kernel: usb 1-1.4: Manufacturer: MACROSILICON
Jul 28 23:40:47 pikvm kernel: hid-generic 0003:534D:2109.0001: hiddev96,hidraw0: USB HID v1.10 Device [MACROSILICON] on usb-0000:01:00.0-1.4/input4
~~~

I haven't seen mine "sleep". However, I needed to update my Raspberry Pi 4's firmware to improve USB power management. Raspbian and other OS'es will power off PCIe devices, including USB, using ACPI (Advanced Configuration and Power Interface) and, if the device does not support being powered off.

Instructions for updating the Raspberry Pi 4 EEPROM can be found here:
[https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md](https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md)

If that does not fix the power management issues with your device, you may want to contact the seller of the device for exchange or refund. I've read @mtlynch 's page and understand there are a lot of devices on the market based on the same chipset but the prices and quality may vary greatly between manufacturers and resellers.
 
[image]

mdevaev, 2020-08-17:
@bbeaudoin Do you use this with Pi-KVM?

rgl, 2020-08-17:
> What do you mean "shipped with pikvm?"

I'm sorry, I meant to say tinypilot.

mdevaev, 2020-08-17:
@rgl 

OK, all the described problems are [hardware-based](https://github.com/pikvm/pikvm#a-few-words-about-hdmi-usb-dongle).

My only advice is to try using something like this to reboot the USB port instead of reconnecting the cable. Will this work?
https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line

bbeaudoin, 2020-08-17:
@mdevaev I do use that dongle with PiKVM; no loss of connectivity seen over days of uninterrupted testing. I agree the issue sounds like something with the hardware, from what I've seen of uStreamer it doesn't support specific hardware directly, only software support provided by v4l2 compatible drivers packaged with the OS or the Linux Kernel project itself.

mdevaev, 2021-03-07:
Okay, apparently the question is already irrelevant, and the hardware problems are not directly related to ustreamer.

closed 2021-03-07 by mdevaev