pikvm/ustreamer #30: chrome memory leak when leaving the /stream endpoint open

rgl, 2020-08-16:
When running in Chrome `Version 84.0.4147.125 (Official Build) (64-bit)` (Ubuntu 20.04 Desktop) the memory footprint of the `/stream` endpoint is always increasing. Here's a screenshot of it using 2.5GB of memory:

[image]

I'm running ustream as:

```bash
./ustreamer \
    --format jpeg \
    --resolution 1280x720 \
    --desired-fps 15 \
    --drop-same-frames 5 \
    --slowdown
```

It's working fine in Firefox 79. Never uses more than 500KB of memory.

mdevaev, 2020-08-16:
Well, this is an obvious Chrome bug. I can add workaround to pikvm/kvmd. Moreover, it seems I can't make a workaround for this, because even reloading the page with the stream doesn't help. If you report an error to Google, I will be very grateful: https://bugs.chromium.org/p/chromium/issues/list

mdevaev, 2020-08-16:
How long did you keep streaming that it ate up so much memory?

LewisSpring, 2021-06-09:
I'm not sure if RGL ever reported the issue to the Chromium Project, but while I'm MJPEG streaming in Edge the GPU Process memory usage increases up to somewhere around 1,200,000K, and then goes back down when streaming stops.

My options: `--host=0.0.0.0 --port=8080 --device-timeout=5 -f 15 -l -q 80 -r 1920x1080 --format=mjpeg`

rgl, 2021-06-09:
Sorry, I did not report this upstream :-(

LewisSpring, 2021-06-09:
> Sorry, I did not report this upstream :-(

No worries. It's a slightly different issue from a while ago anyway.

mdevaev, 2021-06-09:
This is not a ustreamer problem and I can't fix anything here. This is a memory leak in the browser. Please report this issue to the chrome/blink developers, as I don't even have this issue reproducing.

mdevaev, 2022-07-07:
Is it still relevant?

mdevaev, 2022-07-20:
Probably no

closed 2022-07-20 by mdevaev