pikvm/ustreamer #32: Official Docker images

PascalHonegger, 2020-08-17:
Given the popularity of docker it would be really nice if you're able to simply run ustreamer with a one-line docker command without installing anything. Here are a few points which would have to be done in order to achieve said goal:

- [ ] Create Dockerfile / .dockerignore
- [ ] Ensure Dockerfile supports multiple architectures (amd64, armv7, ...)
- [ ] Automatically build and tag new versions, [for example using GitHub Actions and buildx](https://www.docker.com/blog/multi-arch-build-and-images-the-simple-way/ )
- [ ] Add tagged build with Raspberry Pi OMX / GPIO support
- [ ] Update documentation

mdevaev, 2020-09-22:
The latest versions of ustreamer use libgpiod, which is available on all platforms include x86_64.

badsmoke, 2021-01-30:
in the meantime i have already built a docker container for myself, for all who are interested

"https://hub.docker.com/repository/docker/badsmoke/ustreamer"

arch = amd64

tomaszduda23, 2022-11-13:
@mdevaev do you have any plan to make docker image public on dockerhub?

mdevaev, 2022-11-13:
What do I need to do for this?

tomaszduda23, 2022-11-13:
It seems that image is uploaded when you create new tag but repository is private on dockerhub. You can change it in repository settings 'make public'. Do you use pikvm/ustreamer in dockerhub?

mdevaev, 2022-11-14:
Hm, it was pushed to https://hub.docker.com/repository/docker/mdevaev/ustreamer

tomaszduda23, 2022-11-14:
https://github.com/pikvm/ustreamer/commit/1ff021d9aca93fa9eb4175e600909cf1a8315ea5 this could be changed here. Do you own pikvm name? I'm not able to check who is the owner of that.

mdevaev, 2022-11-14:
Yes, I don't know why it's not working :/

tomaszduda23, 2022-11-14:
secrets.DOCKERHUB_USERNAME <- the name is taken from this so you can change it or merge https://github.com/pikvm/ustreamer/pull/189. I'm not able to test it myself though.

mdevaev, 2022-12-27:
https://hub.docker.com/r/pikvm/ustreamer/tags

closed 2022-12-27 by mdevaev