pikvm/ustreamer #39: build fails on FreeBSD; SYS_gettid is Linux-specific

emaste, 2020-09-01:
Attempting to build on FreeBSD I get:
```
-- CC src/device.c
In file included from src/device.c:43:
In file included from src/logging.h:37:
src/threading.h:105:67: error: use of undeclared identifier 'SYS_gettid'
  ...assert(snprintf(name, MAX_THREAD_NAME, "tid=%d", (pid_t)syscall(SYS_gett...
                                                                     ^
```
Worked around by replacing with [thr_self(2)](https://www.freebsd.org/cgi/man.cgi?query=thr_self):
```
diff --git a/src/threading.h b/src/threading.h
index 5cc759d..448b751 100644
--- a/src/threading.h
+++ b/src/threading.h
@@ -34,6 +34,7 @@
 #      if defined(__FreeBSD__) || defined(__OpenBSD__) || defined(__DragonFly__)
 #              include <pthread_np.h>
 #              include <sys/param.h>
+#              include <sys/thr.h>
 #      endif
 #endif
 
@@ -102,7 +103,10 @@ INLINE void thread_get_name(char *name) { // Always required for logging
 #      endif
        if (retval < 0) {
 #endif
-               assert(snprintf(name, MAX_THREAD_NAME, "tid=%d", (pid_t)syscall(SYS_gettid)) > 0);
+               long tid;
+
+               thr_self(&tid);
+               assert(snprintf(name, MAX_THREAD_NAME, "tid=%ld", tid) > 0);
 #ifdef WITH_PTHREAD_NP
        }
 #endif
```

If desired I'm happy to clean this up with more `#ifdef`ery

emaste, 2020-09-01:
The two changes I made to build on FreeBSD are in https://github.com/emaste/ustreamer

mdevaev, 2020-09-01:
Hi! Good catch. What do you think about this? https://github.com/pikvm/ustreamer/commit/deb37986b67c6eca1f2138a4c47d2f91010a0063

emaste, 2020-09-01:
Works for me

closed 2020-09-01 by emaste