pikvm/ustreamer #40: Usage with Raspberry Pi camera?

pixelspark, 2020-09-13:
Hi,

Any chance this can be made to work with the Raspberry Pi's camera? (I am using the V1.0). I have the bcm2835-v4l2 kernel modules installed (so there is a video0 device). Yet:

````
# ustreamer --host 0.0.0.0 --port 1234 -b 3 -f 60
-- INFO  [198.908   tid=230] -- Installing SIGINT handler ...
-- INFO  [198.909   tid=230] -- Installing SIGTERM handler ...
-- INFO  [198.909   tid=230] -- Ignoring SIGPIPE ...
-- INFO  [198.909   tid=230] -- Using internal blank placeholder
-- INFO  [198.914   tid=230] -- Listening HTTP on [0.0.0.0]:1234
-- INFO  [198.914   tid=233] -- Starting HTTP eventloop ...
-- INFO  [198.915   tid=232] -- Using V4L2 device: /dev/video0
-- INFO  [198.915   tid=232] -- Using desired FPS: 60
================================================================================
-- INFO  [198.915   tid=232] -- Device fd=8 opened
-- INFO  [198.916   tid=232] -- Using input channel: 0
-- INFO  [198.916   tid=232] -- Using TV standard: DEFAULT
-- INFO  [198.918   tid=232] -- Using resolution: 640x480
-- INFO  [198.918   tid=232] -- Using pixelformat: YUYV
-- INFO  [198.919   tid=232] -- Using HW FPS: 60
-- INFO  [198.919   tid=232] -- Using IO method: MMAP
-- INFO  [198.922   tid=232] -- Requested 3 HW buffers, got 3
-- ERROR [198.922   tid=232] -- Can't VIDIOC_QUERYBUF: Not a tty
-- INFO  [198.924   tid=232] -- Device fd=8 closed
-- INFO  [198.924   tid=232] -- Sleeping 1 seconds before new stream init ...
-- INFO  [199.590   tid=230] -- ===== Stopping by SIGINT =====
-- INFO  [199.591   tid=233] -- HTTP eventloop stopped
-- INFO  [199.925   tid=230] -- Bye-bye
````

This happens for every combination of frame rates, buffer counts I tried, and also with `-I USERPTR`. Any ideas on what I am missing, or is this simply not supported?

mdevaev, 2020-09-13:
Hi. Ustreamer does not contain any special code for working with raspicam. There's probably something wrong with the driver. it may have some parameters that should be specified. Alas, I can't tell you how to set this up properly, since I don't have a suitable camera at hand.

Show me `v4l2-ctl --list-formats-ext`

mdevaev, 2020-09-28:
Boop

pixelspark, 2020-09-29:
Sorry @mdevaev, I did not yet have a chance to look into this. I am running on a bare kernel with busybox (i.e. would have to compile v4l2-ctl myself statically). I will try with a Raspbian install soon.

mdevaev, 2020-09-29:
Okay, keep me posted. I'll try to help you.

pixelspark, 2020-09-29:
````
pi@raspberrypi:~ $  v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YU12' (Planar YUV 4:2:0)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[2]: 'RGB3' (24-bit RGB 8-8-8)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[3]: 'JPEG' (JFIF JPEG, compressed)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[4]: 'H264' (H.264, compressed)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[5]: 'MJPG' (Motion-JPEG, compressed)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[6]: 'YVYU' (YVYU 4:2:2)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[7]: 'VYUY' (VYUY 4:2:2)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[8]: 'UYVY' (UYVY 4:2:2)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[9]: 'NV12' (Y/CbCr 4:2:0)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[10]: 'BGR3' (24-bit BGR 8-8-8)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[11]: 'YV12' (Planar YVU 4:2:0)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[12]: 'NV21' (Y/CrCb 4:2:0)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[13]: 'BGR4' (32-bit BGRA/X 8-8-8-8)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
```

mdevaev, 2020-09-29:
Try `ustreamer --host 0.0.0.0 --port 1234 -m jpeg -r 2592x1944`

pixelspark, 2020-09-29:
On the same Raspberry:

````
pi@raspberrypi:~ $ ./ustreamer --host 0.0.0.0 --port 1234 -m jpeg -r 2592x1944
-- INFO  [5159.038   tid=672] -- Installing SIGINT handler ...
-- INFO  [5159.041   tid=672] -- Installing SIGTERM handler ...
-- INFO  [5159.044   tid=672] -- Ignoring SIGPIPE ...
-- INFO  [5159.047   tid=672] -- Using internal blank placeholder
-- INFO  [5159.048   tid=672] -- Listening HTTP on [0.0.0.0]:1234
-- INFO  [5159.050   tid=674] -- Starting HTTP eventloop ...
-- INFO  [5159.053   tid=673] -- Using V4L2 device: /dev/video0
-- INFO  [5159.055   tid=673] -- Using desired FPS: 0
================================================================================
-- INFO  [5159.058   tid=673] -- Device fd=8 opened
-- INFO  [5159.061   tid=673] -- Using input channel: 0
-- INFO  [5159.062   tid=673] -- Using TV standard: DEFAULT
-- INFO  [5159.066   tid=673] -- Using resolution: 2592x1944
-- INFO  [5159.066   tid=673] -- Using pixelformat: JPEG
-- INFO  [5159.069   tid=673] -- Using HW FPS: 0 -> 90 (coerced)
-- INFO  [5159.072   tid=673] -- Using IO method: MMAP
-- INFO  [5159.090   tid=673] -- Requested 2 HW buffers, got 2
-- ERROR [5159.092   tid=673] -- Can't VIDIOC_QUERYBUF: Not a tty
-- INFO  [5159.106   tid=673] -- Device fd=8 closed
-- INFO  [5159.108   tid=673] -- Sleeping 1 seconds before new stream init ...
================================================================================
-- INFO  [5160.114   tid=673] -- Device fd=8 opened
-- INFO  [5160.116   tid=673] -- Using input channel: 0
-- INFO  [5160.118   tid=673] -- Using TV standard: DEFAULT
-- INFO  [5160.126   tid=673] -- Using resolution: 2592x1944
-- INFO  [5160.128   tid=673] -- Using pixelformat: JPEG
-- INFO  [5160.135   tid=673] -- Using HW FPS: 0 -> 90 (coerced)
-- INFO  [5160.135   tid=673] -- Using IO method: MMAP
-- INFO  [5160.152   tid=673] -- Requested 2 HW buffers, got 2
-- ERROR [5160.155   tid=673] -- Can't VIDIOC_QUERYBUF: Not a tty
-- INFO  [5160.167   tid=673] -- Device fd=8 closed
-- INFO  [5160.170   tid=673] -- Sleeping 1 seconds before new stream init ...
================================================================================
-- INFO  [5161.175   tid=673] -- Device fd=8 opened
-- INFO  [5161.177   tid=673] -- Using input channel: 0
-- INFO  [5161.179   tid=673] -- Using TV standard: DEFAULT
-- INFO  [5161.188   tid=673] -- Using resolution: 2592x1944
-- INFO  [5161.188   tid=673] -- Using pixelformat: JPEG
-- INFO  [5161.190   tid=673] -- Using HW FPS: 0 -> 90 (coerced)
-- INFO  [5161.192   tid=673] -- Using IO method: MMAP
-- INFO  [5161.210   tid=673] -- Requested 2 HW buffers, got 2
-- ERROR [5161.213   tid=673] -- Can't VIDIOC_QUERYBUF: Not a tty
-- INFO  [5161.224   tid=673] -- Device fd=8 closed
-- INFO  [5161.226   tid=673] -- Sleeping 1 seconds before new stream init ...
^C-- INFO  [5162.198   tid=672] -- ===== Stopping by SIGINT =====
-- INFO  [5162.202   tid=674] -- HTTP eventloop stopped
-- INFO  [5162.231   tid=672] -- Bye-bye
pi@raspberrypi:~ $ 
````

mdevaev, 2020-09-29:
Check dmesg please. Might be related with raspberrypi/linux#3824

pixelspark, 2020-09-30:
Hm, I am not seeing any problematic messages there. Here are some init messages:

````
[   14.999375] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[   15.021244] mc: Linux media interface: v0.10
[   15.025619] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   15.025661] [vc_sm_connected_init]: start
[   15.039113] [vc_sm_connected_init]: installed successfully
[   15.272745] videodev: Linux video capture interface: v2.00
[   15.391971] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   15.395795] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   15.397418] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   15.399029] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   15.454221] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   15.457886] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[   15.464684] bcm2835_audio bcm2835_audio: card created with 8 channels
[   15.498868] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video13
[   15.512420] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   16.116721] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
[   16.127057] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
[   16.127410] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   16.127525] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   16.128659] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video16
[   16.128717] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[   16.128755] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[   16.128785] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[   16.128815] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[   16.140866] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[   16.164749] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   16.164837] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   16.222127] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   16.222229] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   16.282026] bcm2835-v4l2: scene mode selected 0, was 0
[   16.283553] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 1280x720
[   16.358567] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
````

No new messages appeared while ustreamer was running... Also, not seeing any kernel oopses like in https://github.com/raspberrypi/linux/issues/3824.

pixelspark, 2020-09-30:
Note that the device I am interested in appears to be `video0` (the others are there even when the camera is not enabled in `raspi-config` so presumably have to do something with media encoding/decoding). Dmesg appears to note it is in "stills mode > 1280x720" which may have something to do with this issue?

mdevaev, 2020-10-05:
Just for information: I haven't forgotten about this issue. I found a camera among my junk and I'll try to figure it out.

pixelspark, 2020-10-11:
Any ideas yet? I am not too well versed in V4L but I do know C, so if there is anything you need me to test, let me know!

mdevaev, 2020-10-11:
Thanks. I'm sorry I took so long with this. It's just that I'm the only developer of pikvm and a lot of things require my attention. I will try to deal with the camera in the next few days.

pixelspark, 2020-10-11:
Hey no worries, I am super grateful already that you made this project! 👍

pixelspark, 2020-10-11:
See here for an example use of the Pi cam using V4L2 that does appear to work: https://github.com/climberhunt/uvc-gadget/blob/master/uvc-gadget.c

mdevaev, 2020-10-11:
I ran through the code and it seems that the only thing that makes it significantly different from mine is the additional grab method: https://github.com/climberhunt/uvc-gadget/blob/master/uvc-gadget.c#L840

mdevaev, 2020-10-28:
I connected my camera and everything worked for me.
1. I took a fresh raspbian distribution and turned on the camera using [raspi-config](https://www.raspberrypi.org/documentation/configuration/camera.md) and rebooted.
2. Next, I followed this manual: https://github.com/raspberrypi/linux/blob/rpi-3.10.y/Documentation/video4linux/bcm2835-v4l2.txt
```
# Plz note: 1088, not 1080
v4l2-ctl  --set-fmt-video=width=1920,height=1088,pixelformat=4
ustreamer --host :: -m jpeg --device-timeout=5
```

pixelspark, 2020-11-01:
> I connected my camera and everything worked for me.
>
> 1. I took a fresh raspbian distribution and turned on the camera using [...]

Okay, this is strange... I am seeing the following (Raspberry Pi Zero W):

````sh
pi@raspberrypi:~ $ v4l2-ctl  --set-fmt-video=width=1920,height=1088,pixelformat=4
pi@raspberrypi:~ $ ./ustreamer --host :: -m jpeg --device-timeout=5 -r 1920x1088
-- INFO  [392.637   tid=725] -- Installing SIGINT handler ...
-- INFO  [392.639   tid=725] -- Installing SIGTERM handler ...
-- INFO  [392.644   tid=725] -- Ignoring SIGPIPE ...
-- INFO  [392.644   tid=725] -- Using internal blank placeholder
-- INFO  [392.645   tid=725] -- Listening HTTP on [::]:8080
-- INFO  [392.647   tid=727] -- Starting HTTP eventloop ...
-- INFO  [392.651   tid=726] -- Using V4L2 device: /dev/video0
-- INFO  [392.653   tid=726] -- Using desired FPS: 0
================================================================================
-- INFO  [392.653   tid=726] -- Device fd=8 opened
-- INFO  [392.655   tid=726] -- Using input channel: 0
-- INFO  [392.657   tid=726] -- Using TV standard: DEFAULT
-- INFO  [392.662   tid=726] -- Using resolution: 1920x1088
-- INFO  [392.662   tid=726] -- Using pixelformat: JPEG
-- INFO  [392.665   tid=726] -- Using HW FPS: 0 -> 90 (coerced)
-- INFO  [392.666   tid=726] -- Using IO method: MMAP
-- INFO  [392.676   tid=726] -- Requested 2 HW buffers, got 2
-- ERROR [392.678   tid=726] -- Can't VIDIOC_QUERYBUF: Not a tty
-- INFO  [392.685   tid=726] -- Device fd=8 closed
-- INFO  [392.687   tid=726] -- Sleeping 1 seconds before new stream init ...
^C-- INFO  [393.256   tid=725] -- ===== Stopping by SIGINT =====
-- INFO  [393.257   tid=727] -- HTTP eventloop stopped
-- INFO  [393.692   tid=725] -- Bye-bye
pi@raspberrypi:~ $ v4l2-ctl --stream-mmap=3 --stream-count=100 --stream-to=somefile.h264
KKKKKKKKKKKKKKKKKKKKKKKKKKKKKKKKKKKKK^C
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
````

So v4l2-ctl streams just fine (also, raspistill works) but even after following the instructions, ustreamer throws the same error... any thoughts?

pixelspark, 2020-11-01:
Same thing on a custom Linux distribution using Raspberry Pi's firmware from [https://github.com/raspberrypi/firmware](here):

````
/boot # uname -a
Linux (none) 5.4.73+ #1360 Thu Oct 29 15:48:27 GMT 2020 armv6l GNU/Linux
/boot # lsmod
brcmfmac 286720 0 - Live 0xbf375000
brcmutil 20480 1 brcmfmac, Live 0xbf36c000
bluetooth 376832 0 - Live 0xbf2fb000
cfg80211 667648 1 brcmfmac, Live 0xbf219000
lib80211 16384 0 - Live 0xbf211000
rfkill 28672 5 bluetooth,cfg80211, Live 0xbf204000
sha256_generic 16384 0 - Live 0xbf1fc000
sha512_generic 20480 0 - Live 0xbf1f3000
aes_generic 40960 0 - Live 0xbf1e5000
libsha256 20480 1 sha256_generic, Live 0xbf1dd000
ecdh_generic 16384 1 bluetooth, Live 0xbf1d6000
ecc 40960 1 ecdh_generic, Live 0xbf1c8000
libaes 16384 2 bluetooth,aes_generic, Live 0xbf1c1000
bcm2835_v4l2 45056 0 - Live 0xbf1b0000 (C)
videobuf2_v4l2 28672 1 bcm2835_v4l2, Live 0xbf1a4000
videobuf2_common 53248 2 bcm2835_v4l2,videobuf2_v4l2, Live 0xbf190000
videobuf2_vmalloc 16384 1 bcm2835_v4l2, Live 0xbf189000
videobuf2_memops 16384 1 videobuf2_vmalloc, Live 0xbf182000
videodev 225280 3 bcm2835_v4l2,videobuf2_v4l2,videobuf2_common, Live 0xbf138000
bcm2835_mmal_vchiq 28672 1 bcm2835_v4l2, Live 0xbf12d000 (C)
vc_sm_cma 32768 1 bcm2835_mmal_vchiq, Live 0xbf11f000 (C)
mc 45056 3 videobuf2_v4l2,videobuf2_common,videodev, Live 0xbf10c000
snd_bcm2835 28672 0 - Live 0xbf100000 (C)
snd_pcm 94208 1 snd_bcm2835, Live 0xbf0dd000
snd_timer 32768 1 snd_pcm, Live 0xbf0cf000
snd 73728 3 snd_bcm2835,snd_pcm,snd_timer, Live 0xbf0b3000
pwm_bcm2835 16384 0 - Live 0xbf0ab000
g_serial 16384 0 - Live 0xbf0a3000
usb_f_serial 16384 0 - Live 0xbf09b000
usb_f_acm 16384 1 - Live 0xbf093000
u_serial 20480 4 usb_f_serial,usb_f_acm, Live 0xbf08a000
libcomposite 57344 3 g_serial,usb_f_serial,usb_f_acm, Live 0xbf074000
dwc2 163840 0 - Live 0xbf042000
udc_core 53248 4 usb_f_acm,u_serial,libcomposite,dwc2, Live 0xbf02c000
i2c_bcm2835 16384 0 - Live 0xbf024000
i2c_dev 16384 0 - Live 0xbf01c000
spi_bcm2835aux 16384 0 - Live 0xbf014000
spi_bcm2835 24576 0 - Live 0xbf00a000
spidev 20480 0 - Live 0xbf000000
````

mdevaev, 2020-11-01:
Try `./ustreamer --host :: -m jpeg --device-timeout=5 --buffers=3`

pixelspark, 2020-11-01:
Same thing on the custom distro, will try later on Raspbian:

````
/boot # ustreamer --host 0.0.0.0 -m jpeg --device-timeout=5 --buffers=3
-- INFO  [76.425   tid=382] -- Installing SIGINT handler ...
-- INFO  [76.425   tid=382] -- Installing SIGTERM handler ...
-- INFO  [76.425   tid=382] -- Ignoring SIGPIPE ...
-- INFO  [76.426   tid=382] -- Using internal blank placeholder
-- INFO  [76.430   tid=382] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [76.431   tid=385] -- Starting HTTP eventloop ...
-- INFO  [76.431   tid=384] -- Using V4L2 device: /dev/video0
-- INFO  [76.431   tid=384] -- Using desired FPS: 0
================================================================================
-- INFO  [76.432   tid=384] -- Device fd=8 opened
-- INFO  [76.432   tid=384] -- Using input channel: 0
-- INFO  [76.433   tid=384] -- Using TV standard: DEFAULT
-- INFO  [76.435   tid=384] -- Using resolution: 640x480
-- INFO  [76.436   tid=384] -- Using pixelformat: JPEG
-- INFO  [76.437   tid=384] -- Using HW FPS: 0 -> 90 (coerced)
-- INFO  [76.437   tid=384] -- Using IO method: MMAP
-- INFO  [76.439   tid=384] -- Requested 3 HW buffers, got 3
-- ERROR [76.439   tid=384] -- Can't VIDIOC_QUERYBUF: Not a tty
-- INFO  [76.441   tid=384] -- Device fd=8 closed
-- INFO  [76.442   tid=384] -- Sleeping 1 seconds before new stream init ...
================================================================================
-- INFO  [77.442   tid=384] -- Device fd=8 opened
-- INFO  [77.443   tid=384] -- Using input channel: 0
-- INFO  [77.443   tid=384] -- Using TV standard: DEFAULT
-- INFO  [77.446   tid=384] -- Using resolution: 640x480
-- INFO  [77.446   tid=384] -- Using pixelformat: JPEG
-- INFO  [77.447   tid=384] -- Using HW FPS: 0 -> 90 (coerced)
-- INFO  [77.447   tid=384] -- Using IO method: MMAP
-- INFO  [77.449   tid=384] -- Requested 3 HW buffers, got 3
-- ERROR [77.449   tid=384] -- Can't VIDIOC_QUERYBUF: Not a tty
-- INFO  [77.450   tid=384] -- Device fd=8 closed
-- INFO  [77.450   tid=384] -- Sleeping 1 seconds before new stream init ...
-- INFO  [77.975   tid=382] -- ===== Stopping by SIGINT =====
-- INFO  [77.976   tid=385] -- HTTP eventloop stopped
-- INFO  [78.451   tid=382] -- Bye-bye

````

pixelspark, 2020-11-01:
Raspbian same thing:

````
pi@raspberrypi:~ $ v4l2-ctl  --set-fmt-video=width=1920,height=1088,pixelformat=4
pi@raspberrypi:~ $ ./ustreamer --host :: -m jpeg --device-timeout=5 --buffers=3
-- INFO  [133.232   tid=549] -- Installing SIGINT handler ...
-- INFO  [133.234   tid=549] -- Installing SIGTERM handler ...
-- INFO  [133.236   tid=549] -- Ignoring SIGPIPE ...
-- INFO  [133.238   tid=549] -- Using internal blank placeholder
-- INFO  [133.244   tid=549] -- Listening HTTP on [::]:8080
-- INFO  [133.246   tid=551] -- Starting HTTP eventloop ...
-- INFO  [133.248   tid=550] -- Using V4L2 device: /dev/video0
-- INFO  [133.251   tid=550] -- Using desired FPS: 0
================================================================================
-- INFO  [133.251   tid=550] -- Device fd=8 opened
-- INFO  [133.257   tid=550] -- Using input channel: 0
-- INFO  [133.257   tid=550] -- Using TV standard: DEFAULT
-- INFO  [133.262   tid=550] -- Using resolution: 640x480
-- INFO  [133.263   tid=550] -- Using pixelformat: JPEG
-- INFO  [133.266   tid=550] -- Using HW FPS: 0 -> 90 (coerced)
-- INFO  [133.266   tid=550] -- Using IO method: MMAP
-- INFO  [133.271   tid=550] -- Requested 3 HW buffers, got 3
-- ERROR [133.273   tid=550] -- Can't VIDIOC_QUERYBUF: Not a tty
-- INFO  [133.278   tid=550] -- Device fd=8 closed
-- INFO  [133.279   tid=550] -- Sleeping 1 seconds before new stream init ...
````

mdevaev, 2020-11-01:
Could you try https://github.com/jacksonliam/mjpg-streamer with input_uvc plugin? If it works then I have an error in ustreamer.

fred2nice, 2020-12-27:
Hi,

I am trying to stream from Raspberry Cam HQ from Raspberry Pi Zero W
It works fine with ./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=80 -r 1280x720 --image-default  -q 100 -c omx -l

I tried with  1920x1080 / 1640x1232 I obtained  :
- ERROR [757.413    stream] -- Mainloop select() error: Resource temporarily unavailable

Is there a width / height limit? 

PS : it works fine with raspivid & gst
w=1920 / h=1080
raspivid -t 0 -b 2097152 -w $w -h $h -fps 30 -n -o - | gst-launch-1.0 fdsrc ....

Kind Regards.

fred2nice, 2020-12-27:
I tried with :
./ustreamer --host=0.0.0.0 --port=80 device=/dev/video0 -r 1920x1080 --format JPEG --buffers 2 --workers 2 --slowdown --persistent --device-timeout 16 --tcp-nodelay --allow-origin *

no error, but the pixel size is too big, as you can see on the screenshot ( a pixel is about 14x14 or 12x12 ) 

[image]

mdevaev, 2021-01-01:
Try this: `v4l2-ctl  --set-fmt-video=width=1920,height=1088,pixelformat=4`

webgps, 2021-01-26:
> Try this: `v4l2-ctl --set-fmt-video=width=1920,height=1088,pixelformat=4`

same thing...
[image]

mdevaev, 2021-01-27:
Apparently it depends on the revision of the camera. I can't do anything here from the ustreamer side.

Drachenkaetzchen, 2021-02-09:
I can confirm that there are issues with the rpi cameras and the v4l2 interface. My camera is working fine and without issues when using `mjpeg-streamer`.

With non-native resolutions like `640x480` or `1920x1088` (which is never explained why there's an 8 pixel difference) it works with my camera, but with the native resolution I receive a message that the device is disconnected. During my testing, the bcm v4l driver did at least crash once.

Furthermore, the follwing example resulted in a zero byte file:

```
$ v4l2-ctl  --set-fmt-video=width=2592,height=1944,pixelformat=3
$ v4l2-ctl --stream-mmap=3 --stream-count=1 --stream-to=somefile.jpg
```

I'm using a fairly recent kernel from last week. I'll try to investigate, but no warranties.

Drachenkaetzchen, 2021-02-11:
I decided to try on a Rpi4 for comparison. Observations:

- When using the JPEG capture example (see below), it works and the jpg file is properly created
- If I run `./ustreamer --host :: -m jpeg --device-timeout=5 --buffers=3`, `ustreamer` works in `640x480` (output below). I also can run the capture example without problems after shutting down `ustreamer`.
- I then run `./ustreamer --host :: -m jpeg --device-timeout=5 --buffers=3 -r 2592x1944`, which fails with `Mainloop select() error: Inappropriate ioctl for device` (full log below). After that, I can no longer use the capture example as streaming to file simply hangs now.

##  Capture Example 
This is the example capture from [bcm2835-v4l2.txt](https://github.com/raspberrypi/linux/blob/rpi-r.9.y/Documentation/video4linux/bcm2835-v4l2.txt)
```
$ v4l2-ctl  --set-fmt-video=width=2592,height=1944,pixelformat=3
$ v4l2-ctl --stream-mmap=3 --stream-count=1 --stream-to=somefile.jpg
```

## Output from successful run at 640x480

```
 ./ustreamer --host :: -m jpeg --device-timeout=5 --buffers=3
-- INFO  [209.809      main] -- Using internal blank placeholder
-- INFO  [209.810      main] -- Listening HTTP on [::]:8080
-- INFO  [209.810    stream] -- Using V4L2 device: /dev/video0
-- INFO  [209.810    stream] -- Using desired FPS: 0
-- INFO  [209.810      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [209.812    stream] -- Device fd=8 opened
-- INFO  [209.812    stream] -- Using input channel: 0
-- INFO  [209.818    stream] -- Using resolution: 640x480
-- INFO  [209.818    stream] -- Using pixelformat: JPEG
-- INFO  [209.819    stream] -- Using HW FPS: 0 -> 90 (coerced)
-- ERROR [209.819    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [209.819    stream] -- Using IO method: MMAP
-- INFO  [209.821    stream] -- Requested 3 device buffers, got 3
-- INFO  [210.254    stream] -- Capturing started
-- INFO  [210.254    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [210.254    stream] -- Using JPEG quality: encoder default
-- INFO  [210.254    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [210.255    stream] -- Capturing ...
-- INFO  [221.855      http] -- HTTP: Registered client: [::ffff:172.22.117.159]:51646, id=419a5422-4a9e-41f5-968e-c59103ff8727; clients now: 1
```

## Output from unsuccessful run at `2592x1944`

```./ustreamer --host :: -m jpeg --device-timeout=5 --buffers=3 -r 2592x1944
-- INFO  [891.435      main] -- Using internal blank placeholder
-- INFO  [891.436      main] -- Listening HTTP on [::]:8080
-- INFO  [891.437    stream] -- Using V4L2 device: /dev/video0
-- INFO  [891.437    stream] -- Using desired FPS: 0
-- INFO  [891.437      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [891.438    stream] -- Device fd=8 opened
-- INFO  [891.439    stream] -- Using input channel: 0
-- INFO  [891.445    stream] -- Using resolution: 2592x1944
-- INFO  [891.445    stream] -- Using pixelformat: JPEG
-- INFO  [891.445    stream] -- Using HW FPS: 0 -> 90 (coerced)
-- ERROR [891.445    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [891.445    stream] -- Using IO method: MMAP
-- INFO  [891.478    stream] -- Requested 3 device buffers, got 3
-- INFO  [891.977    stream] -- Capturing started
-- INFO  [891.977    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [891.977    stream] -- Using JPEG quality: encoder default
-- INFO  [891.977    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [891.977    stream] -- Capturing ...
-- ERROR [896.982    stream] -- Mainloop select() error: Inappropriate ioctl for device
-- INFO  [896.982    stream] -- Destroying workers pool JPEG ...
-- INFO  [897.006    stream] -- Capturing stopped
-- INFO  [897.018    stream] -- Device fd=8 closed
```

Drachenkaetzchen, 2021-02-11:
I was able to reproduce the hang with the streaming capture example above. As soon as the pixel format gets set to MJPG, the driver hangs with full resolution.

How to reproduce:
```
v4l2-ctl  --set-fmt-video=width=2592,height=1944,pixelformat=5
v4l2-ctl --stream-mmap=3 --stream-count=1 --stream-to=somefile.jpg
```

@mdevaev I noticed that in [device.c](https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/device.c#L44) that both `V4L2_PIX_FMT_MJPEG` and `V4L2_PIX_FMT_JPEG` are both labelled as `JPEG`, is this a mistake?

Drachenkaetzchen, 2021-02-11:
> I noticed that in [device.c](https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/device.c#L44) that both `V4L2_PIX_FMT_MJPEG` and `V4L2_PIX_FMT_JPEG` are both labelled as `JPEG`, is this a mistake?

In fact I have now changed one label from `JPEG` to `MJPEG` and surprise - `ustreamer` now works without a problem with the high resolution! 

I do not know if this solves the problem of @webgps and @fred2nice, but I have created a pull request. Can you both try this out?

mdevaev, 2021-02-11:
@Drachenkaetzchen I am shocked by this solution. I didn't think that was the case. You're doing great! I'll have to check if it's causing any other problems, and then I'll merge it.

Drachenkaetzchen, 2021-02-11:
@mdevaev give me a minute, it seems that the arguments help needs to be updated that MJPEG is also valid

mdevaev, 2021-02-11:
I would appreciate it if you also write a couple of words in the README describing how to work with the Pi camera.

Drachenkaetzchen, 2021-02-11:
Well, it's as simple as `./ustreamer --host :: -m jpeg --device-timeout=5 --buffers=3 -r 2592x1944` (where 2592x1944 is the native resolution). There are several rpi cams out there so the maximum resolution differs.

It's almost 1am now so I won't do that today, but if I forget please remind me :)

mdevaev, 2021-02-11:
Okay)

mdevaev, 2021-02-17:
@Drachenkaetzchen boop :)

Drachenkaetzchen, 2021-02-17:
Done :)

mdevaev, 2021-02-17:
Thanks! Merged. I think this is solved.

closed 2021-02-17 by mdevaev

pixelspark, 2021-06-05:
Works like a charm! Thanks to all involved :+1: