pikvm/ustreamer #42: Make license (more) permissive

dumblob, 2020-09-17:
Hi there, I'm thinking of using this in non-GPL software bundle (containing mostly open source, but also some proprietary stuff - altogether there are hundreds of different licenses) and there'll be some licensing clashes.

Would you consider changing license of uStreamer to a permissive license (MIT, BSD or alike) or at least to LGPL (LGPL is not a win, but has the potential to solve quite many of the problematic cases)?

mdevaev, 2020-09-17:
If you want to use ustreamer as a separate application, you can do so without being hindered. GPLv3 does not oblige you to open the sources of other components, unless you link them to ustreamer for some reason.

Another thing is that ustreamer requires you to open all the patches made for it. Do you want to avoid this?

dumblob, 2020-09-17:
>If you want to use ustreamer as a separate application, you can do so without being hindered. GPLv3 does not oblige you to open the sources of other components, unless you link them to ustreamer for some reason.

It's about bundling and **distributing** ustreamer (in binary form). So GPL unfortunately applies.

>Another thing is that ustreamer requires you to open all the patches made for it. Do you want to avoid this?

Definitely not. Actually my intention is to contribute (I'm starting already as you can see in [my other issue](https://github.com/pikvm/ustreamer/issues/43 ) :wink:), but I can't disclose binaries (nor source code if I'll ever get access to it) of those other bundled apps I'd like to distribute ustreamer with.

mdevaev, 2020-09-17:
> It's about bundling and distributing ustreamer (in binary form). So GPL unfortunately applies.

You have the right to distribute binary builds. The GPL only requires attribution and a link to the source code.

mdevaev, 2020-09-17:
The GPL does not require you to disclose the source code of applications that just use this. The requirement only applies to patches and linking (in the sense of compilation).

dumblob, 2020-09-18:
>The requirement only applies to patches and linking (in the sense of compilation).

...and to the other **binaries** distributed together as bundle (bundle in a sense of "mutually interconnected parts being dependent upon each other") in my understanding. I can't disclose certain other binaries too.

mdevaev, 2020-09-18:
> mutually interconnected parts being dependent upon each other

This means an indissoluble connection like linking. For example, the kernel module. But if you just run the binary and communicate with it over HTTP, this is fine. You might want to consult a lawyer. It seems to me that everything is in order here.

I have already been asked to change the license to BSD. My principled position on the issue of more permissive licenses is refusal. First of all, it's not a library. Secondly, in my opinion, permissive software licenses allow corporations to earn money on it without giving anything in return. They may or may not give away patches. They can earn billions by giving away only leftovers from the table, as in the case of Apple and the BSD network stack.

In your particular case, I believe that you are not going to abuse the permissive license. But if I do this, my software will turn into a cash cow for those who want to abuse it without opening their patches and making money from it.

I'm not greedy, but anyone who wants to make proprietary software should be useful to open source, otherwise it's a one-way game. Either it has to give back the patches, or pay for a double license. I spend a lot of time on this project and because of my country of residence it is extremely difficult for me to monetize my work. I love open source, but I don't want to help proprietary software manufacturers for free. I hope you understand me.

dumblob, 2020-09-18:
>You might want to consult a lawyer. It seems to me that everything is in order here.

Hm, from my understanding this is why **L**GPL exists. But most important is what you as the author want and by stating the above you're giving me permission to use it in a way I need, so I should be good to go if I copy & paste this whole discussion at the end of the GPL license.

>I hope you understand me.

I totally do. I'm trying to do the same - and walking this thin ice with open source and commercial SW is my everyday job :wink:.

mdevaev, 2020-09-18:
> But most important is what you as the author want and by stating the above you're giving me permission to use it in a way I need

Hey, I was just listing the license position on this. I don't know exactly how you are going to use my code, I do not make exceptions to GPLv3, and I don't give explicit permission to use my software without complying with GPLv3, especially on GitHub. Forgive my paranoia, but I can say "yes", and then you can edit your comment after a year and add whatever you want to it. Let's find out if you are eligible under GPLv3. If my understanding of GPLv3 is wrong, then we will either talk about dual licensing explicitly, or I will change the license probably.

mdevaev, 2020-09-18:
Ask a question on StackOverflow or find an appropriate explanation from a reputable source.

dumblob, 2020-09-18:
>Let's find out if you are eligible under GPLv3.

I'm not. I have a lawyer specializing in SW licenses (among other things he's a country "chairman" of Creative Commons) and despite the fact, that his opinion is "yet another lawyers opinion", I stick with it.

Btw. for GPL the rule of thumb is, that if there is anything existing which you could use as an alternative to the SW in question (i.e. something like ustreamer) without significant (not just SLOC but also semantically) changes to the surrounding code, then GPL restrictions on the surrounding/bundled/coupled SW/digital_data do not apply. But there is no such alternative to ustreamer (that's the whole point of ustreamer's existence if I'm not mistaken), so this restriction quite possibly applies :cry:.

With LGPL this restriction is gone as the only restrictions which "stay" in LGPL are those concerning ustreamer itself (i.e. changes to ustreamer itself) which, contrary to GPL, poses no issues in practice (and I bet this is exactly what you want - anything which makes ustreamer better **must** be published in it's original/source form). I don't see a reason though to restrict the surrounding.

>If my understanding of GPLv3 is wrong, then we will either talk about dual licensing explicitly, or I will change the license probably.

I can't say whether it's wrong or not and it doesn't even pertain to me to comment on this. What I'm sure about though is GPL (unlike LGPL) is extremely **unsafe** in this regard preventing **any** use in practice (because laywers rightfully fear other lawyers and future changes in law) thus hindering the sought "publicity" and wider use of ustreamer.

I absolutely understand why you didn't choose any permissive license and wanted a Free Software license. Then I'd suggest relicencing to LGPL to allow bundling (distributing and tightly coupling, possibly but not necessarily linking or FFI calling parts of the binary or reusing chosen parts of the source code) with proprietary SW without the need to disclose anything from the proprietary part.

mdevaev, 2020-09-18:
> But there is no such alternative to ustreamer

There is no perfect copy of the software, but you can use [mjpg-streamer](https://github.com/jacksonliam/mjpg-streamer) as an alternative.

> I can't say whether it's wrong or not and it doesn't even pertain to me to comment on this. What I'm sure about though is GPL (unlike LGPL) is extremely unsafe in this regard preventing any use in practice. It seems that most of the existing open source software is made under the GPL license and this does not bother anyone. Like you, I worked for companies that deal with proprietary software.

GPLv2 does not interfere with the use of Linux in Android in any way. GPLv3, too, only restricts such a vile phenomenon as tivoization.

> thus hindering the sought "publicity" and wider use of ustreamer

Ustreamer was created to solve my problems. This is its main purpose. But because I love open source, I made it an open source project. I don't have a goal to make it used by everyone and in any way. I will not experience the thrill or joy of someone putting ustreamer on their proprietary box, having invested about zero money in the development of the video server. Even if this is not your case, changing the license will make this situation possible. It contradicts my beliefs. So, as I said: either GPLv3 and patches, or dual licensing.

dumblob, 2020-09-18:
>There is no perfect copy of the software, but you can use mjpg-streamer as an alternative.

Good point, thanks. I think this partially solves my issue. The rest is "what would happen if someone (you? :wink:) sued me for choosing ustreamer over mjpg-streamer by default"...

>GPLv2 does not interfere with the use of Linux in Android in any way.

Because everything in Android which is fully dependent on Linux (i.e. wouldn't work with any alternative without significant changes) is also open source. This means e.g., that proprietary firmware doesn't have to be disclosed as it doesn't depend on Linux, but Linux depends on the firmware. But e.g. Dalvik and all of the Android (except for user apps which are not tightly coupled with Linux) does depend on Linux (though Linux doesn't depend on Android obviously :wink:) and as such must be disclosed.

>GPLv3, too, only restricts such a vile phenomenon as tivoization.

Well, there are other important differences to GPLv2, but that's not what this thread is about. Btw. yeah I did even publicly ask Richard Stallman **face to face** few years ago about tivoization. This doesn't change anything on the fact, that GPL does restrict use of surrounding tightly coupled SW (be it scripts using ustreamer or FFI calls or whatever) if distributed.

>Ustreamer was created to solve my problems. This is its main purpose. But because I love open source, I made it an open source project. I don't have a goal to make it used by everyone and in any way.

Oh, understood.

>So, as I said: either GPLv3 and patches, or dual licensing.

Sounds promising. What would be the "patches"? And what would be the second license in case of dual licensing?

mdevaev, 2020-09-18:
OK, another example: Linux and the NVIDIA driver. There is a complex case, but the NVIDIA driver is closed.

By dual licensing, I mean the way it used to be with Qt. This was distributed under the GPL and was great for open source software, or the manufacturer had to pay for an alternative license that allowed them to make closed source software.

In my case, if you want to patch ustreamer and not show its sources (and patch sources), you need my permission to it. I don't make gifts for proprietary software manufacturers, so the terms need to be discussed. However, if you just use ustreamer under the GPL license (open all the patches and don't link to it, etc) then everything is OK.

dumblob, 2020-09-18:
>OK, another example: Linux and the NVIDIA driver. There is a complex case, but the NVIDIA driver is closed.

I think it's a good example. It's though the same principle - the driver's source itself is not dependent on Linux (it's also for Windows etc.), but there is a tiny part which is **only** for Linux and it's open source. It's a shim between the "generic" driver and Linux. IIRC historically this wasn't the case (it was all closed/binary) and there actually were some law suits and since then the shim is a solution.

>By dual licensing, I mean the way it used to be with Qt. This was distributed under the GPL and was great for open source software, or the manufacturer had to pay for an alternative license that allowed them to make closed source software.

I'm afraid this is a misconception. Qt is **L**GPL, not GPL (actually they have some tiny amount of [add-ons under GPL](https://doc.qt.io/qt-5/qtmodules.html#gpl-licensed-addons ), but you most certainly never heard of them). LGPL is fine and I'd be totally happy with LGPL.

>In my case, if you want to patch ustreamer and not show its sources (and patch sources), you need my permission to it. I don't make gifts for proprietary software manufacturers, so the terms need to be discussed.

Ok, if I understand correctly, it'll be a private discussion. May I use  mdevaev@gmail.com for that matter?

mdevaev, 2020-09-18:
> I'm afraid this is a misconception. Qt is LGPL, not GPL

Yes, but I'm talking about the old model. Previously, Qt was distributed under the GPL.

> May I use mdevaev@gmail.com for that matter?

Yes.

mdevaev, 2020-09-19:
Feel free to reopen if you want to discuss anything else.

closed 2020-09-19 by mdevaev