pikvm/ustreamer #48: Is Access-Control-Allow-Origin header in server.c needed?

aggieNick02, 2020-09-22:
Apologies for anything I misunderstand as I've only recently learned a bit about this header. I noticed it is set to "*" in both _http_callback_snapshot and _http_callback_stream_write, and from what I understand it won't allow any bad cross-site stuff since since PREPROCESS_REQUEST protects both code areas and authorization cookies are not sent cross-domain with CORS.

But just to try to understand what the purpose of it was in server.c, I commented both lines out and rebuilt, and everything *seemed* to still work just fine.

Was curious if there are cases where the header is needed or helpful?

mdevaev, 2020-09-22:
Hi. This was necessary to support access to Webcams from various pages. I spied this in mjpg-streamer: 
- https://github.com/jacksonliam/mjpg-streamer/issues/32
- https://github.com/jacksonliam/mjpg-streamer/commit/6955a9e16c046658d0f28ca85756db3db187deef

But I am open to discussion about this. Do you have any suggestions on how this should be done correctly?

aggieNick02, 2020-09-22:
One option could be to have the header off by default, and able to be turned on via a command line parameter (either to set to "*" or some other string. This way ustreamer would be as secure as possible by default, including in pikvm, but if somebody who uses ustreamer needs the header, they can turn it on.

I'll think about other options and try to learn a bit more about CORS tomorrow. I'm pretty sure, from what I've learned, that since ustreamer does authentication, there isn't a security issue for pikvm here.

mdevaev, 2020-09-22:
I agree. It needs to be fixed.

aggieNick02, 2020-09-22:
Happy to. Thanks for looking at this so quickly.

I tested the branch both standalone and under pikvm. The option works well. Verified that everything in pikvm and regular standalone ustreamer still work as before, and that if I really want to, I can set to "*".

Minor fixes for the readme:
For the command line on many shells the * will need to be escaped, e.g., `--allow-origin=\*` .

And then just a minor typo, it should be "Please note that" instead of "Please not that" to start the note about the new option. Thanks again!

mdevaev, 2020-09-22:
Perfect! Fixed. Thanks for the help. I will release it after teting some other features.

closed 2020-09-22 by mdevaev