pikvm/ustreamer #50: ustreamer process name in top should be ustreamer instead of main

rgl, 2020-09-23:
The ustreamer process appear as `main` in top:

```
Tasks: 119 total,   1 running, 118 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.2 us,  0.3 sy,  0.0 ni, 99.5 id,  0.1 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   3827.9 total,   2928.2 free,    112.4 used,    787.3 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.   3575.5 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                         
 4064 root      20   0  139032   5840   1060 S  14.6   0.1   0:03.87 main                                                                                            
```

I think it should appear as ustreamer to better identify it.

Please note that in ps it correclty shows as ustreamer:

```
ps axw | grep ustreamer
 4064 pts/0    Sl+    0:22 ustreamer: ustreamer --device=/dev/video0 --persistent --dv-timings --format=uyvy --encoder=omx --workers=3 --quality=85 --desired-fps=30 --drop-same-frames=30 --process-name-prefix=ustreamer --slowdown --host=0.0.0.0 --port=80
```

mdevaev, 2020-09-23:
This is a feature of top. Ustreamer uses thread renaming in the kernel to make them easier to identify. If you press "c" in top, you will see the path to the command. Btw I recommend using htop. You can configure it to display a tree of threads with names.

In any case, this is not a bug.

[image]

rgl, 2020-09-23:
I though it had to do something like that (hinted by the fact that `setproctitle` is being used); thanks for clearing that up!

But that still doesn't explain from where the `main` name comes from; anyhow, I would expect it to be `ustreamer` instead of `main` :-)

mdevaev, 2020-09-23:
`main` is name of the main thread of ustreamer)

closed 2020-09-23 by mdevaev

rgl, 2020-09-23:
Should the main thread be named ustreamer?

mdevaev, 2020-09-23:
This won't work. Run ustreamer. You will see in square brackets in the log: http, main, etc. If you replace main with ustreamer, this will also change in the log. I'd rather not do that.

rgl, 2020-09-23:
Interesting, I never seen that in the logs. My build made with `make -j$(nproc) WITH_OMX=1 WITH_GPIO=1 install` stdout log does not show the thread name. It shows something like:

```
-- INFO  [68.180 ustreamer] -- Installing SIGINT handler ...
-- INFO  [68.180 ustreamer] -- Installing SIGTERM handler ...
-- INFO  [68.180 ustreamer] -- Ignoring SIGPIPE ...
-- INFO  [68.181 ustreamer] -- Using internal blank placeholder
-- INFO  [68.182 ustreamer] -- Listening HTTP on [0.0.0.0]:80
-- INFO  [68.182    stream] -- Using V4L2 device: /dev/video0
-- INFO  [68.182    stream] -- Using desired FPS: 30
```

Oh... after pasting the text here I can now see it was "hidden"! look at my console screenshot:

[image]

Anyways, I would prefer to have an empty string (like above) in the log to represent main rather than having a `main` being displayed in `top`, but its your call and I respect it.

mdevaev, 2020-09-23:
Your color scheme doesn't display some colors.

[image]

I refuse for a reason. Look in the list of processes and you will see `http`, `stream`, and `worker-1`. Even if we rename the `main` thread, you will still have these. I was thinking about an option for the main thread name, but given the previous statement, it doesn't make much sense. Okay, I could use a prefix. For example, `ustreamer/main`, `ustreamer/worker-1`, etc. But the fact is that the stream name is usually limited to 16 characters. 15, if you subtract `\0`. It just won't fit :+1: 

BTW. I really think the default behavior of top is strange. I didn't even think about the existence of this problem, because I always used htop or ps.

rgl, 2020-09-23:
I understand, no worries :-)

I'm only talking about top, because that is normally the default (at least in ubuntu/raspbian).

I never realized that my terminal was fubar, I'm using a regular gnome-terminal with the built-in scheme, and I "fixed it" by changing the built-in scheme to another value, and back again to the same value... so weird.