pikvm/ustreamer #52: New install - no video: ERROR: Mainloop select() timeout

JohnOCFII, 2020-10-24:
Greetings,

I created and installed a fresh Raspbian image, and followed your build instructions (note -- I also needed to install git).

Build seemed to go well.  I built with WITH_OMX=1.

I've used this same hardware successfully with mjpg-streamer as well as fswebcam and motion.  It is an ELP-cam. `Bus 001 Device 004: ID 05a3:9520 ARC International Camera`

I can reach the web page successfully, but there is no image.  As shown in the issue title, Mainloop select() timeout is repeated.

I've created a gist with debug info here: https://gist.github.com/JohnOCFII/a2649d01e1e66027367e772361b1a3e5

I imagine I'm missing something simple.  

Thanks for your help.

John

mdevaev, 2020-10-24:
Hi. Please show me `v4l2-ctl --list-formats-ext -d /dev/video0` and ustreamer CLI flags.

JohnOCFII, 2020-10-24:
```
pi@octocore:~/ustreamer $ v4l2-ctl --list-formats-ext -d /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1024x768
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x1024
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1600x1200
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 2048x1536
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 2592x1944
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1024x768
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.133s (7.500 fps)
		Size: Discrete 1280x1024
			Interval: Discrete 0.133s (7.500 fps)
		Size: Discrete 1600x1200
			Interval: Discrete 0.333s (3.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.333s (3.000 fps)
		Size: Discrete 2048x1536
			Interval: Discrete 0.333s (3.000 fps)
		Size: Discrete 2592x1944
			Interval: Discrete 0.333s (3.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
```

and

`pi@octocore:~/ustreamer $ ./ustreamer --host=0.0.0.0 --port=8081 --debug`

Thank you.

mdevaev, 2020-10-24:
Try: `./ustreamer --host=0.0.0.0 --port=8081 --format=jpeg --device-timeout=5`.

JohnOCFII, 2020-10-24:
That worked!

Thank you!

I do get an error in the output (`Device does not support setting of HW encoding quality parameters`), but on a quick test, it seems to function properly.

```
pi@octocore:~/ustreamer $ ./ustreamer --host=0.0.0.0 --port=8081 --format=jpeg --device-timeout=5 
-- INFO  [8480.947      main] -- Installing SIGINT handler ...
-- INFO  [8480.947      main] -- Installing SIGTERM handler ...
-- INFO  [8480.947      main] -- Ignoring SIGPIPE ...
-- INFO  [8480.948      main] -- Using internal blank placeholder
-- INFO  [8480.948      main] -- Listening HTTP on [0.0.0.0]:8081
-- INFO  [8480.949    stream] -- Using V4L2 device: /dev/video0
-- INFO  [8480.949    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [8480.949      http] -- Starting HTTP eventloop ...
-- INFO  [8481.031    stream] -- Device fd=8 opened
-- INFO  [8481.031    stream] -- Using input channel: 0
-- INFO  [8481.031    stream] -- Using TV standard: DEFAULT
-- INFO  [8481.032    stream] -- Using resolution: 640x480
-- INFO  [8481.033    stream] -- Using pixelformat: JPEG
-- INFO  [8481.034    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [8481.034    stream] -- Using IO method: MMAP
-- INFO  [8481.038    stream] -- Requested 5 device buffers, got 5
-- INFO  [8481.084    stream] -- Capturing started
-- INFO  [8481.084    stream] -- Switching to HW encoder because the input format is (M)JPEG
-- ERROR [8481.084    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [8481.084    stream] -- Using JPEG quality: encoder default
-- INFO  [8481.084    stream] -- Creating pool with 1 workers ...
-- INFO  [8481.085    stream] -- Capturing ...
-- INFO  [8490.350      http] -- HTTP: Registered client: [192.168.88.135]:60376, id=2f18a369-d38f-4a69-bef4-ac03e3965200; clients now: 1
^C-- INFO  [8585.949      main] -- ===== Stopping by SIGINT =====
-- INFO  [8585.950      http] -- HTTP eventloop stopped
-- INFO  [8586.010    stream] -- Destroying workers pool ...
-- INFO  [8586.011    stream] -- Capturing stopped
-- INFO  [8586.014    stream] -- Device fd=8 closed
-- INFO  [8586.015      main] -- Bye-bye
pi@octocore:~/ustreamer $ 
```

mdevaev, 2020-10-24:
Great! The error you are asking about is not fatal. It's just that your camera can't adjust the compression level. You can remove the `--format=jpeg` parameter and try `--encoder=cpu` along with `--quality=50` if you want to reduce the quality. Or use OMX for the same purpose. In any case, this will load your Pi. If you are not interested in managing compression, then continue using '--format-jpeg` and everything will be fine.

JohnOCFII, 2020-10-24:
Thanks very much for the very quick assistance.  Now that I know the basics will work, I'll try out some of the other options.  This camera will be used to stream video from one of my 3D printers.

Thanks again!

closed 2020-10-24 by JohnOCFII

mdevaev, 2020-10-24:
You're welcome.