pikvm/ustreamer #53: Second capture card causes: "Unable to start capturing: No space left on device"

vedranius, 2020-10-27:
I have two same capture cards and if I try to stream simultaneously from both cards I get this error:
-- ERROR [247.685    stream] -- Unable to start capturing: No space left on device

My code:
./ustreamer --device=/dev/video1 --host=0.0.0.0 --port=12346 -r 1920x1080 -m JPEG -f 30

And second one:
./ustreamer --device=/dev/video3 --host=0.0.0.0 --port=12345 -r 1920x1080 -m JPEG -f 30

SD card is 256GB so there should be enough space.

Tnx!

mdevaev, 2020-10-27:
Hello, which capture cards do you use?

vedranius, 2020-10-28:
Using this one which is working uses MJPG for encoding.
https://www.aliexpress.com/item/4000917130635.html
So far, it works great with all of the software I've tested, and with Ustreamer, but only if I run one.

mdevaev, 2020-10-28:
So other programs don't work when you try to use two devices at once? I remembered that I was once told about a similar problem, but with cameras. Perhaps this is also your case. If other programs are buggy in the same way, it is a kernel bug. Ustreamer does not have its own drivers and uses the V4L2 API to get the image. If the driver can't handle two cameras, then I can't do anything about it. As for the capture card: yes, it works via the uvcvideo driver, they look like webcams for the OS.

vedranius, 2020-10-28:
No, I can use any number of these cards in software Ive tested.
This only didnt worked in Ustreamer.

mdevaev, 2020-10-28:
Okay. Plz give me an example of the program and startup parameters for two devices.

vedranius, 2020-10-28:
For now Ive just tested it with OBS Studio, vMix and Dicaffeine (only supports 1 output right now).

mdevaev, 2020-10-28:
And the camera settings were the same? The error you are seeing may be due to insufficient controller bandwidth.

Could you try this? https://github.com/jacksonliam/mjpg-streamer

mdevaev, 2020-10-29:
Proof: https://stackoverflow.com/questions/11394712/libv4l2-error-turning-on-stream-no-space-left-on-device

vedranius, 2020-10-29:
> Proof: https://stackoverflow.com/questions/11394712/libv4l2-error-turning-on-stream-no-space-left-on-device

Yeah! I think that was the case! Tnx!
I've tried now MJPG streamer and I've set resolution to 1920x1080 first and that didn't worked out for two devices.
Then I've changed it to 1280x720 and that works great ;) 

I'll try Ustreamer too later on.

vedranius, 2020-10-29:
Moved one device to different USB bus (chip) and now it's fine even at 1920x1080.

vedranius, 2020-10-29:
Testing with two same devices on Orange Pi 3 running Armbian and it works awesome! :) 
Doing 1920x1080.

`./ustreamer --device=/dev/video1 --host=0.0.0.0 --port=12345 -r 1920x1080 -m JPEG -f 30 - an FHD action camera
 ./ustreamer --device=/dev/video3 --host=0.0.0.0 --port=12346 -r 1920x1080 -m JPEG -f 30 - laptop FHD YouTube video`

vedranius, 2020-10-29:
@mdevaev If I should open another case, it's fine.
But, is there a way to add an audio from capture card too?
Tnx!

mdevaev, 2020-10-29:
Glad to hear that it works.

MJPG does not contain an audio channel, so this cannot be done using ustreamer. Perhaps in the future I will add the ability to transmit audio in a separate stream. Right now you can use ffmpeg or something like that for audio stream. For video you can keep ustreamer.

closed 2020-10-30 by vedranius