pikvm/ustreamer #58: NDI Output Support

GadgetUK, 2020-11-23:
Would be great to include NDI output support for ultra low latency video output.

mdevaev, 2020-11-23:
Sup. What is NDI output?

GadgetUK, 2020-11-23:
NDI = Network Device Interface, it allows low latency video over IP. The following might give some insight, For me the use case would be as an input to OBS (Open Broadcast Studio).
https://www.youtube.com/watch?v=cX8fo74m6nc&feature=emb_title

mdevaev, 2020-11-24:
Ustreamer currently does not have the ability to implement alternative video transfer protocols. This is a separate big job, which, moreover, I do not have the opportunity to do (at least right now).

As a temporary solution, I can suggest that you use browser source in OBS. Create a small HTML page, add the mjpeg stream using the img tag, and connect this page to OBS.

mdevaev, 2021-01-07:
I implemented the ability to receive frames from ustreamer via shared memory. This allows you to write the stream to a file using ffmpeg or send it for processing further using pipe. I don't think NDI will be supported in the foreseeable future, but you can use ustreamer-dump or its code to make an ndi-proxy. It's better than nothing.

The small example:
```
$ ustreamer --jpeg-sink=test
$ ustreamer-dump --sink=test -o - | ffmpeg -use_wallclock_as_timestamps 1 -i pipe: -c:v libx264 test.mp4
```

closed 2021-01-07 by mdevaev

pinpox, 2024-01-08:
In case anyone stumbles across this: This software can be used instead for this purpouse https://github.com/lplassman/V4L2-to-NDI