pikvm/ustreamer #59: Package for Debian

reedy, 2020-11-25:
It'd be pretty cool if we could (eventually) get ustreamer into debian package repos. Don't suppose there's a DD hanging around the community? ;)

https://www.debian.org/doc/manuals/distribute-deb/distribute-deb.html

Would help with https://github.com/guysoft/OctoPi/issues/698 too

mdevaev, 2020-11-25:
This is a good idea but Debian is its own unique world anduUnfortunately I don't have time to study the features their packaging. If you suggest a patch for packages, I will accept it and support it.

reedy, 2020-11-25:
I do know quite a few DD via work and in other tech circles.

When I've got some free time, I'll try and see if I can get some help (or just nerd snipe them ;)).

In the meantime, if anyone already has the skills to do this and want to work on a patch, please feel free!

reedy, 2020-11-30:
I've made a start... :) https://github.com/reedy/ustreamer/commits/debian

reedy, 2020-12-01:
I've filed an Debian ITP (Intent to Package), waiting on a bug number for that

https://salsa.debian.org/reedy/ustreamer

reedy, 2020-12-01:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=976232

reedy, 2020-12-03:
Nearly ready to go actually now, I think!

```
I: ustreamer: spelling-error-in-binary usr/bin/ustreamer bellow below
W: ustreamer: binary-without-manpage usr/bin/ustreamer
```
Spelling error was fixed in https://github.com/pikvm/ustreamer/pull/66

manpage is https://github.com/pikvm/ustreamer/pull/64

But neither are blockers to packaging. Will see what happens. Ideally need to wait for a tagged release to pull those in easily, but no rush on that. It'll happen eventually

Next step (possibly tomorrow) is to setup a sid/unstable chroot to check I've not missed anything obvious (like relating to dependancies)...

reedy, 2021-01-22:
https://ftp-master.debian.org/new/ustreamer_3.12-1.html

mdevaev, 2021-01-22:
I'm going to get drunk)

By the way, how does debian keep packages up-to-date?

reedy, 2021-02-03:
>ustreamer_3.12-1_amd64.changes ACCEPTED into unstable, unstable

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=976232

https://tracker.debian.org/pkg/ustreamer

mdevaev, 2021-02-18:
I'm closing this issue since everything is done here, apparently. Now the ball is on the debian side.

closed 2021-02-18 by mdevaev