pikvm/ustreamer #60: Building on Ubuntu 20.04

reedy, 2020-11-25:
Not sure how best to integrate this into the readme... But on Ubuntu 20.04 (x86_64) there's package differences between Debian itself (as always)

```
# apt install build-essential libevent-dev libjpeg62-dev uuid-dev libbsd-dev make gcc libjpeg8 libjpeg-turbo8 libuuid1 libbsd0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libbsd0 is already the newest version (0.10.0-1).
libbsd0 set to manually installed.
libuuid1 is already the newest version (2.34-0.1ubuntu9.1).
libuuid1 set to manually installed.
The following additional packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu cpp cpp-9 dpkg-dev fakeroot g++ g++-9 gcc-9 gcc-9-base libalgorithm-diff-perl libalgorithm-diff-xs-perl
  libalgorithm-merge-perl libasan5 libatomic1 libbinutils libc-dev-bin libc6-dev libcc1-0 libcrypt-dev libctf-nobfd0 libctf0 libdpkg-perl libevent-core-2.1-7 libevent-extra-2.1-7
  libevent-openssl-2.1-7 libevent-pthreads-2.1-7 libfakeroot libfile-fcntllock-perl libgcc-9-dev libgomp1 libisl22 libitm1 libjpeg62 liblsan0 libmpc3 libquadmath0 libstdc++-9-dev
  libtsan0 libubsan1 linux-libc-dev manpages-dev
Suggested packages:
  binutils-doc cpp-doc gcc-9-locales debian-keyring g++-multilib g++-9-multilib gcc-9-doc gcc-multilib autoconf automake libtool flex bison gdb gcc-doc gcc-9-multilib glibc-doc
  bzr libstdc++-9-doc make-doc
The following NEW packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu build-essential cpp cpp-9 dpkg-dev fakeroot g++ g++-9 gcc gcc-9 gcc-9-base libalgorithm-diff-perl libalgorithm-diff-xs-perl
  libalgorithm-merge-perl libasan5 libatomic1 libbinutils libbsd-dev libc-dev-bin libc6-dev libcc1-0 libcrypt-dev libctf-nobfd0 libctf0 libdpkg-perl libevent-core-2.1-7
  libevent-dev libevent-extra-2.1-7 libevent-openssl-2.1-7 libevent-pthreads-2.1-7 libfakeroot libfile-fcntllock-perl libgcc-9-dev libgomp1 libisl22 libitm1 libjpeg-turbo8
  libjpeg62 libjpeg62-dev libjpeg8 liblsan0 libmpc3 libquadmath0 libstdc++-9-dev libtsan0 libubsan1 linux-libc-dev make manpages-dev uuid-dev
0 upgraded, 52 newly installed, 0 to remove and 3 not upgraded.
Need to get 40.9 MB of archives.
After this operation, 179 MB of additional disk space will be used.
```

```
$ make
-- CC src/device.c
-- CC src/encoder.c
-- CC src/encoders/cpu/encoder.c
-- CC src/encoders/hw/encoder.c
-- CC src/http/base64.c
-- CC src/http/blank.c
-- CC src/http/mime.c
-- CC src/http/path.c
-- CC src/http/server.c
-- CC src/http/static.c
-- CC src/http/unix.c
-- CC src/http/uri.c
-- CC src/logging.c
-- CC src/main.c
-- CC src/options.c
-- CC src/picture.c
-- CC src/stream.c
-- LD ustreamer
===== Build complete =====
== CC      = cc
== LIBS    = -lm -ljpeg -pthread -levent -levent_pthreads -luuid -lbsd
== CFLAGS  = -O3 -c -std=c11 -Wall
```

And then for GPIO support, `libgpiod-dev` for the package (which would install `libgpiod2`)

```
reedy@ubuntu64-build:~/ustreamer$ WITH_GPIO=1 make
-- CC src/gpio/gpio.c
-- LD ustreamer
===== Build complete =====
== CC      = cc
== LIBS    = -lm -ljpeg -pthread -levent -levent_pthreads -luuid -lgpiod -lbsd
== CFLAGS  = -O3 -c -std=c11 -Wall -Wextra -D_GNU_SOURCE -DWITH_GPIO -DWITH_PTHREAD_NP -DWITH_SETPROCTITLE
== LDFLAGS = 
```

mdevaev, 2020-11-25:
Thank you, I will add information about this.

mdevaev, 2020-11-26:
Done.

closed 2020-11-26 by mdevaev

steve1515, 2021-03-21:
@mdevaev I was recently trying to install this on Ubuntu 20.04 and found some issues with the packages listed here: https://github.com/pikvm/ustreamer#building

The issue I had was that you can't install libjpeg-dev and libjpeg62-dev on Ubuntu at the same time as they conflict. On Ubuntu libjpeg-dev is a reference to libjpeg8-dev/libjpeg8-turbo-dev which conflicts with the 6.2 version.

I've simplified the install line down to: `sudo apt install build-essential libevent-dev libjpeg-dev libbsd-dev`
(I cleaned up the extra packages that are installed by other packages already. e.g. gcc and make are installed by build-essential.)

I did some more digging and found that on Debian libjpeg-dev installs libjpeg62-dev while it installs libjpeg8-dev on Ubuntu. Basically, this lets you simplify your ReadMe page so that you only have to include a line for Debian and can remove the Ubuntu specific line. Just use `sudo apt install build-essential libevent-dev libjpeg-dev libbsd-dev` and it should work automatically on both Debian and Ubuntu 20.04.

mdevaev, 2021-03-21:
@steve1515 fixed, thanks!