pikvm/ustreamer #61: Can't watch video streamed

DarkFighterLuke, 2020-11-25:
Maybe the problem is me, but I can't figure out how to watch the streamed video.

I am streaming /dev/video0 which is an easyCAP dongle to capture analog video, everything seems to be fine in the µStreamer console output, but I can't watch the stream neither on VLC nor on ffplay or in an HTML page with a video tag having for source the address of the server.
I am trying this on my home LAN, so there shouldn't be problems related to the network.
What am I doing wrong?

mdevaev, 2020-11-25:
Show the command line params please. Also specify what it means "can't watch". You see a black screen, or you can't connect to ustreamer at all?

DarkFighterLuke, 2020-11-25:
> 
> 
> Show the command line params please. Also specify what it means "can't watch". You see a black screen, or you can't connect to ustreamer at all?

Hi, thank you for your answer.
The command is:
`./ustreamer --device=/dev/video1 --host=0.0.0.0 --port=80 --encoder=omx --workers=3 --persistent --drop-same-frames=30`

I can't connect to ustreamer at all.

mdevaev, 2020-11-25:
Try `http://ip:8080/stream` in your browser. It should work. Otherwise it may be related with firewall or some local issues.

DarkFighterLuke, 2020-11-25:
> 
> 
> Try `http://ip:8080/stream` in your browser. It should work. Otherwise it may be related with firewall or some local issues.

Now it works, thank you!

However I reinstalled dietpi on the raspberry pi running ustreamer and now I am not able to use the omx encoder...
Both root and dietpi users are in the `video` group.

mdevaev, 2020-11-25:
The OMX setting is a separate song. But you don't need OMX for easycap, `--encoder=cpu` is enough by power. You can try `--encoder=hw`.

DarkFighterLuke, 2020-11-25:
Thank you very much, everything is working now!

I think it could be useful for someone like me to know that it is necessary to use thre stream endpoint, maybe it can be added to te README 😃 .

Wonderful project in any case 😆

closed 2020-11-25 by DarkFighterLuke