pikvm/ustreamer #62: Create manpage

reedy, 2020-12-01:
Noticed this was missing when I was poking around with #59

Creating a manpage would be useful more universally - https://www.cyberciti.biz/faq/linux-unix-creating-a-manpage/

mdevaev, 2020-12-01:
Well, the same as with deb packages. That's great, but I don't have time right now :)

reedy, 2020-12-01:
Yeah, you don't have to do every request. And I don't expect you to! ;)

But someone else can almost certainly do it (feels an easy ish PR for someone looking to give back to the projects. Writing docs is often a good way for beginners and non coders to help) Or I can when I get around to it. I will file this sort of task as a todo/reminder when I come across them

reedy, 2020-12-01:
And I think I have the deb somewhere near now locally too ;p

reedy, 2020-12-01:
Boilerplate done, will flesh it out before I put it through as a proper PR

[image]

reedy, 2020-12-01:
Output is something currently like

```
USTREAMER(1)                                                                 General Commands Manual                                                                USTREAMER(1)

NAME
       ustreamer - stream MJPG video from any V4L2 device to the network

SYNOPSIS
       ustreamer [OPTION]

DESCRIPTION
       µStreamer (ustreamer) is a lightweight and very quick server to stream MJPG video from any V4L2 device to the network. All new browsers have native support of this video
       format, as well as most video players such as mplayer, VLC etc. µStreamer is a part of the Pi-KVM project designed to stream VGA and HDMI screencast hardware  data  with
       the highest resolution and FPS possible.

OPTIONS
   Capturing options
       -d /dev/path, --device /dev/path Path to V4L2 device. Default: /dev/video0.

       -i N, --input N
              Input channel. Default: 0.

       -r WxH, --resolution WxH
              Initial image resolution. Default: 640x480.

       -m fmt, --format fmt
              Image format.  Available: YUYV, UYVY, RGB565, RGB24, JPEG; default: YUYV.

       -a std, --tv-standard std
              Force TV standard.  Available: PAL, NTSC, SECAM; default: disabled.

       -I method, --io-method method
              Set V4L2 IO method (see kernel documentation). Changing of this parameter may increase the performance. Or not.  Available: MMAP, USERPTR; default: MMAP.

       -f N, --desired-fps N
              Desired FPS. Default: maximum possible.

       -z N, --min-frame-size N
              Drop frames smaller then this limit. Useful if the device produces small-sized garbage frames. Default: 128 bytes.

       -n, --persistent
              Don't re-initialize device on timeout. Default: disabled.

       -t, --dv-timings
              Enable DV timings querying and events processing to automatic resolution change. Default: disabled.

       -b N, --buffers N
              The  number  of buffers to receive data from the device. Each buffer may processed using an independent thread.  Default: 2 (the number of CPU cores (but not more
              than 4) + 1).

       -w N, --workers N
              The number of worker threads but not more than buffers.  Default: 1 (the number of CPU cores (but not more than 4)).

       -q N, --quality N
              Set quality of JPEG encoding from 1 to 100 (best). Default: 80.  Note: If HW encoding is used (JPEG source format selected), this parameter attempts to  configure
              the camera or capture device hardware's internal encoder. It does not re-encode MJPG to MJPG to change the quality level for sources that already output MJPG.

       -c type, --encoder type
              Use  specified  encoder. It may affect the number of workers.  CPU ─ Software MJPG encoding (default). HW ─ Use pre-encoded MJPG frames directly from camera hard‐
              ware.  Available: CPU, HW; default: CPU.

       --device-timeout sec
              Timeout for device querying. Default: 1.

       --device-error-delay sec
              Delay before trying to connect to the device again after an error (timeout for example). Default: 1.

   Image control options
       --image-default Reset all image settings bellow to default. Default: no change.
[... 112 lines cut ...]
              Disable color logging. Default: ditto.

   Help options
       -h, --help Print this text and exit.

       -v, --version
              Print version and exit.

       --features
              Print list of supported features.

BUGS
       Please file any bugs and issues at https://github.com/pikvm/ustreamer/issues

AUTHOR
       Maxim Devaev <mdevaev@gmail.com>

HOMEPAGE
       https://pikvm.org/

COPYRIGHT
       GNU General Public License v3.0

November 2020                                                                      version 2.2                                                                      USTREAMER(1)
```

closed 2020-12-03 by mdevaev

reedy, 2020-12-03:
At some point in the future I'll come back around and add the GPIO stuff too... :)

reedy, 2020-12-03:
And I guess it wants plumbing into the various packaging too