pikvm/ustreamer #69: orangepi+USBtoHDMI（ID 534d:2109） ERROR ：Can't set input channel

TelDragon, 2020-12-15:
`v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : USB Video: USB Video
        Bus info      : usb-1c1b000.usb-1
        Driver version: 4.19.67
        Capabilities  : 0x84A00001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'MJPG'
        Field             : None
        Bytes per Line    : 0
        Size Image        : 4147200
        Colorspace        : sRGB
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             : 
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 1920, Height 1080
        Default     : Left 0, Top 0, Width 1920, Height 1080
        Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1920, Height 1080
Selection: crop_bounds, Left 0, Top 0, Width 1920, Height 1080
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness (int)    : min=-128 max=127 step=1 default=-11 value=-11
                       contrast (int)    : min=0 max=255 step=1 default=148 value=148
                     saturation (int)    : min=0 max=255 step=1 default=180 value=180
                            hue (int)    : min=-128 max=127 step=1 default=0 value=0`

`-- INFO  [33494.066    stream] -- Device fd=8 opened
-- INFO  [33494.066    stream] -- Using input channel: 100
-- ERROR [33494.066    stream] -- Can't set input channel
-- INFO  [33494.066    stream] -- Device fd=8 closed
-- INFO  [33494.066    stream] -- Sleeping 1 seconds before new stream init ...`

mdevaev, 2020-12-15:
Why you're using `--input=100`?

TelDragon, 2020-12-16:
@mdevaev 
> Why you're using `--input=100`? Show CLI arguments.

Sorry, the default value was used at first, but the value that was changed at will still not work.

TelDragon, 2020-12-16:
@mdevaev 

Sorry, I changed to a brand new zero and a brand new OS system.
It can be executed normally.

But I found a problem, this USB capture card and Centos7 kernel 3.10x on my other system shows that this card supports hardware encoding JPEG

But on zero, YUYV is displayed

What is the driving reason?

mdevaev, 2020-12-16:
I can't say. I don't have any Orange Pi Zero to test this. Perhaps bad drivers are to blame. If you use hardware JPEG compression, the load will always be lower.

TelDragon, 2020-12-17:
@mdevaev 

Okay, thank you very much for providing such a convenient tool. Next, I will look for different hardware and solutions to achieve the best results. If there is anything that can help you, I will submit my suggestions without reservation.

closed 2020-12-17 by TelDragon