pikvm/ustreamer #70: Centos7 make ustreamer error  "V4L2_EVENT_SOURCE_CHANGE’ undeclared (first use in this function)"

TelDragon, 2020-12-16:
```
Linux bogon 3.10.0-957.el7.x86_64 #1 SMP Thu Nov 8 23:39:32 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

```
[root@bogon ustreamer]# make
-- CC src/device.c
src/device.c: In function ‘device_consume_event’:
src/device.c:366:9: error: ‘V4L2_EVENT_SOURCE_CHANGE’ undeclared (first use in this function)
    case V4L2_EVENT_SOURCE_CHANGE:
         ^
src/device.c:366:9: note: each undeclared identifier is reported only once for each function it appears in
src/device.c: In function ‘_device_open_dv_timings’:
src/device.c:431:14: error: ‘V4L2_EVENT_SOURCE_CHANGE’ undeclared (first use in this function)
   sub.type = V4L2_EVENT_SOURCE_CHANGE;
              ^
make: *** [build/src/device.o] Error 1
```

mdevaev, 2020-12-16:
Your kernel is too old. Is it possible to upgrade to 4.x?

TelDragon, 2020-12-17:
@mdevaev 

Okay, thank you very much, I will find a new machine to install a newer OS system to try

closed 2020-12-17 by mdevaev