pikvm/ustreamer #72: Stream won't load in OctoPrint

thesmart, 2020-12-25:
I have ustreamer working flawlessly when I type in the URL directly in a browser, but when loaded in OctoPrint's control tab, I get a CORS error:

```
Cross-Origin Read Blocking (CORB) blocked cross-origin response http://10.0.30.101/webcam/?action=stream&1608856928650 with MIME type text/html. See https://www.chromestatus.com/feature/5629709824032768 for more details.
```

Here are the options I'm passing to ustreamer:
```
-device=/dev/video1 -r 1920x1080 --format JPEG --buffers 2 --workers 2 --slowdown --persistent --device-timeout 16 --tcp-nodelay --allow-origin *
```

When I look at the response header for the request, I do not see the `Access-Control-Allow-Origin` header set.  I have also tired providing a value that is not wildcard, but this doesn't nothing either.  

Wondering if this is a bug or not. Thanks!

thesmart, 2020-12-25:
I setup a proxy server to force the CORS headers, which resolved the console error message about CORS. However, the stream still fails to load in the control frame.  See screenshot:

[image]

This feature works when using mjpg-streamer.

thesmart, 2020-12-25:
Ok, I have resolved the issue.  It appears that ustreamer's mjpg-streamer compatibility layer doesn't work when OctoPrint attempts to add a random number argument like `/?action=stream&123456789`.  So the "fix" would be to either change the webcam URL to from e.g. `/?action=stream` to `/stream` or for ustreamer source to update its regex or whatever is failing to interpret additional parameters as a stream route.

mdevaev, 2020-12-25:
Hi. I'm glad you've already solved the problem. I've just checked the behavior of the URL parser. There doesn't seem to be an easy way to fix this for libevent, requiring a rewrite of the parser. However, you can do something like this: `http://localhost:8080/?action=stream&123456789=1`.

closed 2020-12-27 by mdevaev