pikvm/ustreamer #73: CS2 HDMI :: OMX Green bug

fred2nice, 2020-12-31:
Hi,

I test on a Raspberry 4 8Go with CSI 2 HDMI ( https://raspberrypiwiki.com/HDMI_TO_CSI-2 )
Output HDMI from the raspberry itself to the CSI HDMI 

Good display, but lower perf
./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=80 -r 1280x720 -n -l -m RGB24 --tcp-nodelay  -q 50
[image]

If I use -c omx I obtained a bad display, but good perf.
./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=80 -r 1280x720 -n -l -m RGB24 --tcp-nodelay  -q 50 -c omx
( I tried with several differents resolutions )
[image]

Settings :
v4l2-ctl --set-edid=file=../1080P50EDID.txt --fix-edid-checksums
v4l2-ctl --set-dv-bt-timings index=6

        Index: 6
        Active width: 1280
        Active height: 720
        Total width: 1980
        Total height: 750
        Frame format: progressive
        Polarities: +vsync +hsync
        Pixelclock: 74250000 Hz (50.00 frames per second)
        Horizontal frontporch: 440
        Horizontal sync: 40
        Horizontal backporch: 220
        Vertical frontporch: 5
        Vertical sync: 5
        Vertical backporch: 20
        Standards: CTA-861
        CTA-861 VIC: 19
        Flags: CE-video, has CTA-861 VIC

config.txt
start_x=1
gpu_mem=256
dtoverlay=tc358743

cmdline
cma=128M

If you have any idea ,-)
Thank You !

Kind Regards.

mdevaev, 2021-01-01:
Try `--dv-timings` and `-m uyvy`.

fred2nice, 2021-01-04:
Hi !

It works with --dv-timings.
Thank you !

Happy new year ,-)

mdevaev, 2021-01-04:
Happy New Year! :)

closed 2021-01-04 by mdevaev