pikvm/ustreamer #74: Next release

reedy, 2020-12-31:
Any idea when the next release will be?

Waiting on a couple of my additions (manpage and typos) to be able to include in and finish off https://github.com/pikvm/ustreamer/issues/59, rather than having patches in the packaging repo

Thanks!

mdevaev, 2021-01-01:
Sup. I'm finishing a major change and I think it will be ready in a week.

mdevaev, 2021-01-06:
@reedy If you want, you can help me. We need another man page for the new utility: ustreamer-dump. This is a thing to write a local stream from ustreamer to a file or redirect it to other utilities, such as ffmpeg:
```
$ ustreamer --jpeg-sink=test
$ ustreamer-dump --sink=test -o - | ffmpeg -use_wallclock_as_timestamps 1 -i pipe: -c:v libx264 test.mp4
```

reedy, 2021-01-06:
I can have a look tomorrow :)

reedy, 2021-01-07:
Looks like it shouldn't be too much work

```
uStreamer-dump - Dump uStreamer's memory sink to file
═══════════════════════════════════════════════════════
Version: 2.2; license: GPLv3
Copyright (C) 2018 Maxim Devaev <mdevaev@gmail.com>

Example:
════════
    ustreamer-dump --sink test -o - \
        | ffmpeg -use_wallclock_as_timestamps 1 -i pipe: -c:v libx264 test.mp4

Sink options:
═════════════
    -s|--sink <name>  ─── Memory sink ID. No default.

    -t|--timeout <sec>  ─ Timeout for the upcoming frame. Default: 1.

    -o|--output  ──────── Filename to dump. Use '-' for stdout. Default: just consume the sink.

Logging options:
════════════════
    --log-level <N>  ──── Verbosity level of messages from 0 (info) to 3 (debug).
                          Enabling debugging messages can slow down the program.
                          Available levels: 0 (info), 1 (performance), 2 (verbose), 3 (debug).
                          Default: 0.

    --perf  ───────────── Enable performance messages (same as --log-level=1). Default: disabled.

    --verbose  ────────── Enable verbose messages and lower (same as --log-level=2). Default: disabled.

    --debug  ──────────── Enable debug messages and lower (same as --log-level=3). Default: disabled.

    --force-log-colors  ─ Force color logging. Default: colored if stderr is a TTY.

    --no-log-colors  ──── Disable color logging. Default: ditto.

Help options:
═════════════
    -h|--help  ─────── Print this text and exit.

    -v|--version  ──── Print version and exit.
```

mdevaev, 2021-01-07:
Updated help a little

reedy, 2021-01-07:
```
uStreamer-dump - Dump uStreamer's memory sink to file
═════════════════════════════════════════════════════
Version: 2.2; license: GPLv3
Copyright (C) 2018 Maxim Devaev <mdevaev@gmail.com>

Example:
════════
    ustreamer-dump --sink test --output - \
        | ffmpeg -use_wallclock_as_timestamps 1 -i pipe: -c:v libx264 test.mp4

Sink options:
═════════════
    -s|--sink <name>  ──────── Memory sink ID. No default.

    -t|--sink-timeout <sec>  ─ Timeout for the upcoming frame. Default: 1.

    -o|--output  ───────────── Filename to dump. Use '-' for stdout. Default: just consume the sink.

    -j|--output-json  ──────── Format output as JSON. Required option --output. Default: disabled.

Logging options:
════════════════
    --log-level <N>  ──── Verbosity level of messages from 0 (info) to 3 (debug).
                          Enabling debugging messages can slow down the program.
                          Available levels: 0 (info), 1 (performance), 2 (verbose), 3 (debug).
                          Default: 0.

    --perf  ───────────── Enable performance messages (same as --log-level=1). Default: disabled.

    --verbose  ────────── Enable verbose messages and lower (same as --log-level=2). Default: disabled.

    --debug  ──────────── Enable debug messages and lower (same as --log-level=3). Default: disabled.

    --force-log-colors  ─ Force color logging. Default: colored if stderr is a TTY.

    --no-log-colors  ──── Disable color logging. Default: ditto.

Help options:
═════════════
    -h|--help  ─────── Print this text and exit.

    -v|--version  ──── Print version and exit.
```

mdevaev, 2021-01-07:
OK done with help and pkgs

mdevaev, 2021-01-11:
It's ready as 3.0

closed 2021-01-11 by mdevaev