pikvm/ustreamer #75: --quality not working?

nevets963, 2021-01-04:
I'm using ustreamer with USB cameras and I would like to adjust/reduce the bitrate of the MJPG streams by adjusting the quality.

When I use --quality, I don't see any noticable change to the MJPG stream. The quality still looks exactly exactly the same if I use 1 or 100.

Perhaps I've misunderstood how this works? Or that this doesn't work with USB cameras?

Any help would be appreciated. Thanks!

mdevaev, 2021-01-04:
When using hardware encoding by means of the camera itself, ustreamer accepts an already compressed jpeg stream via usb. Many cameras do not allow you to adjust the image quality.

nevets963, 2021-01-04:
I see. Makes sense. Do you know other means to further compress the stream? I have 2 different USB cameras, both set to use 800x600. But 1 camera is roughly 3x the bitrate.

mdevaev, 2021-01-06:
Probably the second camera does not use jpeg quantization tables. The only way to get a lower bitrate from it is to use compression on the cpu or omx.

nevets963, 2021-01-06:
That sounds interesting. Any pointers/info to how I could do compression on CPU or OMX?

mdevaev, 2021-01-07:
Use `--encoder=cpu` or `--encoder=omx` (only on raspberry pi).

mdevaev, 2021-01-11:
Perhaps it's solved

closed 2021-01-11 by mdevaev