pikvm/ustreamer #82: install: /usr/ports/multimedia/ustreamer/work/stage/usr/local/man1/ustreamer.1: No such file or directory

yurivict, 2021-01-13:
Version 3.2 fails:
```
===>  Staging for ustreamer-3.2
===>   Generating temporary packing list
gmake[1]: Entering directory '/usr/ports/multimedia/ustreamer/work/ustreamer-3.2'
install -Dm755 ustreamer /usr/ports/multimedia/ustreamer/work/stage/usr/local/bin/ustreamer
install -Dm755 ustreamer-dump /usr/ports/multimedia/ustreamer/work/stage/usr/local/bin/ustreamer-dump
install -Dm644 man/ustreamer.1 /usr/ports/multimedia/ustreamer/work/stage/usr/local/man1/ustreamer.1
install: /usr/ports/multimedia/ustreamer/work/stage/usr/local/man1/ustreamer.1: No such file or directory
gmake[1]: *** [Makefile:87: install] Error 71
gmake[1]: Leaving directory '/usr/ports/multimedia/ustreamer/work/ustreamer-3.2'
*** Error code 2
```

FreeBSD 12.2

mdevaev, 2021-01-13:
Any ideas what's wrong with `install`? I'm using this makefile without any problems on arch

yurivict, 2021-01-13:
You forgot to create the ```man1``` directory.

mdevaev, 2021-01-13:
This is done by the -D option on Linux. Is this not the case on FreeBSD?

mdevaev, 2021-01-13:
UPD: No. Awesome.
Linux: `-D     create all leading components of DEST except the last, or all components of --target-directory, then copy SOURCE to DEST`
FreeBSD: `-D	destdir	     Specify the DESTDIR (top of the file hierarchy) that the items are installed in to.`

yurivict, 2021-01-13:
Aparently not.

yurivict, 2021-01-13:
That's why GNU make is outdated, because you need to deal with such low-level issues.

mdevaev, 2021-01-13:
It's a part of coreutils not make, but I understand what you means. The problem is rather that different implementations use different options for the same things. I don't know who's to blame for this, but I'll have to fix it.

Try now. If everything works, I'll make a 3.3 release

yurivict, 2021-01-13:
It works now.

mdevaev, 2021-01-13:
Ok, released

closed 2021-01-13 by mdevaev