pikvm/ustreamer #85: Streaming a virtual camera: Device or resource busy

KUGA2, 2021-01-16:
I create a virtual camera streaming only a static image.
See https://askubuntu.com/a/881341

Then i want to stream this device. This fails:
```
$ ./ustreamer --device /dev/video1
-- INFO  [380693.354      main] -- Using internal blank placeholder
-- INFO  [380693.354      main] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [380693.354    stream] -- Using V4L2 device: /dev/video1
-- INFO  [380693.354    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [380693.355      http] -- Starting HTTP eventloop ...
-- INFO  [380693.355    stream] -- Device fd=11 opened
-- INFO  [380693.355    stream] -- Using input channel: 0
-- ERROR [380693.355    stream] -- Unable to set pixelformat=YUYV, stride=1280, resolution=640x480: Device or resource busy
-- INFO  [380693.355    stream] -- Device fd=11 closed
-- INFO  [380693.355    stream] -- Sleeping 1 seconds before new stream init ...
```

It is probably only a settings issue but i do not know how to resolve it.
_mjpg-streamer_ has a similar issue, _motion_ can stream this device, however it does not output a still image (aka `http://localhost:8080/?action=snapshot`) which I need for OctoPrint.

PS: Streaming my real cam works.
PS: Running as sudo does also not help.

mdevaev, 2021-01-16:
v4l2loopback is broken. It only works for some limited conditions. I have encountered the fact that it does not correctly implement the V4L2 buffer management API (but the author did not fix this, although I pointed out the problem). You are facing another problem. There is nothing to fix on the part of ustreamer, the problem is in v4l2loopback.

KUGA2, 2021-01-17:
I am sad to hear that :(
I am happy that i am coding on a laptop with a real cam then. Otherwise i could not develop my octorpint plugin :(

Off topic (thats why i closed the issue):
Do you know an alternative to simutlate a webcam? Preferably a static image with timestamp?

closed 2021-01-17 by KUGA2

mdevaev, 2021-01-17:
No idea to be honest. For development, I also use a regular webcam, even for Pi-KVM.