pikvm/ustreamer #86: Hook/callback when someone is watching stream

KUGA2, 2021-01-17:
I doubt there is such a feature yet, so probably a feature request:
Can ustreamer call a hook when someone is watching the stream (or a snapshot picture)?

Use case: I want to turn on a light (via raspi GPIO) when the stream is watched.

mdevaev, 2021-01-17:
Build with `WITH_GPIO=1`.

> 
> GPIO options:
> ═════════════
>     --gpio-device </dev/path>  ───── Path to GPIO character device. Default: /dev/gpiochip0.
> 
>     --gpio-consumer-prefix <str>  ── Consumer prefix for GPIO outputs. Default: ustreamer.
> 
>     --gpio-prog-running <pin>  ───── Set 1 on GPIO pin while uStreamer is running. Default: disabled.
> 
>     --gpio-stream-online <pin>  ──── Set 1 while streaming. Default: disabled.
> 
>     **--gpio-has-http-clients <pin>  ─ Set 1 while stream has at least one client. Default: disabled.**

mdevaev, 2021-01-17:
I made these options for debugging with an oscilloscope, I never thought that someone would need it.

KUGA2, 2021-01-17:
Note: you need libgpiod for this. Ubuntu/rasbian:
`sudo apt install libgpiod-dev`

KUGA2, 2021-01-17:
This works for stream but not for snapshot. I will try to modify it.
Maybe i will also change it to run a callback, because you might wanna do something different, like sending a command to your home automation.

mdevaev, 2021-01-17:
I don't think this is a good idea, because it's not a streamer's job. The flexibility of one particular implementation will always be lacking for other needs. Of course, you can do this for yourself, but for the upstream it is unlikely to work.

KUGA2, 2021-01-17:
I agree with you. ustreamer should only do one job and this is streaming.

Any idea how to do this? Maybe set up a webserver in front, that switches the GPIO then forwards the messages.

Sadly, my knowledge of http is too poor to do this.

If you got the spare time, i would appreciate your input on my fork:
https://github.com/KUGA2/ustreamer/tree/feature/gpio_on_snapshot

closed 2021-01-17 by KUGA2

mdevaev, 2021-01-18:
I believe that the correct way is to turn on the backlight when someone is looking at the page. You can make a simple http server in python for this. The page opens a web socket while the socket is open - the light is on.