pikvm/ustreamer #93: Options for rotating/flipping output?

cdthomas9, 2021-02-11:
Is there a way to rotate the image 90 or 180 degrees for the output?

djkraven, 2021-02-11:
You can do this on the web page. Here is a sample of part of the code I am using.

`<style>
#FlipStream {
    transform: rotateY(180deg)rotateX(180deg);
}
</style>
`

`<div id="FlipStream">
                [image]
</div>

`

mdevaev, 2021-02-11:
If the camera does not support this, then nothing can be done. There is no image conversion function via the processor in ustreamer. But the advice above is quite good. Use HTML for any image processing. I do the same thing in Pi-KVM

cdthomas9, 2021-02-12:
OK, thanks. I'm not sure how I can go about putting that into my setup, I'm adding a webcam to a 3D printer and it has to be mounted upside down. The built in web control software has an option to rotate 180 degrees which works fine, but when I click on the image to get a real time view it shows it normal, upside down. I can't seem to find where I would need to add that code in my setup.

mdevaev, 2021-02-18:
Well, here I can't help you, because I don't know what kind of software is used there. In any case, parameters In any case, this functionality does not concern ustreamer directly. If you want to discuss something else, you can do it here or in the new issue.

closed 2021-02-18 by mdevaev