pikvm/ustreamer #95: Questions

FD-, 2021-02-12:
Hi,
I recently discovered your great project and am trying to evaluate whether it provides all I need for an idea of a hobby project. Please forgive me asking questions in an issue report, but I just couldn't find these answered anywhere else (I have read through all previous issues open and closed):

1) What frame rate can I expect from running ustreamer on a Pi Zero (W) using a HDMI to CSI2 bridge at 1080p? What is actually the bottleneck here? Is it the MJPEG encoding?
2) I read that you are working on support for H264 instead of MJPEG. How does decoding on the client side work for this scheme? Will it still run in a web browser (eg through WebRTC)? What about latency and throughput on a Pi Zero W?

I'm looking forward to your response!

mdevaev, 2021-02-17:
Hi!
1. The bottleneck is CSI bus. It has only 2 lanes so 1080p is limited by 25fps.
2. WebRTC or something like that. I'm still experementing.

FD-, 2021-02-17:
Thanks for the response! 

1. Does that mean 25fps 1080p MJPEG can be achieved using the current implementation on a Pi Zero W?

mdevaev, 2021-02-17:
No. Zero is too weak for fast image compression.

FD-, 2021-02-17:
But doesn't the image compression happen on the GPU (which is the same on every Pi)?

mdevaev, 2021-02-17:
GPU is also weak)

FD-, 2021-02-17:
AFAIK, all Pis except Pi 4 use the same VIdeocore IV GPU.

mdevaev, 2021-02-17:
Maybe it's the frequency. In any case zero is very slow and it is impossible to squeeze an acceptable image out of it

FD-, 2021-02-17:
I see. Thanks for the info and I'm looking forward to the h264 support!

mdevaev, 2021-02-18:
Welcome

closed 2021-02-18 by mdevaev