pikvm/ustreamer #96: Unable to set pixelformat=YUYV, resolution=640x480: Invalid argument

ewertonvasconcelos, 2021-02-17:
Error message when trying to use ustreamer with Droidcam camera. The camera works with another applications, such as Cheese.

ewertonvasconcelos, 2021-02-17:
================================================================================
-- INFO  [3706.363    stream] -- Device fd=8 opened
-- INFO  [3706.363    stream] -- Using input channel: 0
-- INFO  [3706.363    stream] -- Using TV standard: DEFAULT
-- ERROR [3706.363    stream] -- Unable to set pixelformat=YUYV, resolution=640x480: Invalid argument
-- INFO  [3706.363    stream] -- Device fd=8 closed
-- INFO  [3706.363    stream] -- Sleeping 1 seconds before new stream init ...
================================================================================

mdevaev, 2021-02-17:
Try another pixel format, for example `-m jpeg`.

ewertonvasconcelos, 2021-02-17:
Hi @mdevaev , thanks for the fast feedback. I tried all the possible formats and it still not working. Checking the device, I can see itts using a not supported format, this 'YU12'.  Do you know if can I change the format of the device (it's  use v4l2loopback for the virtual device) ?

ewerton@matatabi:/opt$ v4l2-ctl --all --device /dev/video0
Driver Info (not using libv4l2):
	Driver name   : Droidcam
	Card type     : Droidcam
	Bus info      : platform:v4l2loopback_dc-000
	Driver version: 4.15.18
	Capabilities  : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x85200001
		Video Capture
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
Priority: 0
Video input : 0 (loopback: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YU12'
	Field             : None
	Bytes per Line    : 640
	Size Image        : 460800
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Streaming Parameters Video Capture:
	Frames per second: 30.000 (30/1)
	Read buffers     : 16
                    keep_format 0x08000000 (bool)   : default=0 value=0
              sustain_framerate 0x08000001 (bool)   : default=0 value=0
                        timeout 0x08000002 (int)    : min=0 max=100000000 step=1 default=0 value=0
               timeout_image_io 0x08000003 (bool)   : default=0 value=0

mdevaev, 2021-02-17:
v4l2loopback is broken by design. I don't know how to make it work, because it even does not properly handle v4l buffers. The author of v4l2loopback doesn't care if he doesn't want to or can't fix it. Try to do without v4l2loopback.

ewertonvasconcelos, 2021-02-17:
Ok, I will try it. thanks for the support! Br's!!

mdevaev, 2021-02-18:
Welcome

closed 2021-02-18 by mdevaev

mfussenegger, 2024-05-16:
Just in case someone else runs into this. I got the format issue sorted out using `v4l2loopback-ctl set-caps /dev/video2 "YUYV:2560x1440"`

Here are the full commands I used to stream my desktop:

```bash
modprobe v4l2loopback exclusive_caps=1 card_label=desktop_view1
v4l2loopback-ctl set-caps /dev/video2 "YUYV:2560x1440"
wf-recorder --muxer=v4l2 --codec=rawvideo --file=/dev/video2 -x yuyv422 -t
ustreamer -d /dev/video2 -p 3000 -r 2560x1440
```

While iterating on the options I sometimes got errors like:

```
unable to set requested format: Device or resource busy
```

Which were solved by reloading the module (`modprobe -r v4l2loopback`)