pikvm/ustreamer #97: Elgato CamLink Unsupported ?

karepiu, 2021-03-06:
I got below with Elgato Cam Link. Any idea ? 

```
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: ================================================================================
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [87.886    stream] -- Device fd=8 opened
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [87.886    stream] -- Using input channel: 0
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [87.966    stream] -- Requested resolution=1920x1080 is unavailable
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [88.046    stream] -- Using resolution: 3840x2160
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [88.046    stream] -- Could not obtain the requested pixelformat=JPEG; driver gave us unsupported
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- ERROR [88.046    stream] -- Unsupported pixelformat=NV12 (fourcc)
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [88.046    stream] -- Device fd=8 closed
[2021-03-01 09:21:19 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: -- INFO  [88.046    stream] -- Sleeping 1 seconds before new stream init ...
[2021-03-01 09:21:20 kvmd.service] --- kvmd.apps.kvmd.streamer           INFO --- Console: ================================================================================
``

mdevaev, 2021-03-06:
uStreamer-specific task. It will be done in the future.

Vincent056, 2023-11-15:
I have a patch added NV24 and NV16 supported needed for rk3588

Eric-168, 2023-11-16:
> I have a patch added NV24 and NV16 supported needed for rk3588 [Vincent056@793f904](https://github.com/Vincent056/ustreamer/commit/793f904c3f3ab2f2d06fcd7549f7dee1f3776482)
> 
> also this is needed, it is tested on orangePi5 Plus: [Vincent056@6a1c16f](https://github.com/Vincent056/ustreamer/commit/6a1c16f5334764c8238f97ec13c17bceb7aefe67)

@Vincent056 
First of all, thank you very much, I am using your rk3588-b branch
The process is this:
git clone -b rk3588-b https://github.com/Vincent056/ustreamer.git
cd ustreamer
make
And replace the successfully compiled ustreamer and ustreamer-dump to /usr/bin
chmod +x ustreamer
chmod +x ustreamer-dump
I use
ustreamer --device=/dev/video0 --host=0.0.0.0
it no longer prompts me
Video capture is not supported by device
Moreover, I can also see the video stream on HTTP port 8080
I don't know what I did wrong but I still can't see the video stream in pikvm

Eric-168, 2023-11-16:
My previous version was uStreamer 5.37. I don’t know if it’s because the new uStreamer 5.43 is incompatible with my pikvm.

Eric-168, 2023-11-16:
[image]
I modified it by
/etc/kvmd/main.yaml
inside
--device=/dev/video0
Now I can get the video stream in pikvm, but I don't know why its color is yellowish. Is it because of the capture card firmware? Thank you very much again.
@Vincent056 @mdevaev

Vincent056, 2023-11-17:
> [image] I modified it by /etc/kvmd/main.yaml inside --device=/dev/video0 Now I can get the video stream in pikvm, but I don't know why its color is yellowish. Is it because of the capture card firmware? Thank you very much again. @Vincent056 @mdevaev

could you post what you have in the end of `journalctl -u kvmd` it might be a issue with where you placed ustreamer bin file I suspect

Eric-168, 2023-11-18:
> > [image] I modified it by /etc/kvmd/main.yaml inside --device=/dev/video0 Now I can get the video stream in pikvm, but I don't know why its color is yellowish. Is it because of the capture card firmware? Thank you very much again. @Vincent056 @mdevaev
> 
> could you post what you have in the end of it might be a issue with where you placed ustreamer bin file I suspect`journalctl -u kvmd`

I was wrong. Regarding the color issue, it was because the captured machine turned on the night mode.

pl804, 2024-09-22:
Hi @Eric-168 @Vincent056

I also have an extra Elgato CamLink lying around that I'm trying to get working with PiKVM, to avoid buying another component.

My supported formats are similar to @Eric-168 (although slightly different):

`root@pikvm bin]# v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'NV12' (Y/UV 4:2:0)
                Size: Discrete 3840x2160
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'NV12' (Y/UV 4:2:0)
                Size: Discrete 3840x2160
                        Interval: Discrete 0.033s (30.000 fps)
        [2]: 'YU12' (Planar YUV 4:2:0)
                Size: Discrete 3840x2160
                        Interval: Discrete 0.033s (30.000 fps)`

Like @Eric-168, I cloned @Vincent056's branch and ran the exact same commands, and edited main.yaml accordingly to also specify video0.

The good news is I no longer get the unsupported device error.

The bad news is I hit an error during capture.

I noticed @Eric-168 somehow was able to set the resolution to 1920x1080.  For some reason, mine won't set to anything other than 4K, even after setting the resolution on the target host to 1024x768.  I was worried 4K was producing output that was too large for PiKVM to handle.

Any idea what I'm doing wrong?

Thanks.

`-- INFO  [602.756      http] -- Starting HTTP eventloop ...
-- INFO  [603.010    stream] -- Device fd=8 opened
-- INFO  [603.010    stream] -- Using input channel: 0
-- INFO  [603.010    stream] -- Probing device format=YUYV, stride=3840, resolution=1920x1080 ...
-- INFO  [603.011    stream] -- Querying device format ...
-- INFO  [603.011    stream] -- Query pix format: width=3840, height=2160, pixelformat=NV12, field=1, bytesperline=3840
-- INFO  [603.011    stream] -- Current device set to for pix: width=1920, height=1080, pixelformat=NV12, field=0, bytesperline=3840
-- INFO  [603.011    stream] -- Supported device formats:
-- INFO  [603.011    stream] --   NV12
-- INFO  [603.011    stream] --   NV12
-- INFO  [603.011    stream] --   unsupported
-- ERROR [603.096    stream] -- Requested resolution=1920x1080 is unavailable
-- INFO  [603.096    stream] -- Probing device format=NV12, stride=7680, resolution=3840x2160 ...
-- INFO  [603.096    stream] -- Querying device format ...
-- INFO  [603.097    stream] -- Query pix format: width=3840, height=2160, pixelformat=NV12, field=1, bytesperline=3840
-- INFO  [603.097    stream] -- Current device set to for pix: width=3840, height=2160, pixelformat=NV12, field=0, bytesperline=7680
-- INFO  [603.097    stream] -- Supported device formats:
-- INFO  [603.097    stream] --   NV12
-- INFO  [603.097    stream] --   NV12
-- INFO  [603.097    stream] --   unsupported
-- INFO  [603.178    stream] -- Using resolution: 3840x2160, size:24883200
-- INFO  [603.178    stream] -- Using format: NV12
-- INFO  [603.324    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [603.324    stream] -- Using IO method: MMAP
-- INFO  [603.408    stream] -- Requested 5 device buffers, got 5
-- INFO  [603.439    stream] -- Capturing started
-- INFO  [603.439    stream] -- Using JPEG quality: 80%
-- INFO  [603.439    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [603.440    stream] -- Capturing ...
-- ERROR [604.441    stream] -- Mainloop select() error: Invalid argument
-- INFO  [604.441    stream] -- Destroying workers pool JPEG ...
-- INFO  [604.443    stream] -- Capturing stopped
-- INFO  [604.459    stream] -- Device fd=8 closed`