pikvm/ustreamer #98: Whether a remote mjpg-server can be used instead of a built-in HTTP server

Luupw3d, 2021-03-07:
Hi！
This is prefect project, but I want to transport mjpeg-stream to remote server to the Internet instead of LAN.How can I do this?
Do you have a plan to do a **Client/Server separation project**？
Thanks a lot.

mdevaev, 2021-03-07:
I'm a little confused, why do you need an mjpg-streamer for these purposes, if ustreamer can do the same?

mdevaev, 2021-03-07:
Ignore the deleted comment, I turned on my head and reread your post again. How do you imagine sending a stream to a remote server? For example, there is nothing stopping you from using nginx right now to restream the video.

Luupw3d, 2021-03-07:
Use the  `--host MY_SERVER_IP` and `--port MY_SERVER_PORT` and how should I receive MJPG-Stream on my server？Just install the Nginx? Sorry for this stupid question because I'm a rookie in this field:(

mdevaev, 2021-03-09:
@Luupw3d Hey, I'm not refusing to solve your problem. I'm just like a squirrel in a wheel right now because of the release of my hardware. Sorry, I completely forgot to write to you. The host and port parameters are responsible for the native ustreamer server. It does not support push data, which means that your remote web server must pull the video stream.

Like this:
* https://github.com/pikvm/kvmd/blob/master/configs/nginx/kvmd.ctx-http.conf#L5
* https://github.com/pikvm/kvmd/blob/master/configs/nginx/kvmd.ctx-server.conf#L86

Luupw3d, 2021-03-10:
Got it.I've successfully implemented server functionality, but as you said, the client can't actively push the stream, only the server can actively pull the stream.

mdevaev, 2021-03-14:
In theory, you can push a stream using rtsp and gstreamer, if that means anything to you. Ustreamer can write a stream to memory, this stream can be extracted and fed to gstreamer, and the latter can push anywhere in a format that suits you. However, I don't guarantee that there will be no delays in this scheme. If I can vouch for the speed of ustreamer, then gstreamer will add 150-300 ms.

closed 2021-05-21 by Luupw3d