pikvm/ustreamer #99: Only one camera work at a time

oh6fub, 2021-03-14:
I have tested mjpg-streamer with four (4) webcameras (Live! Cam Sync 1080p) and apart from limitations and issues with mjpg-streamer, I can all of them stream simultaneously on Raspberry Pi 4.

However, with ustreamer I can only get one work at a time. Next camera will just stay in infinite loop, waiting for something eternally:

-- DEBUG [5590.448    stream] -- Waiting for worker ...
-- DEBUG [5590.448    stream] -- Calling select() on video device ...
-- DEBUG [5590.448      jw-0] -- Hello! I am a worker jw-0 ^_^
-- DEBUG [5590.448      jw-0] -- Worker jw-0 waiting for a new job ...
-- DEBUG [5591.456      http] -- HTTP: Repeating exposed ...
-- DEBUG [5592.461      http] -- HTTP: Repeating exposed ...
-- DEBUG [5593.471      http] -- HTTP: Repeating exposed ...
-- DEBUG [5594.493      http] -- HTTP: Repeating exposed ...
-- DEBUG [5595.453    stream] -- Device select() --> 0
-- ERROR [5595.453    stream] -- Mainloop select() error: Success
-- INFO  [5595.453    stream] -- Destroying workers pool JPEG ...
-- DEBUG [5595.453      jw-0] -- Bye-bye (worker jw-0)
-- DEBUG [5595.453    stream] -- Calling ioctl(VIDIOC_STREAMOFF) ...
-- INFO  [5595.455    stream] -- Capturing stopped
-- DEBUG [5595.455    stream] -- Releasing device buffers ...
-- DEBUG [5595.455    stream] -- Closing device ...
-- INFO  [5595.456    stream] -- Device fd=8 closed

and so on. Any ideas?

[device_first_unit.txt](https://github.com/pikvm/ustreamer/files/6137239/device_first_unit.txt)
[device_second_unit.txt](https://github.com/pikvm/ustreamer/files/6137241/device_second_unit.txt)

mdevaev, 2021-03-14:
I think I've seen it somewhere before. This is a driver bug. It. cannot be fixed on the ustreamer side.

oh6fub, 2021-03-14:
Hmm, all right, I will see if something can be done about it... Creative doesn't publish any drivers as the camera should be uvc compatible. I'll post my findings.

mdevaev, 2021-05-30:
What's up?

djkraven, 2021-06-15:
I am getting something similar using a Pi 4 with two USB webcam attached. The same setup on a Pi 3 works. Initially, on the Pi4, both cameras were working but not flipped for the page/port I want them to use. After several attempts to get the systemctl service flipped I was able to get the one cam on the page it belonged on but then the second started failing.

` /usr/bin/ustreamer --process-name-prefix ustreamer-2 --log-level 3 --device /dev/video2 --device-timeout=10  --quality  40 --resolution 1280x720  --desired-fps=29 --host=0.0.0.0 --port=8089 --static /var/www/html/ustreamer-2/ -I USERPTR --format MJPEG
-- INFO  [30966.215      main] -- Using internal blank placeholder
-- DEBUG [30966.215      main] -- Installing SIGINT handler ...
-- DEBUG [30966.215      main] -- Installing SIGTERM handler ...
-- DEBUG [30966.215      main] -- Ignoring SIGPIPE ...
-- INFO  [30966.216      main] -- Enabling HTTP file server: /var/www/html/ustreamer-2/
-- DEBUG [30966.216      main] -- Binding HTTP to [0.0.0.0]:60809 ...
-- INFO  [30966.216      main] -- Listening HTTP on [0.0.0.0]:60809
-- INFO  [30966.217    stream] -- Using V4L2 device: /dev/video2
-- INFO  [30966.217    stream] -- Using desired FPS: 29
-- DEBUG [30966.217    stream] -- _stream_init_loop: stream->run->stop=0
-- INFO  [30966.217      http] -- Starting HTTP eventloop ...
================================================================================
-- DEBUG [30966.235      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [30966.235      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
-- INFO  [30966.311    stream] -- Device fd=8 opened
-- DEBUG [30966.311    stream] -- Calling ioctl(VIDIOC_QUERYCAP) ...
-- INFO  [30966.311    stream] -- Using input channel: 0
-- DEBUG [30966.311    stream] -- Using TV standard: DEFAULT
-- DEBUG [30966.311    stream] -- Calling ioctl(VIDIOC_S_FMT) ...
-- INFO  [30966.327    stream] -- Using resolution: 1280x720
-- INFO  [30966.327    stream] -- Using pixelformat: MJPEG
-- DEBUG [30966.327    stream] -- Calling ioctl(VIDIOC_G_PARM) ...
-- INFO  [30966.344    stream] -- Using HW FPS: 29 -> 30 (coerced)
-- ERROR [30966.344    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [30966.344    stream] -- Using IO method: USERPTR
-- DEBUG [30966.344    stream] -- Calling ioctl(VIDIOC_REQBUFS) for V4L2_MEMORY_USERPTR ...
-- INFO  [30966.344    stream] -- Requested 5 device buffers, got 5
-- DEBUG [30966.344    stream] -- Allocating device buffers ...
-- DEBUG [30966.370    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 0 ...
-- DEBUG [30966.372    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 1 ...
-- DEBUG [30966.373    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 2 ...
-- DEBUG [30966.374    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 3 ...
-- DEBUG [30966.375    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 4 ...
-- DEBUG [30966.377    stream] -- Device fd=8 initialized
-- DEBUG [30966.377    stream] -- Calling ioctl(VIDIOC_STREAMON) ...
-- INFO  [30966.385    stream] -- Capturing started
-- INFO  [30966.385    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [30966.385    stream] -- Using JPEG quality: encoder default
-- INFO  [30966.385    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [30966.385    stream] -- Capturing ...
-- DEBUG [30966.385      jw-0] -- Hello! I am a worker jw-0 ^_^
-- DEBUG [30966.385      jw-0] -- Worker jw-0 waiting for a new job ...
--------------------------------------------------------------------------------
-- DEBUG [30966.386    stream] -- Waiting for worker ...
-- DEBUG [30966.386    stream] -- Calling select() on video device ...
-- DEBUG [30966.451    stream] -- Device select() --> 1
-- DEBUG [30966.451    stream] -- Frame is ready
-- DEBUG [30966.451    stream] -- Grabbing device buffer ...
-- DEBUG [30966.451    stream] -- Grabbed new frame in device buffer: index=0, bytesused=0
-- DEBUG [30966.451    stream] -- Dropped too small frame sized 0 bytes, assuming it was broken
-- DEBUG [30966.451    stream] -- Releasing device buffer index=0 (broken frame) ...
--------------------------------------------------------------------------------
-- DEBUG [30966.452    stream] -- Waiting for worker ...
-- DEBUG [30966.452    stream] -- Calling select() on video device ...
-- DEBUG [30966.452    stream] -- Device select() --> 1
-- DEBUG [30966.452    stream] -- Frame is ready
-- DEBUG [30966.452    stream] -- Grabbing device buffer ...
-- DEBUG [30966.452    stream] -- Grabbed new frame in device buffer: index=1, bytesused=0
-- DEBUG [30966.452    stream] -- Dropped too small frame sized 0 bytes, assuming it was broken
-- DEBUG [30966.452    stream] -- Releasing device buffer index=1 (broken frame) ...
--------------------------------------------------------------------------------
-- DEBUG [30966.453    stream] -- Waiting for worker ...
-- DEBUG [30966.453    stream] -- Calling select() on video device ...
-- DEBUG [30966.453    stream] -- Device select() --> 1
-- DEBUG [30966.453    stream] -- Frame is ready
-- DEBUG [30966.453    stream] -- Grabbing device buffer ...
-- DEBUG [30966.453    stream] -- Grabbed new frame in device buffer: index=2, bytesused=0
[... 260 lines cut ...]
-- DEBUG [30977.457    stream] -- Frame is ready
-- DEBUG [30977.457    stream] -- Grabbing device buffer ...
-- DEBUG [30977.457    stream] -- Grabbed new frame in device buffer: index=0, bytesused=0
[... cut ...]
-- DEBUG [30977.459    stream] -- Device select() --> 1
-- DEBUG [30977.459    stream] -- Frame is ready
-- DEBUG [30977.459    stream] -- Grabbing device buffer ...
-- DEBUG [30977.459    stream] -- Grabbed new frame in device buffer: index=2, bytesused=0
-- DEBUG [30977.459    stream] -- Dropped too small frame sized 0 bytes, assuming it was broken
-- DEBUG [30977.460    stream] -- Releasing device buffer index=2 (broken frame) ...
--------------------------------------------------------------------------------
`

djkraven, 2021-06-17:
I have success getting both webcams to run. It appears the issue may be related to the USB hub on the Pi4. After trying a different combination of which port they were plugged into the second finally came on. It was stable for a while yesterday then it went to No Imagine. I have updated the EEPROM to the latest this morning and will see if that improves the issue. The failing camera would disappear from the lsusb listing which lead to looking into the USB hub.

mdevaev, 2021-07-04:
Okay, I think the problem is solved here.

closed 2021-07-04 by mdevaev